zephyrproject-rtos/zephyr #30394: TLS tests failing with sanitycheck (under load)

nashif, 2020-12-02:
**Describe the bug**
Getting this everytime I run sanitycheck with default options:

```
gale:zephyr(master): sanitycheck -cM
Deleting output directory /home/anashif/zephyrproject/zephyr/sanity-out
INFO    - Zephyr version: zephyr-v2.4.0-1953-gf68146f74e
INFO    - JOBS: 112
INFO    - Selecting default platforms per test case
INFO    - Building initial testcase list...
INFO    - 1040 test suites (13542 configurations) selected, 18921 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:  813/5418  15%  skipped: 8269, failed:    0
ERROR   - native_posix              samples/subsys/usb/hid/sample.usb.hid_composite    FAILED: Failed
ERROR   - see: /home/anashif/zephyrproject/zephyr/sanity-out/native_posix/samples/subsys/usb/hid/sample.usb.hid_composite/handler.log
INFO    - Total complete: 3423/5418  63%  skipped: 8740, failed:    1
ERROR   - mps2_an385                tests/kernel/threads/tls/kernel.threads.tls.userspace FAILED: Timeout
ERROR   - see: /home/anashif/zephyrproject/zephyr/sanity-out/mps2_an385/tests/kernel/threads/tls/kernel.threads.tls.userspace/handler.log
INFO    - Total complete: 3438/5418  63%  skipped: 8742, failed:    2
ERROR   - mps2_an521                tests/kernel/threads/tls/kernel.threads.tls.userspace FAILED: Timeout
ERROR   - see: /home/anashif/zephyrproject/zephyr/sanity-out/mps2_an521/tests/kernel/threads/tls/kernel.threads.tls.userspace/handler.log
INFO    - Total complete: 3755/5418  69%  skipped: 8822, failed:    3

```

**Expected behavior**
should pass

**Impact**
Something is not quite right with TLS that we should look at.

**Logs and console output**

Failing test output:
```
*** Booting Zephyr OS build zephyr-v2.4.0-1953-gf68146f74ec9  ***
Running test suite thread_tls
===================================================================
START - test_tls
thread 0: result 0 (expecting 0)
thread 1: result 0 (expecting 0)
thread 2: result 0 (expecting 0)
 PASS - test_tls
===================================================================
Test suite thread_tls succeeded
Running test suite thread_tls_userspace
===================================================================
START - test_tls_userspace
E: 0x200 is not a valid z_thread_stack_element
E: address is not a known kernel object
E: syscall z_vrfy_k_thread_create failed check: bad stack object
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x20000318 (unknown)
E: Halting system
```
**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.12
 - Commit: Tip of master

dcpleung, 2020-12-02:
Looks like the same issue as #30393

nashif, 2020-12-02:
lets close this one then

closed 2020-12-02 by nashif