zephyrproject-rtos/zephyr #30395: Add possibility to use alternative list of platforms default for testing.

PerMac, 2020-12-02:
**Is your enhancement proposal related to a problem? Please describe.**
This relates mostly to the usage of sanitycheck in downstream CIs. 
A set of platforms to be used can be given with `-p` argument. However, using it filters out all platforms which were not given in this way. In particular, this prevents one from using "integration mode" and being able to expand the list of platforms to be tested on the test level (with `integration_platforms` list in the test/sample yaml. 

**Describe the solution you'd like**
Sanitycheck will be called with extra arg, e.g. `--alt-default` and a path to the file containing the list of alt-default platforms. This list will then replace the list of default platforms based on boards.yamls attribute. 

I think the proposed solution would be useful for downstream CIs. The default list is useful only for the upstream CI.  This seems like a non-intrusive patch that will allow downstream CIs to also use "integration mode" with their own set of default platforms and benefit from listing `integration_platforms` in non-upstream tests and plans.

nashif, 2020-12-02:
> A set of platforms to be used can be given with `-p` argument. However, using it filters out all platforms which were not given in this way. In particular, this prevents one from using "integration [...]

so if I understand this correctly, with `-p board1 -T test/test1 -G` you want to be able to also run whatever platforms listed as integration_platforms in testcase.yaml of test1 on top of board1?

When you use '-p board1 -p board2', this basically becomes you default platforms, not sure yet I understand how you want this to behave with `-G`

PerMac, 2020-12-02:
> so if I understand this correctly, with -p board1 -T test/test1 -G you want to be able to also run whatever platforms listed as integration_platforms in testcase.yaml of test1 on top of board1?

Yes. This is the effect I would like to achieve. Maybe sometimes also using "integration_platforms" to limit the scope.  

> When you use '-p board1 -p board2', this basically becomes you default platforms, not sure yet I understand how you want this to behave with -G
But using `-p` works as a filter and blocks whatever can be found in `integration_platforms` so does not work with -G. I think allowing for an alt-default list wouldn't require breaking the main usage of `-p` (to be a filter) and will add benefits of flexibility in downstream CIs as it should work well with -G. Mostly for tests not in the upstream, when downstream users could have their own tests with yamls using "integration_platforms"

But maybe I am missing something or another approach would be better?

nashif, 2020-12-02:
> Yes. This is the effect I would like to achieve. Maybe sometimes also using "integration_platforms" to limit the scope.

but this violates the contract where sanitycheck should only be building/running on platforms you specify with -p. If you only have board1, why would you want to deal with something that you do not have just because it is part of integration_platforms?

PerMac, 2020-12-02:
This feature is not meant for --device-testing, maybe hence the confusion. 
I came to this idea directly from the use case of non-upstream CI. We are running some non-upstream tests based on sanitycheck, developed internally. I would like to be able to mimic the functionality of the upstream CI. E.g by default, on-commit, only tests for plat_A, plat_B and plat_C will be run/build. If needed, an extra platform can be added to the "integration_platforms" for a certain test so it is also built/tested on the given platform in the on-commit CI. 
I think this approach increases the maintainability and flexibility of the CI. Currently, we have to use `-p` when calling sanitycheck in the CI, which prevents the flexibility of "integraton_platform" listing on the test level.

PerMac, 2020-12-02:
For example, I think this should help for sdk-nrf repo (https://github.com/nrfconnect/sdk-nrf), where some sanitycheck based tests are running on-commit. 
Why have you thought that adding `integration_platforms` and -G option was beneficial for the upstream CI? I guess the same reasons will apply to other CIs :)

nashif, 2020-12-08:
One option that would give us this:

Define board groups, so instead of 

```
testing:
  default: true
```

we would have:

```
testing:
  groups:
    - default
    - nordic
    - bluetooth
```

This way you can select the group instead of having to feed a list of boards.

PerMac, 2020-12-09:
I am worried that this solution would be harder to maintain. In the sdk-nrf CI we would probably prefer to have higher flexibility, not relying on updating the upstream, whenever we decide to change the scope of tested platforms in our own plan. And we are already directly feeding a list of boards with `-p`.
 
Another solution could be that if  `-G` is used it adds/filters platforms from `integration_platforms` also when `-p` is used, but to me, it seems more confusing than using sth like `-G -alt-default LIST`

closed 2021-05-17 by PerMac