zephyrproject-rtos/zephyr #30397: tests:latency_measure is not counting semaphore results on the ARM boards

maksimmasalski, 2020-12-02:
**Describe the bug**
When I run latency_measure test on the ARM boards (reel board and the Microchip board) semaphore results are empty
```
DEBUG   - DEVICE: Semaphore take time (context switch)                        :       0 cycles ,        0 ns
DEBUG   - DEVICE: Semaphore give time (context switch)                        :       0 cycles ,        0 ns
```
I tested on the x86 up_squared board, it works
```
Semaphore take time (context switch)                        :    1274 cycles ,    12740 ns
Semaphore give time (context switch)                        :    1808 cycles ,    18080 ns
```
on qemu it works too
```
Semaphore take time (context switch)                        :    1274 cycles ,    12740 ns
Semaphore give time (context switch)                        :    1808 cycles ,    18080 ns
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use ARM board
2. Run
`sanitycheck --device-testing --hardware-map ~/reel_board_hw_map.file -p reel_board -T tests/benchmarks/latency_measure/ -vv`
3. See handler.log
or use west
2. Run
`west build  -p -b reel_board tests/benchmarks/latency_measure/`
3. Run
`west flash`
4. See the output

**Expected behavior**
Test have to provide all the results.

**Impact**
No data to know Semaphore take time (context switch) and Semaphore give time (context switch) duration on the ARM board.

**Logs and console output**
Above

**Environment (please complete the following information):**
 - OS: Fedora/Ubuntu
 - Toolchain Zephyr SDK zephyr-sdk-0.11.4
 - Commit SHA b44a90c9e2b61a209551d492b381109ad40387a0

maksimmasalski, 2020-12-02:
@dcpleung Could you take a look?

nashif, 2020-12-03:
was introduced in ```
commit 9be37553eee22bc16f8740fa2e2ad17b7bccd509 (HEAD, refs/bisect/bad)
Author: Daniel Leung <daniel.leung@intel.com>
Date:   Wed Oct 14 11:15:40 2020 -0700

    timing: do not repeatedly do init()/start()/stop()

    We should not be initializing/starting/stoping timing functions
    multiple times. So this changes how the timing functions are
    structured to allow only one initialization, only start when
    stopped, and only stop when started.

    Signed-off-by: Daniel Leung <daniel.leung@intel.com>
```

PR #30666 (2021-01-04): timing: fix timing_stop() ref counting
When there are more timing_stop() calls then timing_start(),
the reference counter will go negative, resulting in the next
timing_start() call not starting the timer. Without timer
running, getting cycles elasped would not work. So fix
the ref counting so it won't dip below zero.

Fixes #30397
commit 6ab48865 (2021-01-04): timing: fix timing_stop() ref counting

When there are more timing_stop() calls then timing_start(),
the reference counter will go negative, resulting in the next
timing_start() call not starting the timer. Without timer
running, getting cycles elasped would not work. So fix
the ref counting so it won't dip below zero.

Fixes #30397
closed 2021-01-04 by ioannisg