zephyrproject-rtos/zephyr #30408: CI build error on scheduler testing

yingmingx, 2020-12-03:
When I submit a new PR about scheduler testing, CI build error. So I resubmit a PR just add some notes to reconfirm the issue. 
See https://github.com/zephyrproject-rtos/zephyr/pull/30379.
Maybe there are some bug on qemu arc platform.

nashif, 2020-12-03:
@yingmingx Please submit bugs using the template and provide more context, pointing a PR that fails is not really needed if you can reproduce this locally and can provide the failure details directly here.

The issue:

The test tests/kernel/sched/schedule_api is failing after 0875740a3457ecfddc70c93cacd44932f64058a4 on m2gl025_miv:

```
10:22:15.6576 [WARNING] plic: Unhandled write to offset 0x200000, value 0x0.
10:22:15.6633 [WARNING] uart: Unhandled write to offset 0xC. Unhandled bits: [0] when writing value 0x1. Tags: BIT8 (0x1).
10:22:15.6717 [DEBUG] uart: [+0.41s host +0.1ms virt 0.1ms virt from start] *** Booting Zephyr OS build zephyr-v2.4.0-1999-g349c50a6826d  ***
10:22:15.6762 [DEBUG] uart: [+4.53ms host +0.1ms virt 0.2ms virt from start]   Running test suite threads_scheduling
10:22:15.6767 [DEBUG] uart: [+0.49ms host +0s virt 0.2ms virt from start]   ===================================================================
10:22:15.6769 [DEBUG] uart: [+0.25ms host +0s virt 0.2ms virt from start]   START - test_bad_priorities
10:22:15.6809 [DEBUG] uart: [+3.94ms host +0.1ms virt 0.3ms virt from start]    PASS - test_bad_priorities
10:22:15.6813 [DEBUG] uart: [+0.5ms host +0s virt 0.3ms virt from start]   ===================================================================
10:22:15.6816 [DEBUG] uart: [+0.28ms host +0s virt 0.3ms virt from start]   START - test_priority_cooperative
10:22:15.6872 [DEBUG] uart: [+5.61ms host +1.2ms virt 1.5ms virt from start]    PASS - test_priority_cooperative
10:22:15.6878 [DEBUG] uart: [+0.51ms host +0s virt 1.5ms virt from start]   ===================================================================
10:22:15.6882 [DEBUG] uart: [+0.43ms host +0.1ms virt 1.6ms virt from start]   START - test_priority_preemptible
10:22:15.6908 [DEBUG] uart: [+2.61ms host +1.1ms virt 2.7ms virt from start]    PASS - test_priority_preemptible
10:22:15.6913 [DEBUG] uart: [+0.54ms host +0.1ms virt 2.8ms virt from start]   ===================================================================
10:22:15.6916 [DEBUG] uart: [+0.33ms host +0s virt 2.8ms virt from start]   START - test_priority_preemptible_wait_prio
10:22:15.6953 [DEBUG] uart: [+3.6ms host +0.7ms virt 3.5ms virt from start]    PASS - test_priority_preemptible_wait_prio
10:22:15.6957 [DEBUG] uart: [+0.49ms host +0s virt 3.5ms virt from start]   ===================================================================
10:22:15.6961 [DEBUG] uart: [+0.41ms host +0.1ms virt 3.6ms virt from start]   START - test_yield_cooperative
10:22:15.6975 [DEBUG] uart: [+1.31ms host +0.1ms virt 3.7ms virt from start]    PASS - test_yield_cooperative
10:22:15.6990 [DEBUG] uart: [+1.51ms host +0s virt 3.7ms virt from start]   ===================================================================
10:22:15.6992 [DEBUG] uart: [+0.24ms host +0s virt 3.7ms virt from start]   START - test_sleep_cooperative
10:22:15.7017 [DEBUG] uart: [+2.51ms host +1.1ms virt 4.8ms virt from start]    PASS - test_sleep_cooperative
10:22:15.7023 [DEBUG] uart: [+0.54ms host +0.1ms virt 4.9ms virt from start]   ===================================================================
10:22:15.7025 [DEBUG] uart: [+0.29ms host +0s virt 4.9ms virt from start]   START - test_sleep_wakeup_preemptible
10:22:15.7035 [DEBUG] uart: [+0.94ms host +0.1ms virt 5ms virt from start]    PASS - test_sleep_wakeup_preemptible
10:22:15.7040 [DEBUG] uart: [+0.49ms host +0s virt 5ms virt from start]   ===================================================================
10:22:15.7043 [DEBUG] uart: [+0.36ms host +0.1ms virt 5.1ms virt from start]   START - test_pending_thread_wakeup
10:22:15.7050 [DEBUG] uart: [+0.65ms host +0s virt 5.1ms virt from start]    PASS - test_pending_thread_wakeup
10:22:15.7056 [DEBUG] uart: [+0.6ms host +0.1ms virt 5.2ms virt from start]   ===================================================================
10:22:15.7059 [DEBUG] uart: [+0.31ms host +0s virt 5.2ms virt from start]   START - test_time_slicing_preemptible
10:22:15.9017 [DEBUG] uart: [+0.2s host +5.1ms virt 10.3ms virt from start]    PASS - test_time_slicing_preemptible
10:22:15.9032 [DEBUG] uart: [+1.48ms host +0s virt 10.3ms virt from start]   ===================================================================
10:22:15.9040 [DEBUG] uart: [+0.82ms host +0.1ms virt 10.4ms virt from start]   START - test_time_slicing_disable_preemptible
10:22:16.0938 [DEBUG] uart: [+0.19s host +5ms virt 15.4ms virt from start]    PASS - test_time_slicing_disable_preemptible
10:22:16.0944 [DEBUG] uart: [+0.61ms host +0.1ms virt 15.5ms virt from start]   ===================================================================
10:22:16.0947 [DEBUG] uart: [+0.32ms host +0s virt 15.5ms virt from start]   START - test_lock_preemptible
10:22:16.1353 [DEBUG] uart: [+40.5ms host +2.1ms virt 17.6ms virt from start]    PASS - test_lock_preemptible
10:22:16.1369 [DEBUG] uart: [+1.61ms host +0.1ms virt 17.7ms virt from start]   ===================================================================
10:22:16.1373 [DEBUG] uart: [+0.51ms host +0s virt 17.7ms virt from start]   START - test_unlock_preemptible
10:22:16.1743 [DEBUG] uart: [+36.89ms host +1ms virt 18.7ms virt from start]    PASS - test_unlock_preemptible
10:22:16.1749 [DEBUG] uart: [+0.66ms host +0.1ms virt 18.8ms virt from start]   ===================================================================
[... cut ...]
10:22:16.4423 [DEBUG] uart: [+0.57ms host +0.1ms virt 42.6ms virt from start]   ===============================================================
====
10:22:16.4426 [DEBUG] uart: [+0.26ms host +0s virt 42.6ms virt from start]   START - test_priority_scheduling
10:22:16.4458 [DEBUG] uart: [+3.21ms host +0.4ms virt 43ms virt from start]   ABCDEFGHI
10:22:16.4491 [DEBUG] uart: [+3.26ms host +0.5ms virt 43.5ms virt from start]   ABCDEFGHI
10:22:16.4518 [DEBUG] uart: [+2.69ms host +0.6ms virt 44.1ms virt from start]   ABCDEFGHI
10:22:16.4550 [DEBUG] uart: [+3.18ms host +0.6ms virt 44.7ms virt from start]   ABCDEFGHI
:

```

yingmingx, 2020-12-04:
@nashif  Your suggestion is helpful. I'll provide more information when I report bugs later.

nashif, 2021-01-05:
@andyross  any updates on this?

PerMac, 2021-01-11:
This issue is a duplicate of #29705

yingmingx, 2021-01-12:
Close this issue because this is duplicate of #29705

yingmingx, 2021-01-25:
@ PerMac  This issue fail on test_slice_scheduling, but you mentioned issue #29705 failed on test_slice_reset. May be they are different.

andyross, 2021-01-27:
@yingmingx : This really does look like a duplicate of the timeslicing bug @ceolin has been working.  There is a fix up at https://github.com/zephyrproject-rtos/zephyr/pull/30916 , can you try that please?

nashif, 2021-01-27:
@andyross  #30916 does not fix this bug, in fact, it is failing in CI on that PR

yingmingx, 2021-01-28:
@andyross  I patch ceolin's code on this testing PR https://github.com/zephyrproject-rtos/zephyr/pull/30379. It still failed.

INFO    -  9/40 m2gl025_miv               tests/kernel/sched/schedule_api/kernel.scheduler.multiq FAILED Failed (renode 9.791s)

ceolin, 2021-01-28:
> @andyross I patch ceolin's code on this testing PR #30379. It still failed.
> 
> INFO - 9/40 m2gl025_miv tests/kernel/sched/schedule_api/kernel.scheduler.multiq FAILED Failed (renode 9.791s)

this target contains another problem that that pr does not fix indeed.

andyross, 2021-01-28:
It's a soc-level kconfig bug: https://github.com/zephyrproject-rtos/zephyr/issues/31726

Unfortunately with my toddler-level renode knowledge, all I can submit is a workaround.  I'm not sure who the right owner should be.

andyross, 2021-02-01:
There's a workaround patch (a full fix requires some work with renode configuration) at #31841

andyross, 2021-02-03:
Close this, as the root cause and workaround are tracked in #31726

closed 2021-02-03 by andyross