zephyrproject-rtos/zephyr #30433: zephyr client automatic joiner failed on nRF52840dk

IHIDCHAOS, 2020-12-04:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
I want to use the auto-join feature in this [link](https://github.com/zephyrproject-rtos/zephyr/pull/13926)
I use echo_server as an example.
I added the following option to the overlay-ot.conf configuration file:
```
CONFIG_OPENTHREAD_JOINER=y
CONFIG_OPENTHREAD_JOINER_AUTOSTART=y
CONFIG_OPENTHREAD_JOINER_PSKD="J01NME"
CONFIG_OPENTHREAD_SLAAC=y
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=6
```
But my CLI doesn't automatically join the Thread network.
My operation on the OTBR is as follows:
```
pi@raspberrypi:~ $ sudo ot-ctl commissioner start
Commissioner: petitioning
Done
pi@raspberrypi:~ $ sudo ot-ctl commissioner joiner add f4ce36f896f4868d J01NME
Done
```
Zephyr Cli's logs are as follows:
```
uart:~$ kernel reboot cold

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-2040-gd66d748b710e  ***
[00:00:00.254,577] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.267,303] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.267,333] <inf> fs_nvs: alloc wra: 6, ba8
[00:00:00.267,333] <inf> fs_nvs: data wra: 6, 780
[00:00:00.320,892] <inf> net_l2_openthread: State changed! Flags: 0x001fc300 Current role: 0
[00:00:00.321,228] <inf> net_l2_openthread: OpenThread version: OPENTHREAD/g07f430dac; Zephyr; Dec  4 2020 05:40:08
[00:00:00.333,984] <inf> net_l2_openthread: Network name: OpenThreadDemo
[00:00:00.389,465] <inf> net_l2_openthread: State changed! Flags: 0x0100103d Current role: 1
[00:00:00.395,904] <inf> net_config: Initializing network
[00:00:00.396,026] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.396,453] <inf> net_echo_server_sample: Run echo server
[00:00:00.396,484] <inf> net_echo_server_sample: Network connected
[00:00:00.396,514] <inf> net_echo_server_sample: Starting...
[00:00:00.396,667] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:02.402,587] <inf> net_l2_openthread: State changed! Flags: 0x00000040 Current role: 1
[00:00:05.005,737] <inf> net_l2_openthread: State changed! Flags: 0x000012a5 Current role: 4

```
When I started running it again on the OTBR, and commissioning it, and trying to manually join the Thread network:
```
uart:~$ ot state
leader
Done
uart:~$ ot thread stop
Done
uart:~$ ot eui64
f4ce36f896f4868d
Done
uart:~$ ot joiner start J01NME
Done
[00:01:13.954,345] <inf> net_l2_openthread: State changed! Flags: 0x0800000b Current role: 0
Join failed [NotFound]

```
**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr/
2. west build -s samples/net/sockets/echo_server/ -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf"
3. west flash
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used
commit d66d748b710ef7520d25d01752ea8a8c89c9e788 (HEAD -> master, origin/master, origin/HEAD)
Author: Xavier Chapron <xavier.chapron@stimio.fr>
Date:   Thu Nov 19 11:41:01 2020 +0100

**Additional context**
Add any other context about the problem here.

rlubos, 2020-12-04:
Tested current master and two nRF52840 devices against each other, autojoin works just fine for me.

It seems that you have not erased the OT settings area on flash (`west flash --erase`). Looking at the logs, your device already had a dataset stored and it simply formed a network. The startup code won't bother with the joining procedure if it has a valid dataset stored in flash.

As for the second, manual attempt, I can't really say. Perhaps the joiner was removed by the commissioner due to a timeout from its joiner list (the default timeout is 2 minutes AFAIR, but I might be wrong here). In any case, it would be way easier to tell what's going on with Wireshark logs provided.

IHIDCHAOS, 2020-12-07:
@rlubos Thank you! After I used the option you suggested, everything is ok!

rlubos, 2020-12-07:
I'll take the liberty and close the issue then.

closed 2020-12-07 by rlubos