zephyrproject-rtos/zephyr #30434: Memory map executing test case failed when code coverage enabled in x86_64 platform

enjiamai, 2020-12-04:
**Describe the bug**

This bug was found when I try to enable the code coverage feather for qemu_x86_64. Please refer to PR https://github.com/zephyrproject-rtos/zephyr/pull/30150
When using gcov to generate code coverage data,  it will stub some extra code and variables into the origin code first in order to record statistical data.

But for one special test case in X86 64 platform, it would cause a unexcepted fault.
See tests/kernel/mem_protect/mem_map/src/main.c , one test case called “test_z_mem_map_exec()”
The error message is here:

START - test_z_mem_map_exec
E: Page fault at address 0xc0021e5e (error code 0x0)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to read
E: PDPTE: not present
E: RAX: 0x000000000014c73f RBX: 0x0000000000131f40 RCX: 0x0000000000130001 RDX: 0x00000000bfff4000
E: RSI: 0x0000000000000000 RDI: 0x000000000014c73f RBP: 0x000000000014c720 RSP: 0x000000000014c720
E:  R8: 0x0000000000000008  R9: 0xffffffffffffffff R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x0000000000000001 R13: 0x0000000000000000 R14: 0x0000000000000000 R15: 0x0000000000000000
E: RSP: 0x000000000014c720 RFLAGS: 0x0000000000000202 CS: 0x0018 CR3: 0x0000000000178000
E: call trace:
E: RIP: 0x00000000bfff4008
E:      0x00000000001004f2
E:      0x000000000010c64d
E:      0x000000000010c7bb
E:      0x0000000000101053
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x131e80 (ztest_thread)
Caught system error -- reason 0
Unexpected fault during testFAILED: zephyr/CMakeFiles/run
cd /home/mm/zephyrproject/zephyr/build && /home/mm/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-x86_64 -m 9 -cpu qemu64,+x2apic -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -device loader,file=/home/mm/zephyrproject/zephyr/build/zephyr/zephyr-qemu-main.elf -smp cpus=2 -kernel /home/mm/zephyrproject/zephyr/build/zephyr/zephyr-qemu-locore.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/mm/zephyrproject/zephyr/build --target run

And I tried to diagnose this issue, following message is what I know so far:
In this test case, we want to map memory for using code snippet of  transplanted_function().
And gcov also stub into this function:  (this assembly code comes from zephyr.lst)

0000000000100412 <transplanted_function>:
#ifndef SKIP_EXECUTE_TESTS
static void transplanted_function(bool *executed)
{
  100412:       55                                 push   %rbp
  100413:       48 89 e5                       mov    %rsp,%rbp
  100416:       48 89 7d f8                   mov    %rdi,-0x8(%rbp)
  **10041a:       48 8b 05 4f de 02 00    mov    0x2de4f(%rip),%rax         # 12e270 <__gcov0.transplanted_function>**
  100421:       48 83 c0 01                   add    $0x1,%rax
  100425:       48 89 05 44 de 02 00    mov    %rax,0x2de44(%rip)        # 12e270 <__gcov0.transplanted_function>
        *executed = true;
  10042c:       48 8b 45 f8                    mov    -0x8(%rbp),%rax
  100430:       c6 00 01                        movb   $0x1,(%rax)
}
  100433:       90                                 nop
  100434:       5d                                 pop    %rbp
  100435:       c3                                 retq

As you can see, in x86_64 platform, the stubbed gcov variable addressing here is indirect addressing by %rip.
But in the memory mapping case, the %rip will be , and the addressing here will be incorrect.  

The &mapped_rw is bfff4000. So the incorrect addressing is 2de4f+bfff4008(rip) is close to c0021e5e, where is not preset in Page table. 
The correct addressing should be 2de4f+10041a(rip) is close to 12e270

For x86 platform,  the address of gcov stub here complied as absolute addressing, it won’t cause problem in this case.  
Below code snippet comes from zephyr.lst of qemu_x86 with coverage. 

00100898 <transplanted_function>:
#ifndef SKIP_EXECUTE_TESTS
static void transplanted_function(bool *executed)
{
  100898:       55                              push   %ebp
  100899:       89 e5                         mov    %esp,%ebp
  **10089b:       a1 18 b6 12 00          mov    0x12b618,%eax**
  1008a0:       8b 15 1c b6 12 00      mov    0x12b61c,%edx
  1008a6:       83 c0 01                     add    $0x1,%eax
  1008a9:       83 d2 00                     adc    $0x0,%edx
  1008ac:       a3 18 b6 12 00           mov    %eax,0x12b618
  1008b1:       89 15 1c b6 12 00      mov    %edx,0x12b61c
        *executed = true;
  1008b7:       8b 45 08                    mov    0x8(%ebp),%eax
  1008ba:       c6 00 01                     movb   $0x1,(%eax)
}
  1008bd:       90                              nop
  1008be:       5d                              pop    %ebp
  1008bf:       c3                               ret

It seems like if we want to execute code snippet which got by memory mapping, the assembly code which gcov stubbed should using absolute addressing mode.

**To Reproduce**
Steps to reproduce the behavior:
1. sanitycheck -T tests/kernel/mem_protect/mem_map/ -p qemu_x86_64 --coverage
or 
1. west build tests/kernel/mem_protect/mem_map/ -b qemu_x86_64 -DCONFIG_COVERAGE=y -DCONFIG_COVERAGE_GCOV=y
2. west build -t run

**Expected behavior**
Tests must pass without timeout errors or freezing

**Impact**
It makes the memory map test case of executing failed when code coverage enabled for x86_64 platform.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04.3 LTS
 - Toolchain Zephyr SDK zephyr-sdk-0.11.4 and Zephyr SDK zephyr-sdk-0.12.0-beta-2
 - Commit SHA 8f2ba929

**Additional context**

andrewboie, 2020-12-15:
Interesting bug. Looks like the current test design of memcpy()ing program text into another virtual page is unsound and needs a different approach.

If we can supply a testcase specific linker script, could place the test function into its own page, which could then be first remapped to not allow execution, show that faults, and then remap with execution and show that works. No virtual address would change, just the permissions.

nashif, 2021-01-27:
@dcpleung  can you take a look at this one?

dcpleung, 2021-01-27:
Could you re-try? I cannot reproduce the fault on my machine.

nashif, 2021-01-28:
@enjiamai  also can't reproduce here, please retry.

enjiamai, 2021-01-28:
Hi, @dcpleung, sorry for reply late,  here is reproducing step I tried today:

1. mainline commit 3c3504f7e582d3e4cd5adb07e4e495d4c1fbf5af
2. modified tests/kernel/mem_protect/mem_map/src/main.c,  comment the line 20, like that

/* Skip the memory map executing case when coverage enabled in x86_64,
 * because it will crash due to incorrect address accessing for gcov variables.
 * See issue#30434 for more details.
 */
#if defined(CONFIG_X86_64) && defined(CONFIG_COVERAGE)
//#define SKIP_EXECUTE_TESTS   <= comment here.
#endif

3. execute "twister -T tests/kernel/mem_protect/mem_map/ -p qemu_x86_64 --coverage"

I think you cannot reproduce it because I already skip this test case.
And @nashif, do you think we really need to fix it? I don't think it's our code problem.  It only happens on qemu_x86_64 with coverage enable.

PR #31760 (2021-02-02): tests: mem_map: fix execution test for x86_64 with coverage
When coverage is enabled on x86_64, GCC uses relative addressing
to increment the gcov counters. The generated code of the test
function assumes execution is in the same location where
the linker places the test function. This does not work with
the execution test as it copies the function into another part
of memory and tries to execute from there. When the copied
function starts to run, the instruction pointer is at the newly
copied function. So any relative addressing with regard to
the instruction pointer now is invalid. Instead o [... cut]
commit af37e9d9 (2021-02-02): tests: mem_map: fix execution test for x86_64 with coverage

When coverage is enabled on x86_64, GCC uses relative addressing
to increment the gcov counters. The generated code of the test
function assumes execution is in the same location where
the linker places the test function. This does not work with
the execution test as it copies the function into another part
of memory and tries to execute from there. When the copied
function starts to run, the instruction pointer is at the newly
copied function. So any relative addressing with regard to
the instruction pointer now is invalid. Instead  [... cut]
closed 2021-02-02 by nashif