zephyrproject-rtos/zephyr #30441: hci_uart uses wrong BT_BUF_ACL_SIZE on dual chip solutions + multicore

jostn, 2020-12-04:
**Describe the bug**
hci_raw.h has a dependency on CONFIG_BT_CTLR_TX_BUFFER_SIZE

This option can not be set for dual chip solutions for multicores platorms like nrf5340pdk_nrf5340_cpuapp, leading to a max hci data payload of 69 bytes.

**To Reproduce**
Steps to reproduce the behavior:
west build -p auto -b nrf5340pdk_nrf5340_cpuapp -dapp samples/bluetooth/hci_uart
west build -p auto -b nrf5340pdk_nrf5340_cpunet -dnet samples/bluetooth/hci_rpmsg

Use HCI_LE_Set_Data_Length to establish a connection and update the data length (251 octets, 2120us)
HCI data packets with data length > 69 bytes are now dropped because they do not fit in the buffer between the application core and the network core. 

**Expected behavior**
It should be possible to send the maximum LL payload between the cores (251 bytes)

**Impact**
Many bluetooth conformance tests are failing due to this.

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK

carlescufi, 2020-12-06:
@jostn would you mind sending a Pull Request yourself if @joerchan is too busy?

carlescufi, 2021-02-09:
@jostn and @joerchan could you please follow-up on this one?

PR #30552 (2021-03-12): Bluetooth: HCI: Add option CONFIG_BUF_ACL_DATA_LEN
Add the option CONFIG_BT_BUF_ACL_DATA_LEN which allows the user
to configure the max HCI ACL payload. This is needed for platorms
where the BLE LL, HCI uart and host run on three different cores.

Fixes: #30441
closed 2021-03-12 by jhedberg