zephyrproject-rtos/zephyr #30462: Improve how dependent PRs are managed

wal003, 2020-12-05:
**Is your feature request related to a problem? Please describe.**
Actually this is a suggestion to improve how dependent PRs are handled in this project. Example of a PR that depends on another one: https://github.com/zephyrproject-rtos/zephyr/pull/30424. If you do a quick search, you can easily find several other PRs like this.

**Describe the solution you'd like**
Introduce to this project [Dpulls](https://github.com/marketplace/dpulls), a Github app that make easier to manage dependent PRs. Basically:

- :star2: It prevents a PR to be merged before its dependencies.
- :eyes: It gives more visibility to the developers so they know why a PR cannot be merged.
- :heart: Avoid any accidental merge that can break your master branch.

In addition to that:
- Dpulls has support for cross-referencing (dependency from another project).
- Doesn't require any special config to add.
- Nothing to change in your current workflow. As you usually do, just write `depends on #pr` or `depends on owner/repo#pr` if from another project, and Dpulls will do the job :muscle:  

**Additional context**
Dpulls in action :rocket: :
[image]

I hope that you will like this suggestion :relaxed:

carlescufi, 2020-12-06:
Thanks for the suggestion @wal003. Does Dpulls also come in the form of a GitHub Action? We've been switching to those lately.

wal003, 2020-12-06:
Hey @carlescufi , thanks for your response. Dpulls doesn't come in the form of a Github Action, and I honestly don't think this will change in the future mainly because a Github App is suited better for what Dpulls is trying to achieve (better performance, accessibility, etc..).

erwango, 2020-12-08:
@wal003 are you caring to improve zephyrproject or wanting to sell your own project?
https://github.com/search?q=Introduce+to+this+project+Dpulls%2C&type=issues

wal003, 2020-12-08:
@erwango You're completely right, and your question is legitimate, so better be fully transparent.
In reality, I'm not really familiar with zephyr's project. I just noticed after a quick search that working with dependent PRs is a common thing in this project, so I wanted to propose this tool. Am I caring to improve this project ? Of course, like any other project I think I can provide some added value to it. I proposed this tool to other projects too. Some weren't interested, some really liked it and adopted it, and some were even asking where they can find the payment page even though I didn't ask for anything :smile: (Maybe this was the point that actually led me to keep working on and promoting Dpulls haha). So I just keep proposing in the hope to gain more users. Right now, I'm not making any money out of it, I'm just focusing on improving it and having new users actually helps a lot. And as long as I'm seeing that Dpulls is providing some benefits to some projects, I'm gonna keep proposing it (Btw, sorry I forgot to mention it but Dpulls is completely free for open source projects).
Maybe I should've mentioned this in the issue description but I didn't want to make it sound like I'm just here to sell something. Generally people tend to immediately close your issue if something sounds like an ad, which is completely understandable.
I hope my intention is more clear now :stuck_out_tongue: @erwango .

erwango, 2020-12-08:
@wal003 Thanks for this effort of transparency! :+1: