zephyrproject-rtos/zephyr #30465: Spurious interrupts not handled in ARMv7-R code with GICv2.

remy-luisant, 2020-12-06:
Having needed NEON, tens of megabytes of RAM and a very powerful CPU, I have a port of Zephyr to ARMv7-A. The port, quite predictably, shares a lot of code with the ARMv7-R version. I do believe my issue is relevant to your code. Please disregard this report should I be wrong in this respect.

Code in arch/arm/core/aarch32/isr_wrapper.S is as follows, edited for clarity:

```
bl arm_gic_get_active
push {r0, r1}
lsl r0, r0, #3	/* table is 8-byte wide */
cpsie i
ldr r1, =_sw_isr_table
add r1, r1, r0	/* table entry: ISRs must have their MSB set to stay
			 * in thumb mode */

ldm r1!,{r0,r3}	/* arg in r0, ISR in r3 */
blx r3		/* call ISR */
```

This is a problem because GICv2 can return 0x3ff as an indicator of a spurious interrupt. This is not handled, causing a read of memory which is not part of _sw_isr_table. Performing the blx instruction with the results of such a read is calamitous.

The fix is a simple comparison of the interrupt number against 0x3ff and exiting without calling any handlers.

Reference: ARM IHI 0048B.b, pages 1-20 and 3-50.

nashif, 2020-12-15:
@carlocaione  can you take a look?

carlocaione, 2020-12-16:
This is not the first time someone reported this, see #29058 for AArch64 so I think this is a real problem. I'll resurrect the PR for the AArch64, for the fix on AArch32 I think @ioannisg is the right person

carlescufi, 2021-02-09:
> This is not the first time someone reported this, see #29058 for AArch64 so I think this is a real problem. I'll resurrect the PR for the AArch64, for the fix on AArch32 I think @ioannisg is the right person

@carlocaione did you manage to resurrect the PR you mentioned?

carlocaione, 2021-02-09:
Yes, that was fixed with the merge in #29058 but only for AArch64

ioannisg, 2021-02-10:
@stephanosio  could you take a quick look?

stephanosio, 2021-02-10:
> @stephanosio could you take a quick look?

@ioannisg will do.

PR #34396 (2021-04-20): arch: arm: aarch32: Fix spurious interrupt handling
The GIC can return 0x3ff to indicate a spurious interrupt.  Other
interrupt controllers could return something different.  Check that the
pending interrupt is valid in order to avoid indexing past the end of
the isr_table.

This fixes #30465 and is based on the aarch64 fix in 9dd2731d.
commit 6734c6e8 (2021-04-20): arch: arm: aarch32: Fix spurious interrupt handling

The GIC can return 0x3ff to indicate a spurious interrupt.  Other
interrupt controllers could return something different.  Check that the
pending interrupt is valid in order to avoid indexing past the end of
the isr_table.

This fixes #30465 and is based on the aarch64 fix in 9dd2731d.
closed 2021-04-20 by nashif