zephyrproject-rtos/zephyr #30467: replace device define macros with devicetree-based macro

pabigot, 2020-12-06:
With the merge of #30196 we need to convert any device object definitions that are based on a devicetree node to use `DEVICE_DT_DEFINE` instead of:

* `DEVICE_DEFINE` 17 driver source files
* `DEVICE_AND_API_INIT` 361 driver source files

The vast majority of these are for devices that are associated with a devicetree node.  Changing allows use of new API to access the device object directly rather than a runtime lookup through the `label` property, and provides access to additional common devicetree information like dependencies.

Examples of the sort of changes required include:
* 63043bc5acc26f9a2203afaf58039d5dc86d11cd when instance macros based on iterating over devices are already used;
* cf339279338b0859ea1efc13a2806cb527fdc093 when interrupts require local direct access to the device before it is defined.

pabigot, 2020-12-08:
What this will enable is replacement of:

```
struct gpio_cfg {
  const char *label;
  gpio_pin_t pin;
  gpio_dt_flags_t flags;
};

static const struct gpio_cfg cfg = {
   .label = DT_GPIO_LABEL(NODE_ID, foo_gpios),
   .pin = DT_GPIO_PIN(NODE_ID, foo_gpios),
   .flags = DT_GPIO_FLAGS(NODE_ID, foo_gpios),
};
static const struct device *gpio;

...
gpio = device_get_binding(cfg->label);
if (gpio) {
  gpio_configure(gpio, cfg->pin, GPIO_OUTPUT_INACTIVE | cfg->flags);
}

```

with

```
struct gpio_cfg {
  const struct device *dev;
  gpio_pin_t pin;
  gpio_dt_flags_t flags;
};

// the need for this this may go away with tooling enhancements
DEVICE_DT_DECLARE(DT_GPIO_CTLR(NODE_ID, foo_gpios));

static const struct gpio_cfg cfg = {
   .dev = DEVICE_DT_GET(DT_GPIO_CTLR(NODE_ID, foo_gpios)),
   .pin = DT_GPIO_PIN(NODE_ID, foo_gpios),
   .flags = DT_GPIO_FLAGS(NODE_ID, foo_gpios),
};

...

if (device_is_ready(cfg->dev)) {
  gpio_configure(cfg->dev, cfg->pin, GPIO_OUTPUT_INACTIVE | cfg->flags);
}
```

pabigot, 2020-12-09:
Based on TSC discussion we'll just go ahead with this, in separate PRs grouped by reasonable categories (subsystem and/or SOC family).  If you're willing to help please assign yourself, and where possible identify the areas where you intend to work so we don't duplicate effort.  We do need a decision soon on #30400 so we know which style of macro to use: my current position is [here](https://github.com/zephyrproject-rtos/zephyr/pull/30400#pullrequestreview-548383275).

galak, 2020-12-09:
I've got a PR for all the I2C drivers coming

jfischer-no, 2020-12-09:
@pabigot I can help/take-over display controller drivers and USB relevant things.

galak, 2020-12-09:
Converted ADC (#30567) and SPI (#30568)

galak, 2020-12-09:
Converted PWM (#30570)

galak, 2020-12-10:
Converted watchdog (#30592)

galak, 2020-12-10:
Converted GPIO (#30597)

galak, 2020-12-10:
Converted Entropy (#30600)

pabigot, 2021-01-05:
I'm going to call this done, because enough conversions have been completed that #31108 passes CI.  There will be some stragglers, especially for PRs that are in flight.  When reviewing please request that any PRs that define new device instances for devicetree-defined devices switch to the DT-based definition before they're merged.

closed 2021-01-05 by pabigot