zephyrproject-rtos/zephyr #30483: Sanitycheck: When platform is nsim_hs_smp, process "west flash"  become defunct, the grandchild "cld" process can't be killed

wangnuannuan, 2020-12-07:
**Describe the bug**
When the platforms is `nsim_hs_smp`, the `BinaryHandler` call command `west flash xxx` to run testcase, and the process `west flash` start the child process `mdb mdb -multifiles=core0,core1 -run -cl`, and finally child process `mdb mdb -multifiles=core0,core1 -run -cl` create grandchild process `MetaWare/arc/bin32/cld`. 

In normal circumstances, the child process `mdb mdb -multifiles=core0,core1 -run -cl` and grandchild process `MetaWare/arc/bin32/cld` should terminate here.
https://github.com/zephyrproject-rtos/zephyr/blob/5cbf2ec1f7a927a57595c72c5cfbaec1d5bc56c0/scripts/sanity_chk/sanitylib.py#L497

But the process `west flash xxx` become defunct after grandchild process `MetaWare/arc/bin32/cld` started. So the the child process `mdb mdb -multifiles=core0,core1 -run -cl`  become orphan process, and if grandchild process `MetaWare/arc/bin32/cld` doesn't terminate,  the sanitycheck will  hang up forever.

carlescufi, 2021-01-26:
@evgeniy-paltsev do you have time to look at this for 2.5?

carlescufi, 2021-01-26:
Downgrading priority to low since it's specific to a single platform and might already be fixed.

evgeniy-paltsev, 2021-01-27:
@wangnuannuan as I understand this issue has already been fixed as we kill 'cld' process by pid.
Could you please confirm that?

Thanks.

wangnuannuan, 2021-01-28:
Yes, this issue has already been fixed.

closed 2021-01-28 by wangnuannuan