zephyrproject-rtos/zephyr #30486: updatehub demo for nrf52840dk

neo-rink106, 2020-12-07:
HI,
I am beginner on Nordic MCU and zephyr sdk. I am going to start a new project. Before starting my work on updatehub, I need some basic information about updatehub demo with open thread.

1- is it possible to do multibroadcasting? If yes, How many number of maximum device I can connect?
2 Using openthread, how much time(Total Time) it will take for new OTA image update?

nandojve, 2020-12-07:
Hi @neo-rink106 ,

> I am beginner on Nordic MCU and zephyr sdk. I am going to start a new project. Before starting my work on updatehub, I need some basic information about updatehub demo with open thread.

My suggestion is before try UpdateHub, make sure you have a Thread Networking with Zephyr RTOS running. UpdateHub uses sockets on top of network stack. We opened #30460 to add network management and overlays, it covers OT too. This will help you to test UpdateHub over OT. We try keep same configurations that OT have on the https://openthread.io/guides/border-router section. If you can reproduce that, using a Zephyr Joiner UpdateHub probably will work.

> 1- is it possible to do multibroadcasting? If yes, How many number of maximum device I can connect?

I believe this is related to OT instead UpdateHub. I think @rlubos can give you the best answer.

2 Using openthread, how much time(Total Time) it will take for new OTA image update?

I think there is no accurate answer here and it depends on many factors. I'll try this week a run to measure the time from a PC+NCP <-> Zephyr+Joiner+UpdateHub config and post here. You are not the first person that asked about it.

CC: @otavio

rlubos, 2020-12-08:
> 1- is it possible to do multibroadcasting? If yes, How many number of maximum device I can connect?

As for Thread in general, you can send multicast messages by using one of the multicast addresses (for instance `FF03::1` "all-nodes" address). Not sure though if updatehub can make use of it (from what I see it works in PULL mode, i. e. Block2 option is used).

As for device limits in Thread mesh, it's 32 routers with up to 511 children per router which gives a maximum theoretical value of 16352 devices in the network. In the default OT configuration the maximum number of children is limited though to save memory.

nandojve, 2020-12-08:
> As for Thread in general, you can send multicast messages by using one of the multicast addresses (for instance `FF03::1` "all-nodes" address). Not sure though if updatehub can make use of it (from [...]

The current UpdateHub pulls server for data and right now and there is no benefit using a multicast address. I have ideas about enhancements that can able update by multicast. The thing is, some node need continue to pull the server than broadcast the received data to others, This usually should be done on the border router but can be made by any node on the network.

nandojve, 2020-12-11:
> 2 Using openthread, how much time(Total Time) it will take for new OTA image update?

The below scenario uses OT 1.1 with NCP on a Linux PC as border router. It uses Tayga for NAT64 and OT example default configs. This is based on PR #30460. The UpdateHub-CE is a docker hosted at an US service provider at NY. I'm running at South America and ping is around 138 ms from OT node to US server.

Download: ~ 16m 30s
Swap: ~30s
Confirm: ~1 min

```
*** Booting Zephyr OS build zephyr-v2.4.0-2049-gdd5f0994f9d2  ***
[00:00:00.393,005] <inf> mcuboot: Starting bootloader
[00:00:00.399,475] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.409,942] <inf> mcuboot: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[00:00:00.420,501] <inf> mcuboot: Boot source: none
[00:00:00.426,208] <inf> mcuboot: Swap type: none
[00:00:00.878,601] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:00.886,535] <inf> mcuboot: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-2239-gacd86316efd9  ***
[00:00:00.004,272] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.016,815] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.016,815] <inf> fs_nvs: alloc wra: 0, f40
[00:00:00.016,815] <inf> fs_nvs: data wra: 0, 178
[00:00:00.030,334] <inf> main: UpdateHub sample app started
[00:00:00.030,334] <inf> main: Confirming the boot image
[00:00:00.030,395] <inf> main: Network connected
[00:00:00.030,395] <inf> main: Starting UpdateHub polling mode
[00:00:00.030,395] <inf> updatehub: SHA-256 verification on download and from flash
[00:00:02.031,982] <err> updatehub: Could not receive data
uart:~$ updatehub info 
Unique device id: d7eed6b80dd18d94
Firmware Version: 1.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: XXXXXXX
[00:00:16.942,413] <inf> updatehub: No update available
[00:01:02.200,653] <inf> updatehub: No update available
[00:02:02.393,066] <inf> updatehub: Probe metadata received
[00:18:31.403,961] <inf> updatehub: Firmware download complete
�*** Booting Zephyr OS build zephyr-v2.4.0-2049-gdd5f0994f9d2  ***u can reboot now
[00:00:00.255,889] <inf> mcuboot: Starting bootloader
[00:00:00.262,390] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x1
[00:00:00.272,827] <inf> mcuboot: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
[00:00:00.283,294] <inf> mcuboot: Boot source: none
[00:00:00.289,001] <inf> mcuboot: Swap type: test
[00:00:33.645,111] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:33.653,045] <inf> mcuboot: Jumping to the first image slot

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-2239-gacd86316efd9  ***
[00:00:00.004,272] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.016,662] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.016,662] <inf> fs_nvs: alloc wra: 0, c60
[00:00:00.016,662] <inf> fs_nvs: data wra: 0, 680
[00:00:00.045,623] <inf> main: UpdateHub sample app started
[00:00:00.045,623] <inf> main: Confirming the boot image
[00:00:00.045,745] <inf> main: Network connected
[00:00:00.045,745] <inf> main: Starting UpdateHub polling mode
[00:00:00.045,745] <inf> updatehub: SHA-256 verification on download and from flash
[00:00:02.047,363] <err> updatehub: Could not receive data
[00:01:02.214,233] <inf> updatehub: No update available
uart:~$ updatehub info 
Unique device id: d7eed6b80dd18d94
Firmware Version: 2.0.0
Product uid: e4d37cfe6ec48a2d069cc0bbb8b078677e9a0d8df3a027c4d8ea131130c4265f
UpdateHub Server: XXXXXX
uart:~$ net ping XXXXXX
PING XXXXXX
8 bytes from XXXXXX to fd11:22::6a58:2c2c:1425:7900: icmp_seq=0 ttl=48 rssi=0 time=139 ms
8 bytes from XXXXXX to fd11:22::6a58:2c2c:1425:7900: icmp_seq=1 ttl=48 rssi=0 time=136 ms
8 bytes from XXXXXX to fd11:22::6a58:2c2c:1425:7900: icmp_seq=2 ttl=48 rssi=0 time=138 ms
uart:~$ 
```

neo-rink106, 2020-12-14:
hello Sir, 
@nandojve ,
I find some one link of OTA code using thread but without Zephyr , that code update time is just 3 min 12 second , that was amazing but why Zephyr code takes too much time , I want to reduce time . What can i do . 
**Can you tell me particular reason behind it.** 

[image]

https://devzone.nordicsemi.com/nordic/short-range-guides/b/mesh-networks/posts/thread-tutorial-practical-guide-for-device-upgrade-ota 

thanks for your support

nandojve, 2020-12-14:
Hello @neo-rink106,

It is difficult to say the real motive behind this higher time. There are some things that need be in mind:

1) The currently OT support on UpdateHub is experimental. This means, it works as I show to you, but it is not tuned yet.
2) The OTA on the NordicZone uses a dedicated local solution and I showed to you a true internet FW upgrade from US to Brasil. The ping from internet is around 138ms and it can not be ignored because will delay every request.
3) UpdateHub was configured to use CONFIG_UPDATEHUB_COAP_BLOCK_SIZE_EXP=2 by default. This means, 64 byte CoAP data.

Let's do some math:
I remember that image size was around 380k bytes. Using CoAP block of 64 bytes we will need (380.000 / 64) ~6000 packages. If we add request delay it will need (6000 * 0.138) = 828s download [0:13:48]. The other thing that takes time is flash erase/write.
Since the application took [0:16:30] the calculation seems reasonable to give perspective. The variables that most influentiate are block size and ping, in this case.

Notes:

1) A CoAP block size bigger than 64 will fragment the 6LoWPAN package. By default, UpdateHub will not try fragment any 6LoWPAN packages. This can decrease overall performance but can be better in high density networks. This is an assumption and need tests!
2) The maximum CoAP block is 1024 bytes. This is necessary  to avoid IPv6 fragmentation. With this size you can have the best internet performance but you can struggle in the Border Router since the max air frame on the IEEE-802.15.4 is 128 bytes [this is not payload].

As you can see, there is no right or wrong instead there are a plenty of things to be considered for a real scenario.

CC: @nehalrohit

retfie, 2020-12-16:
Hello @neo-rink106,
here is an example of what time you could achieve with the following conditions:

CONFIG_LWM2M_COAP_BLOCK_SIZE=512
example: samples/net/lwm2m_client + OpenThread + https://github.com/retfie/zephyr/commit/3b9dd7281845c281106b2e85eca604f2665919a6 (zephyr.signed.bin is ~344k)
file server: Californium, cf-simplefile-server into local IPv6 network
Multiple updates are possible in parallel.

using /5/0/1 lwm2m resource manually from leshan UI or automated with [leshan.py](https://github.com/foundriesio/dm-lwm2m/blob/master/scripts/leshan.py): 

```
python leshan.py -c <client endpoint> -u coap://[<file server ipv6 addr>]:5683/data/zephyr.signed.bin -host http://<leshan server>:8080 -d "OMA-LWM2M Sample Client"
2020-12-16 14:33:15,669 [INFO]  ready for firmware update
2020-12-16 14:33:15,842 [INFO]  requested firmware download from coap://[<file server ipv6 addr>]:5683/data/zephyr.signed.bin
2020-12-16 14:33:31,022 [INFO]  downloading firmware
2020-12-16 14:33:36,099 [INFO]  downloading firmware
...

2020-12-16 14:35:23,733 [INFO]  downloading firmware
2020-12-16 14:35:28,887 [INFO]  downloading firmware
2020-12-16 14:35:33,959 [INFO]  ready to apply update
2020-12-16 14:35:34,849 [INFO]  requested firmware update execution
2020-12-16 14:35:50,025 [ERROR]  <Response [504]>
2020-12-16 14:36:05,092 [ERROR]  <Response [504]>
2020-12-16 14:36:17,684 [ERROR]  <Response [500]>
2020-12-16 14:36:27,828 [INFO]  firmware update successful
2020-12-16 14:36:28,813 [INFO] (MainThread) UPDATE SUMMARY:
2020-12-16 14:36:28,813 [INFO] (MainThread)  update SUCCESS (191 seconds)
2020-12-16 14:36:28,813 [INFO] (MainThread) 1 update(s) attempted which took 193 seconds total
```

nandojve, 2020-12-16:
Hello @retfie,

> file server: Californium, cf-simplefile-server into local IPv6 network

It will be nice have data from a real deploy using internet from one continent to another with ping info on Zephyr console, for the records. Please, share better your scenario, ping? hardware? Etc? Are you using native IPv6 or NAT 6to4?

Why ping is so important: From 138ms to 20ms (I'm assuming 20ms local network, it can be even better).
(6000 * 0.138) = 828s
(6000 * 0.02) = 120s

> Multiple updates are possible in parallel.

I don't follow here, what is the difference between UpdateHub and LWM2M here? Could you elaborate?

It will be nice compare apples with apples otherwise is pointless.

retfie, 2020-12-16:
Dear @nandojve,
nothing to sell here, just an alternative working solution for FOTA into Zephyr.
OpenThread + LWM2M on nrf52840 based HW, native IPv6, local, no Clouds.
My apologies for posting into wrong topic.

nandojve, 2020-12-16:
Hi @retfie,

It is ok and that is the beauty of Zephyr, it gives us many choices!
👍

nandojve, 2021-02-10:
Since there is no activity for almost 2 months, I'm closing this topic.
Feel free to reopen if necessary.

closed 2021-02-10 by nandojve