zephyrproject-rtos/zephyr #30493: Allow a custom location for the final merged configuration file

valeros, 2020-12-07:
Hi,

**Is your enhancement proposal related to a problem?**
At the moment Zephyr integration with PlatformIO ecosystem only supports project configuration using `*.conf` files. Users (see https://skybluetrades.net/projects/nrf52840-tools-comparison/platformio-zephyr) would like have support for an interactive Kconfig interface, for example the `menuconfig` target.

The main issue here is that the build directory within the usual PlatformIO project workflow is much more dynamic and can be cleaned once a user changes the PlatformIO project configuration. Because of that, we also lose the `zephyr/.config` file with the configurations made via `menuconfig`.

**Describe the solution you'd like**
Ideally, the path to the final merged configuration file [DOTCONFIG](https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/kconfig.cmake#L47) could be overridden directly from the CMake command line, for example:

```bash
cmake -GNinja -DBOARD=<board> -DDOTCONFIG=/custom/path/to/.config
```

I'm no expert on the Zephyr build system and maybe there is another way to specify a custom location for that file. Is there any chance that converting the `DOTCONFIG` to a cache variable will be enough? 

```cmake
set(DOTCONFIG {PROJECT_BINARY_DIR}/.config CACHE STRING "Location for the merged configuration file")
```

This will greatly improve the current integration of the Zephyr framework. Instead of modifying `*.conf` files manually, users will be able to configure their project using Zephyr's Kconfig tools. 

Thanks in advance.

cc @ian-ross @nashif @ivankravets

tejlmand, 2020-12-08:
@valeros could you please explain why the existing possibility of saving the configuration file to a custom location is not sufficient ?

[image]

This custom conf file can then be used later with `cmake -DCONF_FILE=<path>/my_hello_world_debug.conf ...`

valeros, 2020-12-08:
Hi @tejlmand , thanks for answering. Please correct me if I'm wrong, but doesn't that approach conflicts with the usual configuration workflow when the Zephyr build system picks up `prj.conf` if it exists in the project by default? Also, there is a solid chance that a user may specify a custom `*.conf` file using the same `CONF_FILE` variable directly in their `CMakeLists.txt` (this method even mentioned in the Zephyr docs).

Our goal here is to keep the usual Zephyr workflow as much as possible just. We can provide users with more seamless integration instead of introducing additional instructions on how to properly use `menuconfig` configurations with PlatformIO when they stumble upon a non-obvious discrepancy between configuration methods.

Thanks!

tejlmand, 2020-12-10:
@valeros Thanks for the update.

> Please correct me if I'm wrong, but doesn't that approach conflicts with the usual configuration workflow when the Zephyr build system picks up prj.conf if it exists in the project by default? 

I was under the impression you wanted to avoid modifying the original `prj.conf`, hence the reason for using a custom name.
That way PlatformIO would not need to modify the original `prj.conf`

The `<build-folder>/.config` is only for testing local configuration changes, as described:
https://docs.zephyrproject.org/latest/guides/kconfig/setting.html
> The menuconfig and guiconfig interfaces can be used to test out configurations during application development. This page explains how to make settings permanent.

So the `.config` itself is to be considered a build folder file, similar to any other generated files, and users are expected to update `prj.conf` or. So storing this file at another location is not a good solution.

It also contains all settings, and not just the `prj.conf`, such settings can be soc/board specific values, and other default settings.

> Our goal here is to keep the usual Zephyr workflow as much as possible.

Using a custom config file instead of `prj.conf` is a perfect valid solution. If you read the above link, it also contains:
> The application configuration can come from the sources below. By default, prj.conf is used.
>
>  1.  If CONF_FILE is set, the configuration file(s) specified in it are merged and used as the application configuration. CONF_FILE can be set in various ways:
>      1. In CMakeLists.txt, before calling find_package(Zephyr)
>      2. By passing -DCONF_FILE=<conf file(s)>, either directly or via west

But if I understand you correctly, it is preferred to keep the `prj.conf` untouched, but have `prj.conf` a the default config file entry, just as today.

I think using a Kconfig fragment with minimal config changes is what you need.
That could be handled with PlatformIO creating an empty `<path>/platformio.conf` and then executing CMake as `cmake -DOVERLAY_CONFIG=<path>/platformio.conf`.
See also:
https://docs.zephyrproject.org/latest/application/index.html?highlight=overlay_config#important-build-system-variables
> **OVERLAY_CONFIG**: Additional Kconfig configuration fragment files. Multiple filenames can be separated with either spaces or semicolons. This can be useful in order to leave CONF_FILE at its default value, but “mix in” some additional configuration options.

When users make changes in `menuconfig`, then those changes can be stored as a minimal config with the `platfornio.conf`.

Of course this requires users to know the config name to use, but might be a better approach than requesting the `.config` to change location.

If this sounds reasonable, then it could probably be better to see if build system could accept default minimal config name to use.

valeros, 2020-12-11:
Indeed, at first glance, it seems like a sane approach, but still it forces users to take all these non-obvious steps with relative paths and a proper file name every single time when they run the `menuconfig` target instead of just saving their configurations when prompted at exit. Besides, that approach can lead to obscure issues when users already use  `OVERLAY_CONFIG`  in their `CMakeLists.txt`. 

> So storing the .config file at another location is not a good solution.

The default location will still be same, we're merely asking just to add a way to override that location when it's needed.

Thanks for your help.

cafe88, 2021-08-30:
Hi! Since we have also encountered this problem in our build system, we would suggest that it should be possible to pass `menuconfig` a parameter that overrides the default `_minconf_filename`.
This would allow a user to at least only have to remember to save the minimal configuration, but not have to remember file paths.
Would this be an approach that could also help with the PIO problem?
Thanks!

tejlmand, 2021-08-31:
> Hi! Since we have also encountered this problem in our build system, we would suggest that it should be possible to pass `menuconfig` a parameter that overrides the default `_minconf_filename`.
> This [...]

Not sure we can do it as argument to menuconfig itself, but we could maybe make an extra configuration option that would allow to specify a default `_minconf_filename`, and still keep the default conf file to `.config`.

cafe88, 2021-09-01:
I think that should do it for us, thanks for your fast reply!

mmahadevan108, 2022-09-12:
@valeros @tejlmand . I am clearing the milestone field as we are past the feature freeze date for release 3.2. Kindly add one for a future release.