zephyrproject-rtos/zephyr #30507: nrf52_bsim fails on some tests after merging 29810

nashif, 2020-12-07:
**Describe the bug**

```west build -b nrf52_bsim  tests/benchmarks/latency_measure -t run```
hangs, before #29810 it used to run all the way to the end.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52_bsim  tests/benchmarks/latency_measure -t run

**Expected behavior**

test completes
**Impact**
Blocking CI in many pending PRs

**Logs and console output**
hangs here:

```
?_??: WARNING: Neither simulation id or the device number have been set. I assume you want to run without a BabbleSim phy (-nosim)
?_??: WARNING: If this is not what you wanted, check with --help how to set them
d_00: @00:00:00.000000  *** Booting Zephyr OS build zephyr-v2.4.0-2121-gde191bf3a6fb  ***
d_00: @00:00:00.000000  START - Time Measurement
d_00: @00:00:00.000000  Timing results: Clock frequency: 0 MHz
```

nordic-krch, 2020-12-08:
@aescolar i think i would need a help from you here. It seems like RTC interrupt is triggered only once. I've logged `COUNTER`, `CC[0]` and `EVTEN` registers after setting new CC and it looks ok.

At 0 CC is set for 1sec, at one sec RTC interrupt occurs and it is set to 2sec timeout but that one never comes.

```
d_00: @00:00:00.000000  COUNTER:0 CC:32768
d_00: @00:00:00.000000  EVTEN:10000
d_00: @00:00:00.000000  *** Booting Zephyr OS build zephyr-v2.4.0-2096-gad3de0a76cec  ***
d_00: @00:00:00.000000  START - Time Measurement
d_00: @00:00:00.000000  Timing results: Clock frequency: 0 MHz
d_00: @00:00:01.000000  COUNTER:32768 CC:65536
d_00: @00:00:01.000000  EVTEN:10000
```

aescolar, 2020-12-08:
>@aescolar i think i would need a help from you here. 

@nordic-krch (Today and tomorrow I'm swamped.) The default when a PR breaks master is to revert it and figure out calmly why and fix it without blocking others. If you cannot find easily what went wrong I would not delay further reverting it.

PR #30522 (2020-12-08): drivers: timer: nrf_rtc_timer: Fix int_mask initialization
Variable was not properly initialized when custom channels were
disabled.

Fixes #30507.
closed 2020-12-08 by nashif