zephyrproject-rtos/zephyr #30517: Interrupt nesting is broken on ARMv7-R / LR_svc corrupted.

remy-luisant, 2020-12-08:
I have a custom board running a port to ARMv7-A with NEON. This shares much code with the ARMv7-R version and I believe this issue is relevant to your code as well. Please disregard this report should this not be the case.

zephyr/arch/arm/core/aarch32/isr_wrapper.S, edited for clarity, comments starting with // are mine:

```
        //Entry in MODE_IRQ

	sub lr, lr, #4
	srsdb #MODE_SYS!          //LR_irq saved to the process stack.
	cps #MODE_SYS             //System mode
	push {r0-r3, r12, lr}     //User registers saved to the process stack.

	cps #MODE_SVC             //SVC mode 

	/* Align stack at double-word boundary */
	and r3, sp, #4
	sub sp, sp, r3
	push {r2, r3}

	/* Increment interrupt nesting count */
	ldr r2, =_kernel
	ldr r0, [r2, #_kernel_offset_to_nested]
	add r0, r0, #1
	str r0, [r2, #_kernel_offset_to_nested]

	/* Get active IRQ number from the interrupt controller */
        // !!!! WRITING LR_svc HERE !!!!
	bl arm_gic_get_active  
	push {r0, r1}
	lsl r0, r0, #3	/* table is 8-byte wide */

	/*
	 * Enable interrupts to allow nesting.
        */
	cpsie i

        //Interrupt happens here, starting this code all over again and corrupting LR_svc.
```

At no point is LR_svc preserved. The fix is to simply store LR_svc before it gets clobbered at the first bl instruction and subsequently restore it before exiting the handler. I have chosen to do this onto the IRQ stack, which appeared to be otherwise unused, to preserve my work on getting OpenOCD to behave on the Cortex-A7 that I work with. It is likely that there is a much better solution as to where to put this register, but I currently lack the resources to investigate that and my fix worked well enough for my needs.

julien-massot, 2020-12-08:
Hi Remy, I'm indeed facing an issue with nested interrupt on Cortex-R,
but didn't figured out what is the reason.
Thanks for the report.

As a temporary workaround, I just disabled nested interrupt, as it's done in Cortex M.
https://github.com/iotbzh/zephyr/commit/a0e4c0ab6f2c009570d16529972e50faf36c3ffc#

ioannisg, 2021-01-21:
@stephanosio any chance you could take a look?

stephanosio, 2021-01-22:
@ioannisg will do.

stephanosio, 2021-01-22:
With the current implementation, if an ISR is interrupted while executing in a branch, the `lr_svc` register will indeed be corrupted and, the branch of the interrupted ISR will exit to the return address of the final branch of the interrupting ISR, which may or may not coincide with the return address of the branch of the interrupted ISR (which is why the interrupt nesting test seems to work fine in the QEMU).

I will create a PR to fix this in a few days.

PR #31519 (2021-01-26): Fix AArch32 interrupt nesting and exception exiting bugs
This series fixes some critical bugs in the nested interrupt handling and fatal exception handling code.

Fixes #31511 
Fixes #30517
closed 2021-01-26 by nashif