zephyrproject-rtos/zephyr #30526: tests: drivers: timer: Tests from drivers.timer.nrf_rtc_timer.basic fail on nrf52 platforms

PerMac, 2020-12-08:
**Describe the bug**
Tests from drivers.timer.nrf_rtc_timer.basic (zephyr/tests/drivers/timer/nrf_rtc_timer) fail on nrf52 platforms. 

**To Reproduce**
Steps to reproduce the behavior:
1. go to zephyr dir
2. have nrf52 platform connected (e.g. nrf52dk_nrf52832)
3. run `./scripts/sanitycheck --device-testing -T tests/drivers/timer/nrf_rtc_timer/ -p nrf52dk_nrf52832 --device-serial /dev/ttyACM0 -v -v`
4. See error

**Expected behavior**
The tests pass.

**Impact**
Not clear

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.4.0-2107-gd1c2a5261781  ***
DEBUG   - DEVICE: Running test suite test_nrf_rtc_timer
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_basic
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at CMAKE_SOURCE_DIR/src/main.c:84: test_timeout: (test_data.err not equal to 0)
DEBUG   - DEVICE: Unexpected err: -22
DEBUG   - DEVICE: FAIL - test_basic
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_z_nrf_rtc_timer_compare_evt_address_get
DEBUG   - DEVICE: PASS - test_z_nrf_rtc_timer_compare_evt_address_get
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_int_disable_enabled
DEBUG   - DEVICE: PASS - test_int_disable_enabled
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_get_ticks
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at CMAKE_SOURCE_DIR/src/main.c:172: test_get_ticks: ((ticks >= exp_ticks - 1) && (ticks <= exp_ticks) is false)
DEBUG   - DEVICE: Unexpected result -22 (expected: 18260)
DEBUG   - DEVICE: FAIL - test_get_ticks
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_absolute_scheduling
DEBUG   - DEVICE: RTC event scheduled at 585083us for 590761us,event occured at 2073157196us (uptime)
DEBUG   - DEVICE: RTC event scheduled now, at 598297us,event occured at 2073183624us (uptime)
DEBUG   - DEVICE: PASS - test_absolute_scheduling
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_alloc_free
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at CMAKE_SOURCE_DIR/src/main.c:248: test_alloc_free: (chan[i] >= 0 is false)
DEBUG   - DEVICE: Failed to allocate RTC channel.
DEBUG   - DEVICE: FAIL - test_alloc_free
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_stress
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at CMAKE_SOURCE_DIR/src/main.c:266: test_stress: (chan >= 0 is false)
DEBUG   - DEVICE: Failed to allocate RTC channel.
DEBUG   - DEVICE: FAIL - test_stress
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: Test suite test_nrf_rtc_timer failed.
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: PROJECT EXECUTION FAILED
DEBUG   - run status: nrf52dk_nrf52832/tests/drivers/timer/nrf_rtc_timer/drivers.timer.nrf_rtc_timer failed
INFO    - 1/1 nrf52dk_nrf52832          tests/drivers/timer/nrf_rtc_timer/drivers.timer.nrf_rtc_timer FAILED Failed (device 3.997s)
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: zephyr-v2.4.0-2107-gd1c2a5261781

PerMac, 2020-12-08:
It seems the fail of kernel.common.stack_sentinel.fatal is caused by the issues with the above driver, so I won't create a new issue for that:
```
DEBUG   - Waiting for device nrf52dk_nrf52832 to become available
DEBUG   - DEVICE: ASSERTION FAIL [chan && chan < (0 + 1)] @ /home/maciej/zephyrproject2/zephyr/drivers/timer/nrf_rtc_timer.c:214
DEBUG   - DEVICE: E: r0/a1:  0x00000004  r1/a2:  0x000000d6  r2/a3:  0x00000000
DEBUG   - DEVICE: E: r3/a4:  0x00006d61 r12/ip:  0x00000000 r14/lr:  0x00001e17
DEBUG   - DEVICE: E:  xpsr:  0x41000021
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x0000632c
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
DEBUG   - DEVICE: E: Fault during interrupt handling
DEBUG   - DEVICE: 
DEBUG   - DEVICE: E: Current thread: 0x200002c8 (main)
DEBUG   - DEVICE: Caught system error -- reason 4
DEBUG   - DEVICE: Was not expecting a crash
DEBUG   - Waiting for device nrf52dk_nrf52832 to become available
DEBUG   - Waiting for device nrf52dk_nrf52832 to become available
...
DEBUG   - Timed out while monitoring serial output on nrf52dk_nrf52832
DEBUG   - run status: nrf52dk_nrf52832/tests/kernel/fatal/exception/kernel.common.stack_sentinel timeout
INFO    - 3/5 nrf52dk_nrf52832          tests/kernel/fatal/exception/kernel.common.stack_sentinel FAILED Timeout (device 63.449s)

```

ioannisg, 2020-12-09:
This is observed in several Nordic platforms, so I am raising priority to high.

PR #30487 (2020-12-14): drivers: timer: nrf_rtc_timer: Add clearing of previous CC event
Added clearing of CC event which may occure due to previous
CC value which was closed to current counter value.

Fixes #30526.
closed 2020-12-14 by nashif