zephyrproject-rtos/zephyr #30540: CONFIG_TRACING not working after updating from v2.1.0 to 2.2.0

vandita5, 2020-12-09:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Tried different versions v2.3.0 and v2.2.0 and v2.1.0. tracing logs are only printed for v2.1.0. I am using net/socket/echo_server sample with the following config values added to prj.cnf file
CONFIG_TRACING=y
CONFIG_TRACING_CPU_STATS=y
CONFIG_TRACING_CPU_STATS_LOG=y

(I am using segger rtt to see logs)

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject/zephyr; git checkout zephyr-v2.1.0; west update
2. add above mentioned config values to prj.cnf for the mentioned sample
3. west build -p auto -b nucleo_f429zi .\samples\net\sockets\echo_server\
4. west flash; see cpu usage logs
5. repeat steps 1 to 3 with zephyr-v2.2.0 
6. no cpu usage logs

**Expected behavior**
A clear and concise description of what you expected to happen.
I expected to see cpu usage logs via tracing

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
showstopper

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) windows
 - Toolchain (e.g Zephyr SDK, ...) zephyr sdk
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

nashif, 2020-12-15:
Does this work for you in master?

koffes, 2020-12-15:
Does the system/thread hang after the CPU log print should have happened?
May be related to https://github.com/zephyrproject-rtos/zephyr/issues/30418

vandita5, 2020-12-15:
@koffes  No. the app keeps running normal. Just no cpu usage prints. 
@nashif I have not tried master but I tried zephyr-v2.1.0, zephyr-v2.2.0 and zephyr-v2.3.0 It broke v2.2.0 and beyond. I'll try to find some time this week to test with master as well.

MaureenHelm, 2021-02-03:
Is this still an issue? I don't have a `nucleo_f429zi` board, but I tried to reproduce with `mimxrt1064_evk` and it seems to work on v2.5.0-rc1

```
west build -b mimxrt1064_evk samples/net/sockets/echo_server/ -- -DOVERLAY_CONFIG=tracing_rtt.overlay
```

where the contents of `tracing_rtt.overlay` are:
```
CONFIG_TRACING=y                                                                                    
CONFIG_TRACING_CPU_STATS=y
CONFIG_TRACING_CPU_STATS_LOG=y
CONFIG_USE_SEGGER_RTT=y

CONFIG_UART_CONSOLE=n
CONFIG_RTT_CONSOLE=y
```

Then run `west debug` and in a second terminal:

```
$ telnet localhost 19021
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
SEGGER J-Link V6.62d - Real time terminal output
SEGGER J-Link ARM V10.1, SN=600111924
Process: JLinkGDBServerCLExe
*** Booting Zephyr OS build v2.5.0-rc1  ***
[00:00:00.053,000] <inf> net_config: Initializing network
[00:00:00.053,000] <inf> net_config: Waiting interface 1 (0x8000256c) to be up...
[00:00:03.012,000] <inf> net_config: Interface 1 (0x8000256c) coming up
[00:00:03.012,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
CPU usage: 1
[00:00:03.012,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:03.113,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:03.113,000] <inf> net_echo_server_sample: Run echo server
[00:00:03.113,000] <inf> net_echo_server_sample: Network connected
[00:00:03.113,000] <inf> net_echo_server_sample: Starting...
[00:00:03.167,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6)...
[00:00:03.273,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
[00:00:03.327,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:03.380,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv4)...
CPU usage: 14
CPU usage: 0
CPU usage: 0
CPU usage: 0
CPU usage: 0
```

closed 2021-04-26 by github-actions[bot]