zephyrproject-rtos/zephyr #30553: kconfig.py exits with error when using multiple shields

arludwig, 2020-12-09:
**Describe the bug**
When I try to set two shields
```
list(APPEND SHIELD "shield1")
list(APPEND SHIELD "shield2")
```
`kconfig.py` exits with return code 1 (System can't find file).

When I use a single shield instead
```
list(APPEND SHIELD "shield1")
```
everything works as expected.

I dug a bit through the cmake code, in particular `kconfig.cmake`, and ended up comparing the `execute_process` calls in line 241. I found, that they differed in their `COMMON_KCONFIG_ENV_SETTINGS`, in particular the `SHIELD_AS_LIST` part.

Single shield:
```
SHIELD_AS_LIST=shield1;
```

Two shields:
```
SHIELD_AS_LIST=shield1;shield2;
```

From what I can tell, those are variable definitions for the shell, which are separated by semicolons. My guess is, that the `shield2;` part is responsible for the "File not found" error.

After commenting out line 75 of `kconfig.cmake`
```
  #SHIELD_AS_LIST=${SHIELD_AS_LIST}
```
so far everything seems to work with two shields as well. I don't know whether the setting is needed, as both my custom shields currently are device tree only.

**To Reproduce**
Steps to reproduce the behavior:
1. Create (or find) two device tree only shields, and choose a compatible board
2. CMake configure a zephyr based project using both shields
3. Unless the issue is specific to my system, you should see an error.

**Expected behavior**
CMake configuring should work with zero, one or more shields, as described in the zephyr documentation.

**Impact**
As I need to use multiple shields in my project, this issue forces me to run a personally modified zephyr source tree (which I try to avoid whenever possible), until I can pull the fix from the zephyr repository. Also I don't know (yet) whether my workaround (hack) has any unintended side effects.

**Environment (please complete the following information):**
 - Windows 10
 - CMake 3.17.2
 - Zephyr 2.4 (c2a0b0f)
 - GNU ARM Embedded Toolchain (9 2020-q2-update)

**Additional context**
May be related to #25716.

erwango, 2020-12-10:
@tejlmand would you be able to have a look ?

henrikbrixandersen, 2021-01-18:
This is broken in Zephyr v2.4.0 as well. It works as expected in Zephyr v2.3.0.

henrikbrixandersen, 2021-01-18:
Raising the priority since this is a regression compared to v2.3.0.

galak, 2021-01-26:
@henrikbrixandersen does your PR address this issue?

henrikbrixandersen, 2021-01-26:
> @henrikbrixandersen does your PR address this issue?

@galak No, it does not address this issue. I only stumbled across it while trying to implement another feature using multiple shields.

PR #31635 (2021-01-26): cmake: shields: escaping and quoting list when used for env settings
Fixes: #30553

The `;` in a list must be escaped when used as for execute_process and
add_custom_target, as those functions are using `cmake -E env` to setup
correct environment before executing kconfig and menuconfig.

This commits escapes `SHIELD_AS_LIST` variable list correctly before
using it.
commit d24ca1be (2021-01-26): cmake: shields: escaping and quoting list when used for env settings

Fixes: #30553

The `;` in a list must be escaped when used as for execute_process and
add_custom_target, as those functions are using `cmake -E env` to setup
correct environment before executing kconfig and menuconfig.

This commits escapes `SHIELD_AS_LIST` variable list correctly before
using it.
closed 2021-01-26 by nashif