zephyrproject-rtos/zephyr #30554: tests/kernel/fatal/exception/sentinel test is failing for various nrf platforms

ioannisg, 2020-12-09:
**Describe the bug**
tests/kernel/fatal/exception/ stack_sentinel test variant is failing for various nRF platforms
tested on nRF5340 and nRF9160.

**To Reproduce**

Build flash and run stack sentinel test variant of tests/kernel/fatal/exception

**Expected behavior**
Test should not assert.

**Impact**
kernel test failing

**Logs and console output**

ioannisg, 2020-12-09:
@nordic-krch I suspect a driver issue here, but might be wrong.

ioannisg, 2020-12-09:
SImilar as reported in #30526

jcyrax, 2021-01-10:
This is a bug due to adjusting stack start and size for threads using floating point after sentinel is written to stack. This results in sentinel at wrong location and so it fails on first test.

You can see that in kernel/thread.c `setup_thread_stack()` sentinel is written to address `stack_buf_start = Z_THREAD_STACK_BUFFER(stack);` and right after that in `arch_new_thread()` `arch/arm/core/aarch32/thread.c` stack start is adjusted due to larger guard area used with floating points, but sentinel is never written to this location. 
```
#if FP_GUARD_EXTRA_SIZE > 0
	if ((thread->base.user_options & K_FP_REGS) != 0) {
		/* Larger guard needed due to lazy stacking of FP regs may
		 * overshoot the guard area without writing anything. We
		 * carve it out of the stack buffer as-needed instead of
		 * unconditionally reserving it.
		 */
		thread->stack_info.start += FP_GUARD_EXTRA_SIZE;
		thread->stack_info.size -= FP_GUARD_EXTRA_SIZE;
	}
#endif /* FP_GUARD_EXTRA_SIZE */
```

ioannisg, 2021-01-29:
Cannot reproduce this any more. Closing this.

ioannisg, 2021-01-29:
> This is a bug due to adjusting stack start and size for threads using floating point after sentinel is written to stack. This results in sentinel at wrong location and so it fails on first test. [...]

Sentinel should not be used together with MPU stack guards, so there's no stack adjustment of stack_info.start that applies here, as far as I understand.

closed 2021-01-29 by ioannisg

PerMac, 2021-01-29:
@ioannisg I haven't seen it for a while. I think it was the same root cause as in  #30526 and was fixed by #30487.