zephyrproject-rtos/zephyr #30564: RFC:  GNSS API

sslupsky, 2020-12-09:
## Introduction

Has there been any work on developing a GNSS API for Zephyr?

### Problem description

GNSS is a widely used technology for global positioning.  Zephyr should consider support of an API to interface with various GNSS chips.

### Proposed change

This proposed API was developed to work the u-blox GNSS solutions.  It does not implement advanced GNSS capabilities at the current time.  I am not familiar with the recent GPS solution offered by Nordic so it would be helpful to know what Nordic's thinking is regarding an API and how to make this better.

```
/**
 * @file gnss.h
 * @author Steven Slupsky (sslupsky@scanimetrics.com)
 * @brief 
 * @version 0.1
 * @date 2020-05-20
 * 
 * @copyright Copyright (c) 2020
 * 
 *    _____                 _                _        _          
 *   / ____|               (_)              | |      (_)         
 *  | (___   ___ __ _ _ __  _ _ __ ___   ___| |_ _ __ _  ___ ___ 
 *   \___ \ / __/ _` | '_ \| | '_ ` _ \ / _ \ __| '__| |/ __/ __|
 *   ____) | (_| (_| | | | | | | | | | |  __/ |_| |  | | (__\__ \
 *  |_____/ \___\__,_|_| |_|_|_| |_| |_|\___|\__|_|  |_|\___|___/
 *                                                               
 */

#ifndef ZEPHYR_INCLUDE_DRIVERS_GNSS_H_
#define ZEPHYR_INCLUDE_DRIVERS_GNSS_H_

#include <sys/__assert.h>
#include <sys/slist.h>

#include <zephyr/types.h>
#include <stddef.h>
#include <device.h>
// #include <dt-bindings/gpio/gpio.h>

#ifdef __cplusplus
extern "C" {
#endif

struct gnss_time {
	u32_t timeOfWeek;		 // ms
	u16_t gpsYear;
	u8_t gpsMonth;
	u8_t gpsDay;
	u8_t gpsHour;
	u8_t gpsMinute;
	u8_t gpsSecond;
	struct {
		u8_t validDate : 1;
		u8_t validTime : 1;
		u8_t fullyResolved : 1;
		u8_t validMag : 1;
	} valid;
	u32_t accuracy;
	s32_t gpsNanosecond;
} __packed;

struct gnss_position {
	u8_t fixType;		 //Tells us when we have a solution aka lock
	struct {
		u8_t gnssFixOK : 1;	// valid fix (ie: within DOP nad accuracy masks)
		u8_t diffSoln : 1;	// differential corrections were applied
		u8_t psmState : 4;	
[... 544 lines cut ...]
 *                      about USB device controller state.
 *
 * @return 0 if successful, negative errno code if failure.
 */
static inline int gnss_enable(struct device *dev,
			      gnss_status_callback_t status_cb)
{
	const struct gnss_driver_api *api =
		(const struct gnss_driver_api *)dev->driver_api;

	if (api->enable == NULL) {
		return -ENOTSUP;
	}

	return api->enable(dev, status_cb);
}

#ifdef __cplusplus
}
#endif

#include <syscalls/gnss.h>

#endif /* ZEPHYR_INCLUDE_DRIVERS_GNSS_H_ */
```

jukkar, 2020-12-09:
I just wonder would it make sense to look how gpsd (https://gitlab.com/gpsd/gpsd) is interfacing to different GNSS providers. It has this gps_type_t API https://gitlab.com/gpsd/gpsd/-/blob/master/include/gpsd.h#L423 for that. See https://gitlab.com/gpsd/gpsd/-/tree/master/drivers for different drivers.
I do not mean we would need to do things the same way as gpsd, but we need to make sure that our APIs are able to support all different GNSS providers.

sslupsky, 2020-12-10:
I will have a look at that.  I had done some prior work with the Sparkfun library which was my starting point.

sslupsky, 2020-12-11:
@jukkar Have you used gpsd?  I had a look and it looks quite good.  

A couple observations:

1. I am unsure about the weight of this.  Does it require the network stack to be running?  I think I saw that it uses sockets so I think networking is a requirement which increases the memory requirements significantly.
2. The intro It says you can connect via serial or usb.  Do you know if there are there other input sources?  U-blox devices have an i2c interface as well as serial and usb.  I am not sure which interfaces are available for the gps functions of the nRF9160.  I think I recall that it is an AT style interface so presumably serial?

jukkar, 2020-12-14:
I have used gpsd (but over 10 year ago). I would not want to use it in zephyr as is, but perhaps we can look how it interfaces to different chips as it has lot of supported GNSS providers. Your API looks reasonable but you mentioned it was developed according to u-blox GNSS. So would that API work with other GNSS providers?

sslupsky, 2020-12-14:
@jukkar I agree, I think there are some good things to learn from the studying the gpsd library.  I think the API above would be fine across other GNSS providers.  We need to add additional protocols as the API only contemplates four at the moment:

```
	GNSS_SENTENCE_STATE_UBX,
	GNSS_SENTENCE_STATE_NMEA,
	GNSS_SENTENCE_STATE_RTCM,
	GNSS_SENTENCE_STATE_RTCM3,
```

We also need to understand how the existing and future solutions interface to the mcu.  I think the majority use a serial uart.  Some have USB.  u-blox supports i2c but I do not know if anyone else does.

jukkar, 2020-12-14:
Also cellular modems usually provide support for GNSS. For example GSM modems have GSM muxing implemented where one UART will provide multiple virtual channels to the caller, and the GNSS data is transferred in one of these channels. The muxed UART will look same as normal UART to the application so it should work ok.

beriberikix, 2021-01-07:
@sslupsky I have only limited experience with GPS but the vast majority I've seen either use NMEA or UBX. I've never used RTCM but it seems to also be common and modern. Are there indeed other commonly used protocols than those 4 or is that good enough starting point?

@jukkar I believe this is also implied with your point but there are some modems that have an integrated GNSS and some that a dedicated interface as you described. To further add to the complexity, some have user-programmable application processors (though most need NDA'd permission) and some can only be interfaced via a host processor.

It might be worthwhile to pick a small subset of common standalone GNSS modules as well as GSM+GNSS modems as a starting point and plan to expand the design over time.

mfiumara, 2021-01-07:
I have a quectel BG95 I could experiment with, would love to see this functionality added to Zephyr.

beriberikix, 2021-01-07:
The BG95 would be a good one to target! Actually, many of the NB/Cat-M1 modules are based on the [qcomm 9205](https://www.qualcomm.com/news/releases/2019/06/26/qualcomm-technologies-announces-16-design-wins-latest-lte-iot-chipset), so picking one would provide good representation. 

My personal reqs would be to start with:
* qcomm 9205 (via the BG95) 
* Nordic nRF91
* a standalone U-blox GNSS ([Neo or Max](https://www.u-blox.com/sites/default/files/GNSS_LineCard_%28UBX-13004717%29.pdf))

RobMeades, 2021-03-28:
Hi guys.  For disclosure purposes, I'm from u-blox, though not the GNSS side (cellular).  I am interested in this discussion because we have a C support library (I'd call it a "hopefully useful lump of well-tested code") for our cellular, Wifi, BLE and soon GNSS modules and I'd like it to be compatible with this API.

A *very* quick scan suggests that the proposal above is fine, though since it comes originally from support for a u-blox GNSS chip I guess that will be no surprise :-).  Some comments:

1.  Might we want the user to have control over the size/location of this internal sample buffer?  Could be better to let them pass this in, then it can be static/`malloc()`'ed/stuck-in-some-bit-of-RAM-they-found-spare-from-an-unused-DMA-buffer, etc.  Or does the generic Zephyr `dev` thingy allow this?
2.  As is mentioned above, GNSS devices can often be hidden behind other devices: for instance, many of our customers connect the GNSS chip to one of our cellular modules and then talk to it through that, allowing synergy with cellular-based positioning that is not otherwise possible.  Effectively the [in our case] UBX messages from the host get passed transparently through an AT interface and are muxed in the cellular module with messages from the cellular device itself.  So it would be sensible if the message-level interface within the above was clearly available for such arrangements, though I suppose that could be done on a device-specific basis.
3.  Assistance information: time to first fix may be significantly reduced through provision of assistance information (current time or approximate position or satellite metadata from a previous session, allowing the GNSS chip to find satellites more quickly).  While this might not be included right now, it would be worth thinking about how it would be added to the API.
4.  Looking to the future, this is obviously a GNSS API but parts of it are common with a more abstract "Location" API.  For instance, location can be achieved via cellular (alone or with GNSS), most recently BLE, etc.  Does such a "Location" API exist within Zephyr and, if not, do we want to sketch out what it might look like in order to at least put the abstracted types in the right place?

jukkar, 2021-03-29:
Hi @RobMeades 

couple of random comments to your questions:
* Currently we do not have any location API in zephyr, so we need one. If you have one in mind, just propose such an API
* The above API has not been sent yet to review so people have not really been looking it properly. One thing to note, is that the API needs to be device agnostic, it should not depend on ublox or any other chip manufacturer behavior.
* Please no dynamic allocation in APIs etc. If some buffer is needed by the API, then that could be provided by the API caller. The device API allows one to attach context data into it if needed.

Anyway, it would be nice to see a PR in review, so just propose something if you have the code and we can then start to comment it.

avigreen1978, 2021-04-22:
The GNSS API in zephyr is really desired.
What's the current status of that?

carlescufi, 2021-05-17:
@avigreen1978 as usual with open source work, someone needs to send an RFC Pull Request so we can start considering it for addition to the tree.

allen-k-liu, 2021-07-12:
@sslupsky I'm trying to adopt gnss api for our zephyr porting on silab pg board.  do you have any suggestion or any reference that we can work on?
Thanks!

sslupsky, 2021-07-12:
@allen-k-liu I built an API using the header I posted above and implemented a driver for a u-blox gnss device (SAM-M8Q).  The driver and API work for my use case but it is not a complete solution.  I am a couple zephyr versions behind so the code needs to be rebased to work with the most recent release of Zephyr.  If someone was interested to collaborate on the code development it might make sense for me to generate an RFC PR and use my code as a starting point.

allen-k-liu, 2021-07-12:
thanks a lot @sslupsky ! I'm interested to do so.  We original planed to use u-blox m8q but probably will switch to sony's cxd5605.  I'm not familiar with RFC PR, is it something you will summit to zephyr to merge to main line?

sslupsky, 2021-07-12:
@allen-k-liu I had a look at the PG board:
https://www.silabs.com/documents/public/user-guides/ug474-pg22-dk2503a-user-guide.pdf

does this board have gnss capability?

allen-k-liu, 2021-07-12:
@sslupsky i just did quick scan, i don't see it.  I used efm32pg12 (stk3402a) that zephyr has supported.  But I haven't figure out where to insert gps module.  probably under zephyr/module/gps/

allen-k-liu, 2021-08-03:
@sslupsky Would you mind if you can share your gnss project branch so I can base on that to not go too far off?  Thanks a lot!

sslupsky, 2022-07-06:
@allen-k-liu I posted all my out-of-tree zephyr stuff here:  https://github.com/sslupsky/zephyr-out-of-tree

bperseghetti, 2023-06-30:
Is there a good way we can "revive" the evaluation of this RFC?

bjarki-trackunit, 2024-02-10:
We now have a vendor agnostic GNSS API supporting both Quectel and Ublox, and should support Nordic as well, included from release 3.6 If there is some request in this RFC that is still relevant, please reopen this issue and we can take it from there

closed 2024-02-10 by bjarki-trackunit

beriberikix, 2024-02-10:
Linking the documentation [here](https://docs.zephyrproject.org/latest/hardware/peripherals/gnss.html) so others can find it in the future :)

bperseghetti, 2024-02-10:
@bjarki-trackunit @beriberikix are there any thoughts on extending this to include RTCM3 corrections?

bperseghetti, 2024-02-10:
If not yet should this be re-opened or a new one opened?

bjarki-trackunit, 2024-02-10:
@bperseghetti That would require a new RFC :)

ubieda, 2025-05-22:
> [@bperseghetti](https://github.com/bperseghetti) That would require a new RFC :)

In case it is useful. I've filed an RFC for this here: #90363