zephyrproject-rtos/zephyr #30566: flashing issue with ST Nucleo board H745ZI-Q

vagotec, 2020-12-09:
Hi,
there is a bug with ST Nucleo H745ZI-Q. The same issue (Ticket:#29732). But the solution does not work.
----------------------------------------------------------------------
openocd.cfg
source [find board/st_nucleo_h745zi.cfg]

reset_config srst_only srst_nogate connect_assert_srst

$_CHIPNAME.cpu0 configure -event gdb-attach {
        echo "Debugger attaching: halting execution"
        reset halt
        gdb_breakpoint_override hard
}

$_CHIPNAME.cpu0 configure -event gdb-detach {
        echo "Debugger detaching: resuming execution"
        resume
}

-------------------------------------------------------------------------------------
Application build is ok
run on Hardware not ok (flash)
-------------------------------------------------------------------------------------
ERROR:

[  1%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[  2%] Built target parse_syscalls_target
[  2%] Built target kobj_types_h_target
[  2%] Built target syscall_list_h_target
[  3%] Built target driver_validation_h_target
[  3%] Built target offsets
[  4%] Built target offsets_h
[  4%] Built target zephyr_generated_headers
[ 24%] Built target kernel
[ 25%] Built target app
[ 47%] Built target zephyr
[ 48%] Built target linker_script_target
[ 50%] Built target isr_tables
[ 51%] Built target arch__common
[ 59%] Built target arch__arm__core__aarch32
[ 66%] Built target arch__arm__core__aarch32__cortex_m
[ 68%] Built target arch__arm__core__aarch32__cortex_m__mpu
[ 80%] Built target lib__libc__minimal
[ 82%] Built target lib__posix
[ 84%] Built target soc__arm__common__cortex_m
[ 85%] Built target drivers__gpio
[ 87%] Built target drivers__serial
[ 92%] Built target ..__modules__hal__stm32__stm32cube
[ 94%] Built target zephyr_prebuilt
[ 95%] Built target linker_pass_final_script_target
[ 98%] Built target zephyr_final
[100%] Flashing nucleo_h745zi_q_m7
-- west flash: using runner openocd
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-06-25-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warning : hla does not support multicore debugging
Warning : could not use CTI with a single core device, CTI is disabled
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1800 kHz
Info : STLINK V3J2M1 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.292615
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Error: mem2array: Read @ 0x5c001004, w=4, cnt=1, failed
Error executing event examine-end on target stm32h7x.cpu0:
/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg:199: Error: 
in procedure 'stm32h7x_dbgmcu_mmw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 149
in procedure 'stm32h7x_mmw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 224
in procedure 'stm32h7x_mrw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 205
at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 199
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32h7x.cpu0      hla_target little stm32h7x.cpu       reset

Error: mem2array: Read @ 0x5c001004, w=4, cnt=1, failed
Error executing event examine-end on target stm32h7x.cpu0:
/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg:199: Error: 
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 230
in procedure 'stm32h7x_dbgmcu_mmw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 149
in procedure 'stm32h7x_mmw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 224
in procedure 'stm32h7x_mrw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 205
at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 199
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0802c410 msp: 0x240090f0
16440 bytes written at address 0x08000000
292 bytes written at address 0x08004038
downloaded 16732 bytes in 0.206083s (79.288 KiB/s)

shutdown command invoked
[100%] Built target flash
-------------------------------------------------------------------------
Environment:
OS: Linux, ubuntu 20.04
Toolchain (Zephyr SDK: 11.0.4) and Eclipse Plug-In (Ver.: Version: 0.2.99.202009250453)

erwango, 2020-12-10:
@vagotec can you check that this isn't fixed with https://github.com/zephyrproject-rtos/zephyr/pull/30417 ?

vagotec, 2020-12-10:
Hello erwango,
unfortunately I get another error now.
my changes in relation to this document: https://github.com/zephyrproject-rtos/zephyr/pull/30417/files/768126367154828ceecab4998c712bcf361eab7a

-------------------------------------------------------------------------------------------------------
boards/arm/nucleo_h745zi_q/board.cmake
-------------------------------------------------------------------------------------------------------
# SPDX-License-Identifier: Apache-2.0
board_runner_args(jlink "--device=STM32H745ZI" "--speed=4000")

include(${ZEPHYR_BASE}/boards/common/openocd.board.cmake)
include(${ZEPHYR_BASE}/boards/common/jlink.board.cmake)
----------------------------------------------------------------------------------------------------------
boards/arm/nucleo_h745zi_q/support/openocd.cfg
----------------------------------------------------------------------------------------------------------

source [find board/st_nucleo_h745zi.cfg]

# Use connect_assert_srst here to be able to programm
# even when core is in sleep mode
reset_config srst_only srst_nogate connect_assert_srst

$_CHIPNAME.cpu0 configure -event gdb-attach {
        echo "Debugger attaching: halting execution"
        gdb_breakpoint_override hard
}

$_CHIPNAME.cpu0 configure -event gdb-detach {
        echo "Debugger detaching: resuming execution"
        resume
}

# Due to the use of connect_assert_srst, running gdb requires
# to reset halt just after openocd init.
rename init old_init
proc init {} {
        old_init
        reset halt
}
---------------------------------------------------------------------------------------------------
New Error (Fehler):
-------------------------------------------------------------------------------------------------
[  1%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[  2%] Built target parse_syscalls_target
[  2%] Built target kobj_types_h_target
[  2%] Built target syscall_list_h_target
[  3%] Built target driver_validation_h_target
[  3%] Built target offsets
[  4%] Built target offsets_h
[  4%] Built target zephyr_generated_headers
[ 24%] Built target kernel
[ 25%] Built target app
[ 47%] Built target zephyr
[ 48%] Built target linker_script_target
[ 50%] Built target isr_tables
[ 51%] Built target arch__common
[ 59%] Built target arch__arm__core__aarch32
[ 66%] Built target arch__arm__core__aarch32__cortex_m
[ 68%] Built target arch__arm__core__aarch32__cortex_m__mpu
[ 80%] Built target lib__libc__minimal
[ 82%] Built target lib__posix
[ 84%] Built target soc__arm__common__cortex_m
[ 85%] Built target drivers__gpio
[ 87%] Built target drivers__serial
[ 92%] Built target ..__modules__hal__stm32__stm32cube
[ 94%] Built target zephyr_prebuilt
[ 95%] Built target linker_pass_final_script_target
[ 98%] Built target zephyr_final
[100%] Flashing nucleo_h745zi_q_m7
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/sarvg/code/zephyr_ws/Nuucleo-H745ZI-Q/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-06-25-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warning : hla does not support multicore debugging
Warning : could not use CTI with a single core device, CTI is disabled
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
init
Info : clock speed 1800 kHz
Info : STLINK V3J2M1 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.285638
Error: init mode failed (unable to connect to the target)
in procedure 'init'

FATAL ERROR: command exited with status 1: /home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/sarvg/zephyrproject/zephyr/boards/arm/nucleo_h745zi_q/support -s /home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/sarvg/zephyrproject/zephyr/boards/arm/nucleo_h745zi_q/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/sarvg/code/zephyr_ws/Nuucleo-H745ZI-Q/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/sarvg/code/zephyr_ws/Nuucleo-H745ZI-Q/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
make[3]: *** [zephyr/cmake/flash/CMakeFiles/flash.dir/build.make:58: zephyr/cmake/flash/CMakeFiles/flash] Fehler 1
make[2]: *** [CMakeFiles/Makefile2:3270: zephyr/cmake/flash/CMakeFiles/flash.dir/all] Fehler 2
make[1]: *** [CMakeFiles/Makefile2:3277: zephyr/cmake/flash/CMakeFiles/flash.dir/rule] Fehler 2
make: *** [Makefile:716: flash] Fehler 2

erwango, 2020-12-10:
@vagotec It seems you're using sdk version zephyr-sdk-0.11.4.
Latest beta version (0.12) includes several additions/fixes on h7 support on openocd. Would you mind upgrading to latest SDK and have a new try ?

vagotec, 2020-12-10:
Hello erwango,
i tried with Zephyr SDK 0.12.0 Beta-3: https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.12.0-beta-3 and the same configuration. 
i still get errors:
-------------------------------------------------------------------------------------------
Error:

[  1%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[  2%] Built target parse_syscalls_target
[  2%] Built target kobj_types_h_target
[  2%] Built target syscall_list_h_target
[  3%] Built target driver_validation_h_target
[  3%] Built target offsets
[  4%] Built target offsets_h
[  4%] Built target zephyr_generated_headers
[ 24%] Built target kernel
[ 25%] Built target app
[ 47%] Built target zephyr
[ 48%] Built target linker_script_target
[ 50%] Built target isr_tables
[ 51%] Built target arch__common
[ 59%] Built target arch__arm__core__aarch32
[ 66%] Built target arch__arm__core__aarch32__cortex_m
[ 68%] Built target arch__arm__core__aarch32__cortex_m__mpu
[ 80%] Built target lib__libc__minimal
[ 82%] Built target lib__posix
[ 84%] Built target soc__arm__common__cortex_m
[ 85%] Built target drivers__gpio
[ 87%] Built target drivers__serial
[ 92%] Built target ..__modules__hal__stm32__stm32cube
[ 94%] Built target zephyr_prebuilt
[ 95%] Built target linker_pass_final_script_target
[ 98%] Built target zephyr_final
[100%] Flashing nucleo_h745zi_q_m7
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/sarvg/code/zephyr_ws/myRobot/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01508-gf79c90268-dirty (2020-12-02-20:36)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
init
Info : STLINK V3J2M1 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.291018
Error: init mode failed (unable to connect to the target)

FATAL ERROR: command exited with status 1: /home/sarvg/zephyr-sdk-0.12.0-beta-3/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/sarvg/zephyrproject/zephyr/boards/arm/nucleo_h745zi_q/support -s /home/sarvg/zephyr-sdk-0.12.0-beta-3/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/sarvg/zephyrproject/zephyr/boards/arm/nucleo_h745zi_q/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/sarvg/code/zephyr_ws/myRobot/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/sarvg/code/zephyr_ws/myRobot/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
make[3]: *** [zephyr/cmake/flash/CMakeFiles/flash.dir/build.make:58: zephyr/cmake/flash/CMakeFiles/flash] Fehler 1
make[2]: *** [CMakeFiles/Makefile2:3270: zephyr/cmake/flash/CMakeFiles/flash.dir/all] Fehler 2
make[1]: *** [CMakeFiles/Makefile2:3277: zephyr/cmake/flash/CMakeFiles/flash.dir/rule] Fehler 2
make: *** [Makefile:716: flash] Fehler 2

erwango, 2020-12-11:
Is is possible not to shout :-) ?
Ok, thanks for the additional test. Well this is strange, as I have no issue with the same board on my side.

Would you be able to describe exactly the procedure that doesn't work (flash/plugg/unplugg, ...).
An idea might be to start from a clean sheet and start by fully erasing the flash using STM32CubeProgrammer.

On my side, I'll have few more tests and get back to you. Likely next week, though

erwango, 2020-12-14:
@vagotec I just had a try this morning and everything works fine on my board.
Please provide more information on the steps to reproduce and use west -vvv when running you commands to provide more hints on what is going wrong.

vagotec, 2020-12-17:
Hello Erwan,
After your comments and tests, i ordered a new board and re-tested everything. Now everything works fine.It's hard to believe that my new (old) board was broken (defekt from factory).

Thank you very much for your efforts (Thanksssssss)

erwango, 2020-12-17:
That's a relief !! :-)
@vagotec Thanks for the heads up! Closing the point.

closed 2020-12-17 by erwango