zephyrproject-rtos/zephyr #30574: up_squared: tests/kernel/semaphore/semaphore failed.

chen-png, 2020-12-10:
**Describe the bug**
when running "tests/kernel/semaphore/semaphore" on up_squared, this test will stop at the "test_sem_reset" and fail with timeout.
the last good commit is 6e561570, the first bad commit is da70d20a.
and there were lots of tests failed with the same error:
        tests/posix/eventfd
	tests/posix/common
	tests/portability/cmsis_rtos_v1
        tests/kernel/queue
        tests/kernel/fifo/fifo_timeout

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/kernel/semaphore/semaphore
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
 Booting Zephyr OS build zephyr-v2.4.0-2184-g87ddddae5222 
Running test suite test_semaphore

START - test_k_sem_define
PASS - test_k_sem_define

START - test_k_sem_init
PASS - test_k_sem_init

START - test_sem_thread2thread
PASS - test_sem_thread2thread

START - test_sem_thread2isr
PASS - test_sem_thread2isr

**START - test_sem_reset**

**the test will stop at there.**

**Environment (please complete the following information):**
 - OS: Fedora32
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: 87ddddae5222

nashif, 2020-12-11:
based on quick test, this seems to be caused by #30573

dcpleung, 2020-12-11:
Traced it to `CONFIG_INTEL_VTD_ICTL` being enabled for UP Squared board. After building with it disabled, the test ran fine. So this is not the same cause as #30573, as that problem is still there after disabling `CONFIG_INTEL_VTD_ICTL`. So re-opening this one as the root-cause is different.

dcpleung, 2020-12-11:
@tbursztyka `CONFIG_INTEL_VTD_ICTL` is causing issue on UP Squared board.

tbursztyka, 2020-12-15:
@dcpleung I will check

nashif, 2020-12-17:
this should be closed after the revert in #30716

closed 2020-12-17 by nashif