zephyrproject-rtos/zephyr #30578: Reusing test fixtures between various tests

nordic-krch, 2020-12-10:
There are some tests which may rely on the same fixture. `gpio_loopback` is one example. It is used for `gpio_basic_api` https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/drivers/gpio/gpio_basic_api/testcase.yaml and it could also be reused for `uart` or `spi_loopback` test since all those requires few pins to be shortened. However, I don't see what would be the best way to reuse it. `gpio_basic_api` has binding for that https://github.com/zephyrproject-rtos/zephyr/tree/master/tests/drivers/gpio/gpio_basic_api/dts/bindings.

Anyone have suggestion how to reuse same fixture between those tests?

nordic-krch, 2020-12-10:
@nashif can you help?

pabigot, 2020-12-10:
For the case of `gpio-loopback` as a fixture I think we need to move the binding that supports it up into the zephyr dts hierarchy, possibly as a compatible `zephyr,fixture,gpio-loopback`.  Then we could add nodes to specific boards so they don't have to be replicated in the tests.

I will prototype that for `gpio-loopback` so we can see what it looks like.