zephyrproject-rtos/zephyr #30582: Doxygen doesn't catch errors in argument names in callback functions that are @typedef'd

Thalley, 2020-12-10:
**Describe the bug**
For this bug I'll use a simple callback (bt_ready_cb_t`) as an example, but this seem to be a generic issue. It is at this time declared as:

```
/**
 * @typedef bt_ready_cb_t
 * @brief Callback for notifying that Bluetooth has been enabled.
 *
 * @param err zero on success or (negative) error code otherwise.
 */
typedef void (*bt_ready_cb_t)(int err);
```

If the declaration is updated to
```
typedef void (*bt_ready_cb_t)(int res);
```
instead, but the `@param err` isn't, then doxygen does not capture this. 

**To Reproduce**
Introduce a doxygen error (e.g. create a mismatch between the doxygen argument names and the function/callback argument names) and build doxygen and observe no error. 

**Expected behavior**
Th expected behavior is the same as for any "proper" function where the documentation and declaration is a mismatch, which causes a warning. More speficically I'd expect
```
warning: argument 'err' of command @param is not found in the argument list of (*bt_ready_cb_t)(int res)
warning: The following parameter of  (*bt_ready_cb_t)(int res) is not documented:
  parameter 'res'
```

**Impact**
Risk that some callback functions are not properly documented. 

**Environment (please complete the following information):**
 - OS: Linux
 - Doxygen version: 1.8.20

utzig, 2020-12-17:
I submitted a fix to Doxygen with https://github.com/doxygen/doxygen/pull/8259, which was already merged, so once they ship a new release we can look into upgrading and close this issue.

nashif, 2021-02-08:
fixed by #30799

closed 2021-02-08 by nashif