zephyrproject-rtos/zephyr #30618: Add arduino header/spi support for HSDK board

vanwinkeljan, 2020-12-10:
**Is your enhancement proposal related to a problem? Please describe.**

Currently the HSDK board has an `ILI9340` display added on the arduino spi interface (hsdk `spi2`) which is not physically present on the board and as such a display shield should be used instead.

Problem in removing the display is that the HSDK board is missing the arduino header & spi.
Adding the arduino spi is straight forward but the arduino header pinout is missing documentation.

**Describe the solution you'd like**

Add arduino spi and header to HSDK dts

**Describe alternatives you've considered**

Remove `ILI9340` display from `spi2` as-is

**Additional context**

This problem popped up while cleaning up unnecessary config overlays in the LVGL sample.
In case the arduino spi and headers are supported the file `samples/display/lvgl/boards/hsdk.conf` can be removed.

abrodkin, 2021-01-19:
@IRISZZW could you please check this one and see if there's a possibility to add support of that particular peripheral on the HSDK in Zephyr.

IRISZZW, 2021-01-22:
we have some documents about the Arduino pins in HSDK board: [HSDK doc](https://docs.zephyrproject.org/latest/boards/arc/hsdk/doc/index.html)

There are some problem to add the `arduino header` for HSDK, HSDK board includes a CY8C9520A I/O expander, and some of the `arduino header` pins use the extended I/O. But there is no CY8C9520A driver in zephyr currently.

IRISZZW, 2021-05-18:
I think this issue has been fixed by this PR: #33907.
@vanwinkeljan may we close this issue?

ruuddw, 2021-05-26:
Closing this one now - @vanwinkeljan feel free to reopen or open another issue if #33907 doesn't handle this issue.

closed 2021-05-26 by ruuddw