zephyrproject-rtos/zephyr #30634: frdm_kw41z: Current master fails compilation in drivers/pwm/pwm_mcux_tpm.c

pfalcon, 2020-12-11:
Build error messages:

```
00:13:06.112  2883 | #define DT_FOREACH_OKAY_INST_nxp_kinetis_tpm(fn) fn(0) fn(1)
00:13:06.112       |                                                        ^~
00:13:06.112 /home/buildslave/workspace/zephyr-upstream/zephyr/include/sys/util_internal.h:106:36: note: in expansion of macro 'DT_FOREACH_OKAY_INST_nxp_kinetis_tpm'
00:13:06.112   106 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
00:13:06.112       |                                    ^
00:13:06.112 /home/buildslave/workspace/zephyr-upstream/zephyr/drivers/pwm/pwm_mcux_tpm.c:203:1: note: in expansion of macro 'DT_INST_FOREACH_STATUS_OKAY'
00:13:06.112   203 | DT_INST_FOREACH_STATUS_OKAY(TPM_DEVICE)
00:13:06.112       | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
00:13:06.112 /home/buildslave/workspace/zephyr-upstream/zephyr/include/device.h:186:2: error: type defaults to 'int' in declaration of 'DEVICE_DT_DEFINE' [-Werror=implicit-int]
```

Full log:
https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/5818/PLATFORM=frdm_kw41z,ZEPHYR_TOOLCHAIN_VARIANT=zephyr,label=docker-xenial-amd64-13/console

Specific commit built: d5072d13786edc4bc34a169e743d5aeeee34a819

pfalcon, 2020-12-11:
Posted to capture the issue and avoid possibility of duplicating effort. I didn't perform further triaging so far.

@agansari, @MaureenHelm : FYI

galak, 2020-12-11:
Fixed by #30644

PR #30644 (2020-12-11): drivers: pwm: pwm_mcux_tpm: Fix build failure
Add a missing ',' in DEVICE_DT_INST_DEFINE

Fixes #30634
closed 2020-12-11 by galak