zephyrproject-rtos/zephyr #30635: cpu_stats: Change from printk to LOG_*

koffes, 2020-12-11:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, if you want to print CPU stats, https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/tracing/cpu_stats.c#L155 uses printk.

This has two drawbacks:
1. Printk will interrupt ordinary `LOG_*` making output scrambled and more challenging to read.
2. `CONFIG_TRACING_CPU_STATS_LOG=y `makes the user think that `LOG_*` is used, when this is not the case.

**Describe the solution you'd like**
Suggest that this printout is changed to `LOG_INF`

nashif, 2020-12-11:
sound reasonable, however I am planning to actually change the way we collect stats and move away from this implementation (CPU based) to something based on threads (using `CONFIG_THREAD_RUNTIME_STATS`).

nashif, 2021-06-16:
this was replaced by thread runtime statistics, we no longer use tracing for cpu stats

closed 2021-06-16 by nashif