zephyrproject-rtos/zephyr #30636: TCP stack locks irq's for too long

lund-prevas, 2020-12-11:
In subsys/net/ip/tcp2.c, function "net_tcp_get", the irq_lock() is used to avoid interrupts.

On a Freedom K64 board, with a Kinetis K64 running 120 MHz, this function locks interrupts for ~164 us
Same board provides several UART's with no flowcontrol signals and no FIFO, so each byte has to be read before next byte is received.
With a baudrate on 115200, there is ~86 us between each byte, so locking interrupts for 164 us causes data loss on UART's

TCP is a logical component, so there is no reason to use irq_lock, instead it should use k_mutex or k_semaphore to protect critical data.

jukkar, 2020-12-14:
Hi @lund-prevas, I have a fix proposal at #30707. I did test it but could you report does it help in your use case?

lund-prevas, 2020-12-15:
Hi, 
It does help. I am not seeing data loss on the UART when applying this commit

PR #30707 (2020-12-15): net: tcp2: Use mutex instead of irq_lock
When needing to lock non-connection related access, use k_mutex
instead of locking irq. There is really no reason to prevent the
system from generating interrupts.

Fixes #30636
commit 9080a46e (2020-12-15): net: tcp2: Use mutex instead of irq_lock

When needing to lock non-connection related access, use k_mutex
instead of locking irq. There is really no reason to prevent the
system from generating interrupts.

Fixes #30636
closed 2020-12-15 by jukkar