zephyrproject-rtos/zephyr #30649: Trouble with gpio callback on frdm k64f

EinarHenriksen, 2020-12-11:
**Describe the bug**
I'm trying to set up a gpio callback, but it's not working as I'd expect.
I want the callback to trigger when I receive a high signal at the PTC12 port (arduino header D8) on my frdm k64f.
I suspect that I'm not using the device tree correctly, but I've done everything like in the examples I've seen and can't figure out any other way to do this.
The button example works correctly on my platform.
I've checked with an oscilloscope that the interrupt signal appears on the right pin and looks like it's supposed to, it just doesn't trigger my callback.
If I've done something the wrong way in my code, please let me know.

**To Reproduce**
My code:
#define INT_PORT        DT_LABEL(DT_ALIAS(gpio_c))
#define INT_FLAGS      (GPIO_INPUT | GPIO_ACTIVE_HIGH)

static struct gpio_callback cb_data;
void gpio_interrupt(const struct device *dev, struct gpio_callback *cb, uint32_t pins)
{
        printk("callback triggered\n");
}

void main(void) {
        const struct device *int_port;
        int_port = device_get_binding(INT_PORT);
        gpio_pin_configure(int_port, 12, INT_FLAGS);
        gpio_pin_interrupt_configure(int_port, 12, GPIO_INT_EDGE_TO_ACTIVE);
        gpio_init_callback(&cb_data, gpio_interrupt, BIT(12));
        gpio_add_callback(int_port, &cb_data);

        while(1) {
                k_sleep(K_SECONDS(5));
         }
}

**Expected behavior**
I expect to see "callback triggered" printed in my terminal when my device receives an interrupt signal

**Impact**
Unable to use gpio interrupts

**Environment (please complete the following information):**
 - OS: Linux (ubuntu)
 - Toolchain Zephyr SDK (2.4.99)

jfischer-no, 2020-12-11:
I do not think that there is a bug, try this:

```diff c
diff --git a/boards/arm/frdm_k64f/pinmux.c b/boards/arm/frdm_k64f/pinmux.c
index 8f798e446c..1285489bf3 100644
--- a/boards/arm/frdm_k64f/pinmux.c
+++ b/boards/arm/frdm_k64f/pinmux.c
@@ -58,6 +58,8 @@ static int frdm_k64f_pinmux_init(const struct device *dev)
        /* FXOS8700 INT2 */
        pinmux_pin_set(portc, 13, PORT_PCR_MUX(kPORT_MuxAsGpio));
 
+       pinmux_pin_set(portc, 12, PORT_PCR_MUX(kPORT_MuxAsGpio));
+
        /* SW3 */
        pinmux_pin_set(porta,  4, PORT_PCR_MUX(kPORT_MuxAsGpio));
 
```

EinarHenriksen, 2020-12-13:
> I do not think that there is a bug, try this:
> 
> ```diff
> diff --git a/boards/arm/frdm_k64f/pinmux.c b/boards/arm/frdm_k64f/pinmux.c
> index 8f798e446c..1285489bf3 100644
> --- a/boards/arm/frdm_k64f/pinmux.c
> +++ b/boards/arm/frdm_k64f/pinmux.c
> @@ -58,6 +58,8 @@ static int frdm_k64f_pinmux_init(const struct device *dev)
>         /* FXOS8700 INT2 */
>         pinmux_pin_set(portc, 13, PORT_PCR_MUX(kPORT_MuxAsGpio));
>  
> +       pinmux_pin_set(portc, 12, PORT_PCR_MUX(kPORT_MuxAsGpio));
> +
>         /* SW3 */
>         pinmux_pin_set(porta,  4, PORT_PCR_MUX(kPORT_MuxAsGpio));
>  
> ```

I tried adding this, but it doesn't seem to make any difference.
My callback still isn't getting triggered.

MaureenHelm, 2021-02-01:
Please try enabling the pull down. I was able to make `samples/basic/button` work with this pin with the following changes:

```diff
diff --git a/boards/arm/frdm_k64f/frdm_k64f.dts b/boards/arm/frdm_k64f/frdm_k64f.dts
index f587059e0f..52ed4a15e5 100644
--- a/boards/arm/frdm_k64f/frdm_k64f.dts
+++ b/boards/arm/frdm_k64f/frdm_k64f.dts
@@ -12,7 +12,7 @@
                led0 = &green_led;
                led1 = &blue_led;
                led2 = &red_led;
-               sw0 = &user_button_3;
+               sw0 = &ptc12;
                sw1 = &user_button_2;
        };
 
@@ -52,6 +52,10 @@
                        label = "User SW3";
                        gpios = <&gpioa 4 GPIO_ACTIVE_LOW>;
                };
+               ptc12: button_2 {
+                       label = "PTC12";
+                       gpios = <&gpioc 12 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>;
+               };
        };
 
        arduino_header: connector {
diff --git a/boards/arm/frdm_k64f/pinmux.c b/boards/arm/frdm_k64f/pinmux.c
index c7b3f33e39..2892baeb12 100644
--- a/boards/arm/frdm_k64f/pinmux.c
+++ b/boards/arm/frdm_k64f/pinmux.c
@@ -58,6 +58,8 @@ static int frdm_k64f_pinmux_init(const struct device *dev)
        /* FXOS8700 INT2 */
        pinmux_pin_set(portc, 13, PORT_PCR_MUX(kPORT_MuxAsGpio));
 
+       pinmux_pin_set(portc, 12, PORT_PCR_MUX(kPORT_MuxAsGpio));
+
        /* SW3 */
        pinmux_pin_set(porta,  4, PORT_PCR_MUX(kPORT_MuxAsGpio));
```

MaureenHelm, 2021-02-04:
I think the bug is in your application, but please reopen this issue if you disagree.

closed 2021-02-04 by MaureenHelm