zephyrproject-rtos/zephyr #30657: Friendship ends if LPN publishes to a VA it is subscribed to

stefbo, 2020-12-11:
**Describe the bug**
Given a single LPN and friend node in friendship and a Generic Level Server model on the LPN which has a virtual address as publish address (say 0xbeef) the same LPN is subscribed also to, when publishing a (say) STATUS message on this model (with master credentials), then the friendship ends, because subsequent update messages are ignored by the LPN.

After some investigation, this seems to happen:

- After the STATUS message was published and the friend received it, the friend queues the STATUS message into the LPN's queue. That is, because the friend has an entry in the LPNs subscription list for 0xbeef.

- On the next poll, the friend delivers the STATUS message to the LPN.

- On reception, the LPN starts to decrypt the message in `net.c:net_decrypt()` but then realizes, that one of its elements is the origin. This is the relevant code (w/ extended logging):
```
    if (bt_mesh_elem_find(rx->ctx.addr)) {
      BT_DBG("Dropping locally originated packet: src=0x%04x, dst=0x%04x.\n", rx->ctx.addr, rx->ctx.recv_dst);
      return -EBADMSG;
    }
```

- Because of this, processing the message by the LPN stops at the Network layer. No further processing occurs. The FSN is not toggled.

- The next time the LPN polls for messages, it uses the previously used FSN.

- Because of the unchanged FSN, the friend tries to re-send the previous message, which is again dropped at the LPN side.

- This is continued until the LPN stops retrying and clears friendship.

**To Reproduce**
Steps to reproduce the behavior:
As described above.

**Expected behavior**
Good question. I guess the LPN should either accept its own message or drop it and toggle the FSN. In any case, friendship must not be terminated because of this.

**Impact**
Currently, the combination of a LPN having a model which publishes on a (at least) virtual address, it is also subscribed to, will terminate the friendship when publishing a message. Having such a set-up seems very common to me.

**Logs and console output**

```
[00:00:32.945,404] <dbg> bt_mesh_net.net_decrypt: Dropping locally originated packet
[00:00:32.945,404] <dbg> bt_mesh_net.bt_mesh_net_decode: Unable to find matching net for packet
```

LingaoM, 2020-12-12:
> * After the STATUS message was published and the friend received it, the friend queues the STATUS message into the LPN's queue. That is, because the friend has an entry in the LPNs subscription list for 0xbeef.
> *

This is obviously a specification error. LPN messages should not be stored in friend queue. This error has been corrected by SIG, mesh profile draft R13 3.5.5 friend queue.

LingaoM, 2020-12-12:
> * On reception, the LPN starts to decrypt the message in `net.c:net_decrypt()` but then realizes, that one of its elements is the origin. This is the relevant code (w/ extended logging):
>
> ```
>     if [...]

Network layer discards the message with SRC as its own address. MESH.TS 1.0.1.2 4.8.4 MESH/NODE/RLY/BI-02-C [Ignore Invalid SRC] will specifically check this.

PR #30669 (2020-12-16): Bluetooth: Mesh: Fixes Friend Queue store message
If the SRC field of the received message is a unicast
address of an element of the Low Power node, then the
message shall not be stored in the Friend Queue.

Otherwise, lpn will discard this message, eventually
it breaks friendship.

Fixes: #30657
commit 5b11c053 (2020-12-16): Bluetooth: Mesh: Fixes Friend Queue store message

If the SRC field of the received message is a unicast
address of an element of the Low Power node, then the
message shall not be stored in the Friend Queue.

Otherwise, lpn will discard this message, eventually
it breaks friendship.

Fixes: #30657
closed 2020-12-16 by carlescufi