zephyrproject-rtos/zephyr #30659: High cpu usage with udp sockets

vandita5, 2020-12-11:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
I ran characterization for throughput and cpu usage. I modified sockets/echo_server sample's udp file to run a thread that receives an initial message, then starts sending data in a loop. After sending a certain amount of data, the thread sleeps for a certain interval before the next iteration. The data is received using a python script, which outputs transmission stats including recorded throughput. I used CONFIG_TRACING option to log cpu usage. 
running this on nucleo f429zi dev kit, I am able to push the throughput to 13.5132Mbps at which, no logs print so I'm assuming ~100% usage. I see 97% cpu usage at 13.0986Mbps.  for my use case I require >8Mbps which shows cpu usage of ~62%. Are these values expected? Are there ways to optimize these?

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...) 2.1.99 (version tweak = 0 ;extra version none)
 - Commit SHA or Version used

Hardware setup: Nucleo STM32F429ZI connected via cat5e path cable to USB to Ethernet hub, then connected to a Lenovo Thinkpad.

**Additional context**
Add any other context about the problem here.
Based on sample - net/sockets/echo_server
Modified process_udp function
```
static int process_udp(struct data *data)
{
  const uint32_t SLEEP_TIME_MS         = 2;
  const uint32_t DUMMY_DATA_SIZE_BYTES = 6250;

  int ret                                          = 0;
  bool print_debug_info_first_send                 = true;
  static uint8_t dummy_data[DUMMY_DATA_SIZE_BYTES] = {0};

  memset(dummy_data, 1, DUMMY_DATA_SIZE_BYTES);

  NET_INFO("Waiting for UDP packets on port %d (%s)...", MY_PORT, data->proto);

  struct sockaddr client_addr;
  socklen_t client_addr_len = sizeof(client_addr);

  int received = recvfrom(data->udp.sock, data->udp.recv_buffer,
                          sizeof(data->udp.recv_buffer), 0, &client_addr,
                          &client_addr_len);
  if (received < 0)
  {
    /* Socket error */
    NET_ERR("UDP (%s): Connection error %d", data->proto, errno);
    ret = -errno;
  }
  else
  {
    uint32_t tmp_len = 0;
    uint8_t *tmp_ptr = dummy_data;
    while (true)
    {
      tmp_len = DUMMY_DATA_SIZE_BYTES;
      tmp_ptr = dummy_data;
      do
      {
        ret = sendto(data->udp.sock, tmp_ptr, tmp_len, 0, &client_addr,
                     client_addr_len);
        if (ret < 0)
        {
          NET_ERR("UDP (%s): Failed to send %d", data->proto, errno);
          ret = -errno;
          return ret;
        }
        if (print_debug_info_first_send)
        {
          NET_INFO("sent %d remaining %d", ret, tmp_len);
          print_debug_info_first_send = false;
        }
        tmp_len = tmp_len - ret;
        tmp_ptr = tmp_ptr + ret;

        if (++data->udp.counter % 1000 == 0U)
        {
          NET_INFO("%s UDP: Sent %u packets", data->proto, data->udp.counter);
        }

      } while (tmp_len);

      k_sleep(SLEEP_TIME_MS);
    }
  }

  return ret;
}
```

additional config in prj.cnf
```
# Tracing
# CPU usage
CONFIG_TRACING=y
CONFIG_TRACING_CPU_STATS=y
CONFIG_TRACING_CPU_STATS_LOG=y

# Network related
CONFIG_NET_STATISTICS=y
```

vandita5, 2020-12-11:
@jukkar Have you worked with udp or can you help tag someone who could help with this?

jukkar, 2020-12-14:
Not sure what was the issue as the bug template was not filled fully. Are you just asking if that kind of performance is expected? For UDP I have seen similar numbers but have not tried that hw.
There is a draft PR at #30346 that tries to minimize various latencies in the network stack. You can certainly try that if it improves throughput.

vandita5, 2020-12-14:
@jukkar Yes, that was my intention. Thanks for sharing the PR link. I'll take a look and try it out once it shows up as ready but probably early next year. 
My overall intention is to see if there is other work planned to improve the cpu usage and if not, are there any suggestions available on what I could try to improve the cpu usage. My application requires ~8Mbps throughput and I'd like to minimize the cpu usage as much as possible.

jukkar, 2020-12-15:
Ok, I misunderstood the issue. So basically you are looking a solution where you could cap the performance (mbit/sec) to some number. Right now we do not have that kind of support implemented, patches are welcome if you would like to implement one.

vandita5, 2020-12-15:
Yes. More like I'm hoping to get a lower cpu usage. probably around 20-30% to get 8Mbps throughput from UDP. I honestly, don't have ideas around it as I haven't looked into udp or networking code. I am happy to try things out and implement a patch but I'd need some help as I haven't worked with this kind of code before. So, I was hoping to get ideas or get to connect with someone working on similar issues.

jukkar, 2020-12-15:
Ok, one way to implement this could be something like this:

Modify `subsys/net/ip/net_core.c:net_recv_data()` and add some functionality there to calculate the incoming traffic and optionally
  1. drop traffic directly or
  1. inform the driver that it should stop accepting incoming traffic

For these the 1st is easy, we just drop the received packet with some criteria. This will affect the upper stack, but we are still receiving packets so the driver would still cause load to the system.
The 2nd one is better for the load of the device as we would not receive data at all for some period of time.

Open questions:
* what is the metric to use, is it mbit/sec or the load of the system etc
* how the user configures this performance cap value, is it global or per network interface
* in 2nd option, how the incoming traffic is accepted again (do we need a timer or thread etc)
* in 2nd option, what is the API towards the driver. I would imagine the API would be optional in the driver side, if it does not implement it, no throughput capping would happen

These were just some ideas for discussion. There could be other solutions also for this.

jukkar, 2020-12-16:
IMHO this is more like an enhancement than a bug so changing the label.

jukkar, 2023-11-15:
Updated the issue title to reflect the desired outcome.

carlescufi, 2025-03-12:
Similar to #27662