zephyrproject-rtos/zephyr #30679: puncover does not work with sdk-0.12.0-beta3

pabigot, 2020-12-13:
sdk-0.12.0-beta-3 breaks puncover support for worst-case stack analysis from https://docs.zephyrproject.org/latest/guides/optimizations/tools.html#build-target-puncover.  No stack data is visible in the puncover web; dropping back to sdk-0.11.4 it shows up.

Not clear whether the problem needs to be fixed here, in the sdk, or in HBehrens/puncover

nashif, 2020-12-13:
just verified the generated .su files are the same across gcc versions.. so this is not the problem.

galak, 2020-12-14:
> just verified the generated .su files are the same across gcc versions.. so this is not the problem.

I take that to mean its probably on the puncover side.

nashif, 2021-01-30:
I am not getting worst-case stack analysis with older SDK and with older Zephyr tree, my suspicion is that some dependency in the modules started breaking this, so it is not a bug in zephyr, need to open a bug in puncover issues. The basic functionality of puncover is still there and works fine.

closed 2021-01-30 by nashif

larson-whisper, 2021-12-21:
To ease the search for others, there is a PR open on puncover to fix the issue.  Applying the fix locally restored stack analysis for us.

https://github.com/HBehrens/puncover/pull/43/files

yashi, 2021-12-26:
The development of puncover seems to be halted. (Three commits in 2020 are the last. Before that is 2017)  Is it possible to host it under Zephyr project?  Or integrate it with Zephyr SDK?