zephyrproject-rtos/zephyr #30683: up_squared_adsp:running tests/kernel/sched/schedule_api failed

KangJianX, 2020-12-14:
Describe the bug
running tests/kernel/sched/schedule_api error, it showed assertion failed.

To Reproduce
Steps to reproduce the behavior:

sanitycheck -W -p intel_adsp_cavs15 --device-testing -T tests/kernel/sched/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/up_squared_adsp/tools/up_squared_adsp_flash.sh, /home/ztest/work/sof/rimage/keys/otc_private_key.pem" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/dump_trace.py" -x=CONFIG_IPM=y -x=CONFIG_CONSOLE=y -x=CONFIG_LOG_PRINTK=n

see error:
START - test_slice_reset
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:91: thread_time_slice: (t <= expected_slice_max is false)
timeslice too big, expected 3840384 got 3840934
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:88: thread_time_slice: (t >= expected_slice_min is false)
timeslice too small, expected 3839616 got 15295
FAIL - test_slice_reset

Environment (please complete the following information):
OS: Fedora28
Toolchain: Zephyr-sdk-0.11.4
Commit ID: 90d06cff

KangJianX, 2021-01-07:
@andyross This issue can be reproduce after your patch merged. This test cases still run failed.

andyross, 2021-02-03:
Cleaning up.  This is working for me now.  It's almost certain (given the test at fault) that this was a duplicate of the timeslicing bug that @ceolin merged last week.  Closing, please reopen if you have trouble.

closed 2021-02-03 by andyross