zephyrproject-rtos/zephyr #30685: reel_board: tests/kernel/fatal/exception/ failure

s-kelvin, 2020-12-14:
**Describe the bug**
Running testcase kernel.common.stack_sentinel in tests/kernel/fatal/exception/ on reel board. The output is incomplete and the testcase appears to be a failure in sanity check.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p auto tests/kernel/fatal/exception/ -- -DOVERLAY_CONFIG=sentinel.conf
2. west flash

**Expected behavior**
Test should pass.

**Impact**
Test is stuck and fails.

**Logs and console output**
```
ASSERTION FAIL [chan && chan < (0 + 1)] @ WEST_TOPDIR/zephyr/drivers/timer/nrf_rtc_timer.c:214
E: r0/a1:  0x00000004  r1/a2:  0x000000d6  r2/a3:  0x00000001
E: r3/a4:  0x00000021 r12/ip:  0x00000000 r14/lr:  0x00002163
E:  xpsr:  0x41000021
E: Faulting instruction address (r15/pc): 0x0000ec3e
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20000298 (main)
Caught system error -- reason 4
Was not expecting a crash
```

**Environment (please complete the following information):**
- Linux Ubuntu 18.04.4 LTS 
- OS: zephyr-v2.4.0-2281-g90d06cffecc6
- Toolchain Zephyr SDK 0.11.2

pabigot, 2020-12-14:
This also fails on other Nordic boards including pca10056.

pabigot, 2020-12-14:
@nordic-krch The failure is an assertion in `z_nrf_rtc_timer_compare_set()`.

PerMac, 2020-12-14:
This is already reported here: https://github.com/zephyrproject-rtos/zephyr/issues/30554
and here https://github.com/zephyrproject-rtos/zephyr/issues/30526
Closing this issue due to repetition.

closed 2020-12-14 by PerMac