zephyrproject-rtos/zephyr #30686: getaddrinfo() does not respect socket type

SeppoTakalo, 2020-12-14:
**Describe the bug**
When requesting an address for listening socket using `getaddrinfo()` with `AI_PASSIVE` in the flags field, the returned address is always using socket type `SOCK_STREAM`. Following example demonstrates this:

```.c
static const char *socktype_str(int type) {
    switch (type) {
    case SOCK_DGRAM:    return "SOCK_DGRAM";
    case SOCK_STREAM:   return "SOCK_STREAM";
    case SOCK_RAW:      return "SOCK_RAW";
    default:            return "SOCK_???";
    }
}

void test_getaddrinfo_hints(void) {
    struct addrinfo *res;
    struct addrinfo hints = {
        .ai_family = AF_INET,
        .ai_socktype = SOCK_DGRAM,	/* I'm requesting UDP address */
        .ai_flags = AI_PASSIVE,
    };

    int r = getaddrinfo(NULL, "1234", &hints, &res);
    if (r) {
        printf("Error: getaddrinfo() %d\n", r);
        return;
    }

    for (struct addrinfo *ai = res; ai; ai = ai->ai_next) {
        int s = socket(ai->ai_family, ai->ai_socktype, ai->ai_protocol);
        if (s == -1)
            continue;

        /* Verify that we got a correct socket type */
        if (ai->ai_socktype != SOCK_DGRAM) {
            printf("ERROR: I got type %s\n", socktype_str(ai->ai_socktype));
        }

        if (bind(s, ai->ai_addr, ai->ai_addrlen) == 0) {
            /* Bind succesfull
             * I should have now listening UDP socket on port 1234 */
            break;
        }
        close(s);
    }
    freeaddrinfo(res);
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Prepare test network for QEMU `./net-setup.sh`
2. `west build -b qemu_cortex_m3 -t run`
3. Check output.

**Expected behavior**
When AI_PASSIVE flag is specified in hints, I would expect `getaddrinfo()` to return `struct sockaddr` suitable to be used with `bind()` call, as defined by POSIX.

**Impact**
Usage of `getaddrinfo()` differs from other POSIX compliant systems.

**Logs and console output**
```
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
*** Booting Zephyr OS build zephyr-v2.4.0-2344-g4def9a386b52  ***
ERROR: I got type SOCK_STREAM
thread: started
QEMU: Terminated
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: zephyr-v2.4.0-2344-g4def9a386b

**Additional context**
Add any other context about the problem here.

jukkar, 2020-12-14:
Indeed, the code seems to hard code `SOCK_STREAM` type in this case. See https://github.com/zephyrproject-rtos/zephyr/blob/905f496603b905a05d8ae41d81b2cac7987232e7/subsys/net/lib/sockets/getaddrinfo.c#L129 for details. The fix looks quite simple, @SeppoTakalo would you be able to fix this and submit a PR?

SeppoTakalo, 2020-12-14:
Sure.
Submitted a PR https://github.com/zephyrproject-rtos/zephyr/pull/30711

PR #30711 (2020-12-15): Fix getaddrinfo() to respect socket type hints
If getaddrinfo() was called with AI_PASSIVE flag in hints,
returned address defaulted to SOCK_STREAM and IPPROTO_TCP.
Fixed so that SOCK_DGRAM leads to correct address type
that can be fed to bind() directly.

Same hard coding was visible when numerical IPv4 address string
was converted to binary. That is also fixed to respect hints now.

Fixes #30686
commit f684b7c6 (2020-12-15): net: getaddrinfo: Fix getaddrinfo() to respect socket type hints

If getaddrinfo() was called with AI_PASSIVE flag in hints,
returned address defaulted to SOCK_STREAM and IPPROTO_TCP.
Fixed so that SOCK_DGRAM leads to correct address type
that can be fed to bind() directly.

Same hard coding was visible when numerical IPv4 address string
was converted to binary. That is also fixed to respect hints now.

Also, added functionality to get IPv6 address, when hints contained
AF_INET6.

Fixes #30686
closed 2020-12-15 by jukkar