zephyrproject-rtos/zephyr #30688: Using openthread based  lwm2m_client cannot ping the external network address unless reset once

taoyinzhou, 2020-12-14:
**Describe the bug**
I want to implement the thread terminal to automatically connect to the OTBR by commissioning after it starts, and then connect to the external network.
OTBR runs openwrt Integrated with otbr-posix.
The nrf52840 runs .hex file which is one compiled from samples/net/lwm2m_client .
overlay-ot.conf modified：
`CONFIG_OPENTHREAD_JOINER=y
CONFIG_OPENTHREAD_JOINER_AUTOSTART=y
CONFIG_OPENTHREAD_JOINER_PSKD="H00R11"
CONFIG_OPENTHREAD_SLAAC=y
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=6`
Before the terminal starts, OTBR will create Thread network with dataset first.
`ot-ctl dataset init new
ot-ctl dataset commit active
ot-ctl prefix add fd11:22::/64 pasor
ot-ctl ifconfig up
ot-ctl thread start
ot-ctl netdata register
ot-ctl commissioner start
ot-ctl commissioner joiner add f4ce36dbda49db09 H00R11`
Then the terminal start up,  i can see the print: net_l2_openthread: Join success. 
The role becomes to router.Then ping 64:ff9b::8.8.8.8.

`uart:~$ ot ipaddr
fd11:22:0:0:9f08:e2a4:55fb:1ed2
fd29:ceec:9986:f495:0:ff:fe00:3405
fd29:ceec:9986:f495:4c17:fed1:1e5a:e436
fe80:0:0:0:d465:305:ab6b:802
fdde:ad00:beef:0:0:0:0:1
Done
uart:~$ net ping 64:ff9b::8.8.8.8
PING 64:ff9b::8.8.8.8
Ping timeout`

Then re-plug the power or ot reset. The terminal will be able to ping  the external network
`uart:~$ ot state
router
Done
uart:~$ net ping 64:ff9b::8.8.8.8
PING 64:ff9b::8.8.8.8
8 bytes from 64:ff9b::808:808 to fd11:22::e656:880f:ea71:9128: icmp_seq=1 ttl=112 rssi=0 time=43 ms
8 bytes from 64:ff9b::808:808 to fd11:22::e656:880f:ea71:9128: icmp_seq=2 ttl=112 rssi=0 time=45 ms`

This problem is like #29567, but he is directly set panid and channel to connect, I have tried this way is normal, but using commissioning there will have problem.

**To Reproduce**
Steps to reproduce the behavior:
west build -s samples/net/lwm2m_client -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf"

**Expected behavior**
OTBR start commissioning.  Then theThread device  power on, it can connect to the Thread Network and connect to the internet.

**Impact**
After the Thread device connects to the network, it must be restarted once, otherwise it cannot connect to the external network

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

rlubos, 2020-12-15:
I'm sorry but I cannot reproduce the issue. I've built the `lwm2m_client` sample with the config changes you've listed (nrf52840dk_nrf52840), commissioned my device into the network (wpantund-based border router) and I could ping `net ping 64:ff9b::0808:0808` right away after joining, no reboot required.

taoyinzhou, 2020-12-16:
I used the example samples/net/sockets/echo_server to print less .
overlay-ot.conf modified：
```
CONFIG_OPENTHREAD_JOINER=y
CONFIG_OPENTHREAD_JOINER_AUTOSTART=y
CONFIG_OPENTHREAD_JOINER_PSKD="H00R11"
CONFIG_OPENTHREAD_SLAAC=y
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=6
```
compile
```
west build -s samples/net/sockets/echo_server -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-ot.conf"
```
The steps to reproduce the problem are exactly the same. Below is my log, hopefully that will help you figure out what the problem is
```
[00:05:33.760,375] <inf> net_l2_openthread: State changed! Flags: 0x08000000 Current role: 0
[00:05:33.789,337] <inf> net_l2_openthread: State changed! Flags: 0x08000000 Current role: 0
[00:05:33.852,691] <inf> net_l2_openthread: State changed! Flags: 0x08040100 Current role: 0
[00:05:33.953,033] <inf> net_l2_openthread: Join success
[00:05:33.953,338] <inf> net_l2_openthread: State changed! Flags: 0x0800101f Current role: 1
[00:05:35.353,271] <inf> net_l2_openthread: State changed! Flags: 0x001b32b7 Current role: 2
[00:05:35.353,515] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 2
uart:~$ ot ipaddr
fd11:22:0:0:d25f:6be4:4e33:b7a2
fd2a:2faa:1a6:a8c:0:ff:fe00:5401
fd2a:2faa:1a6:a8c:8b6f:dc69:ca7b:4f6e
fe80:0:0:0:b834:d395:dc24:86d0
fdde:ad00:beef:0:0:0:0:2
Done
[00:05:50.365,447] <inf> net_l2_openthread: State changed! Flags: 0x00000064 Current role: 3
[00:05:53.788,269] <inf> net_l2_openthread: State changed! Flags: 0x00000200 Current role: 3
uart:~$ net ping 64:ff9b::808:808
PING 64:ff9b::808:808
Ping timeout
[00:06:03.826,110] <err> net_otPlat_radio: Error while calling otIp6Send
[00:06:04.826,782] <err> net_otPlat_radio: Error while calling otIp6Send
[00:06:05.827,423] <err> net_otPlat_radio: Error while calling otIp6Send
uart:~$ net ping 64:ff9b::808:808
PING 64:ff9b::808:808
Ping timeout
[00:06:16.574,584] <err> net_otPlat_radio: Error while calling otIp6Send
[00:06:17.575,256] <err> net_otPlat_radio: Error while calling otIp6Send
[00:06:18.575,897] <err> net_otPlat_radio: Error while calling otIp6Send
uart:~$ ot reset

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-2186-gc45e7f3c9643  ***
[00:00:00.253,387] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.266,357] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.266,357] <inf> fs_nvs: alloc wra: 0, f40
[00:00:00.266,357] <inf> fs_nvs: data wra: 0, 17c
[00:00:00.270,111] <inf> net_l2_openthread: State changed! Flags: 0x001fc300 Current role: 0
[00:00:00.270,446] <inf> net_l2_openthread: OpenThread version: OPENTHREAD/gf9fbfad01; Zephyr; Dec 16 2020 15:22:47
[00:00:00.271,209] <inf> net_l2_openthread: Network name: OpenThreadGuide
[00:00:00.274,536] <inf> net_l2_openthread: State changed! Flags: 0x0100103d Current role: 1
[00:00:00.281,097] <inf> net_config: Initializing network
[00:00:00.281,250] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.281,677] <inf> net_echo_server_sample: Run echo server
[00:00:00.281,707] <inf> net_echo_server_sample: Network connected
[00:00:00.281,738] <inf> net_echo_server_sample: Starting...
[00:00:00.281,860] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:00.982,696] <inf> net_l2_openthread: State changed! Flags: 0x000010e4 Current role: 3
[00:00:00.997,253] <inf> net_l2_openthread: State changed! Flags: 0x00000200 Current role: 3
[00:00:00.997,253] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 3
uart:~$ net ping 64:ff9b::808:808
PING 64:ff9b::808:808
8 bytes from 64:ff9b::808:808 to fd11:22::d25f:6be4:4e33:b7a2: icmp_seq=0 ttl=112 rssi=0 time=51 ms
8 bytes from 64:ff9b::808:808 to fd11:22::d25f:6be4:4e33:b7a2: icmp_seq=2 ttl=112 rssi=0 time=54 ms
uart:~$ ipaddr
ipaddr: command not found
uart:~$ ot ipaddr
fd11:22:0:0:d25f:6be4:4e33:b7a2
fd2a:2faa:1a6:a8c:0:ff:fe00:c000
fdde:ad00:beef:0:0:0:0:2
fd2a:2faa:1a6:a8c:8b6f:dc69:ca7b:4f6e
fe80:0:0:0:b834:d395:dc24:86d0
Done
uart:~$ 
```

rlubos, 2020-12-16:
Ok, I think I've figured it out. In your example the OT network uses non-default on-mesh prefix, `fd29:ceec:9986:f495::/64` (which is perfectly fine btw.), but you still register a static address with the default prefix (`fdde:ad00:beef:0:0:0:0:2`). Because of that, this IP address is not recognized by Zephyr as mesh-local and can be selected as source address for external destinations. We end up with two addresses that can be used as a source for external networks:
* `fd11:22:0:0:d25f:6be4:4e33:b7a2` (the one assigned by the BR)
* `fdde:ad00:beef:0:0:0:0:2` (the one statically configured in the app).

Since both addresses are equally good for the Zephyr source address matching algorithm (https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/net_if.c#L2407), the choice really depends on their order in the IPv6 address table (that's why it works after the reset, the addresses are re-ordered). And in case `fdde:ad00:beef:0:0:0:0:2` is used, OpenThread does not recognize it as a valid source address for external destination, hence reports an error.

So, in order to get the external routing to work correctly, you should not register a static address with a custom prefix (set `CONFIG_NET_CONFIG_MY_IPV6_ADDR=""`).

taoyinzhou, 2020-12-17:
Thank you. The problem solved

taoyinzhou, 2020-12-17:
I did what you said and changed CONFIG_NET_CONFIG_MY_IPV6_ADDR="".
At first it was fine, but just now the network was down again, I looked at the ipaddr,
found that there is an additional address. 2001:db8:0:0:0:0:0:1
```
uart:~$ net ping 64:ff9b::8.8.8.8
PING 64:ff9b::8.8.8.8
Ping timeout
[00:05:48.481,048] <err> net_otPlat_radio: Error while calling otIp6Send
[00:05:49.481,689] <err> net_otPlat_radio: Error while calling otIp6Send
[00:05:50.482,330] <err> net_otPlat_radio: Error while calling otIp6Send
uart:~$ ot ipaddr
fdf8:4cb8:aaef:319b:0:ff:fe00:7800
fd11:22:0:0:eeb9:41b3:fdb9:28be
fdf8:4cb8:aaef:319b:6dbb:2144:2f72:698
fe80:0:0:0:602d:c559:5880:78e8
2001:db8:0:0:0:0:0:1
Done
uart:~$ 
```
Even reset didn't help. I searched the code and didn't find that I had set this address.
Then I entered dataset clear and factoryreset into the board. at this time, it was not connected to the thread network, but it already had this address, which means it should have come with it since startup.
```
uart:~$ ot dataset clear
Done
[00:10:39.807,128] <err> net_l2_openthread: Join failed [23]
[00:10:39.807,708] <inf> net_l2_openthread: State changed! Flags: 0x0800800b Current role: 0
uart:~$ ot factoryreset

uart:~$ *** Booting Zephyr OS build zephyr-v2.4.0-2186-gc45e7f3c9643  ***
[00:00:00.253,387] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.266,387] <inf> fs_nvs: 8 Sectors of 4096 bytes
[00:00:00.266,387] <inf> fs_nvs: alloc wra: 0, d70
[00:00:00.266,387] <inf> fs_nvs: data wra: 0, 3c0
[00:00:00.267,333] <inf> net_l2_openthread: State changed! Flags: 0x00038200 Current role: 0
[00:00:00.267,669] <inf> net_l2_openthread: OpenThread version: OPENTHREAD/gf9fbfad01; Zephyr; Dec 17 2020 14:24:39
[00:00:00.267,852] <inf> net_l2_openthread: Starting OpenThread join procedure.
[00:00:00.268,463] <inf> net_l2_openthread: State changed! Flags: 0x0900900b Current role: 0
[00:00:00.275,115] <inf> net_config: Initializing network
[00:00:00.275,238] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 0
[00:00:00.275,634] <inf> net_echo_server_sample: Run echo server
[00:00:00.275,665] <inf> net_echo_server_sample: Network connected
[00:00:00.275,695] <inf> net_echo_server_sample: Starting...
[00:00:00.275,848] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:05.118,103] <err> net_l2_openthread: Join failed [23]
[00:00:05.118,682] <inf> net_l2_openthread: State changed! Flags: 0x0800800b Current role: 0
uart:~$ ot ipaddr
fe80:0:0:0:dc3d:da9e:c436:facc
2001:db8:0:0:0:0:0:1
Done
[00:00:09.965,972] <err> net_l2_openthread: Join failed [23]
```
This address was not there at the beginning and I did not write it in the configuration file.Do you know why this is happening?

taoyinzhou, 2020-12-17:
CONFIG_NET_CONFIG_MY_IPV6_ADDR=""

closed 2020-12-17 by taoyinzhou