zephyrproject-rtos/zephyr #30694: reel_board enables non-minimal peripherals by default

gmarull, 2020-12-14:
**Describe the bug**

`reel_board` enables non-minimal peripherals by default like SPI. This goes against board porting guidelines, as it enables a peripheral that is not necessarily used by all samples.

So for example, if I take the "hello_world" sample I get:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       23224 B         1 MB      2.21%
            SRAM:        4488 B       256 KB      1.71%
        IDT_LIST:         120 B         2 KB      5.86%
```
If I compile the same sample with CONFIG_SPI=n, as it's not needed:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       19244 B         1 MB      1.84%
            SRAM:        4200 B       256 KB      1.60%
        IDT_LIST:          88 B         2 KB      4.30%
```

Not only that, but if one tries to build e.g. MCUboot, it overflows ROM area because of SPI being unnecessarily enabled.

**To Reproduce**

Build a sample that does not use SPI with defaults and compare its ROM size with SPI disabled.

**Expected behavior**

Binary size should be minimal.

**Impact**

Generated binary sizes will not be minimal. Other undesired side effects is that MCUboot can't be compiled on reel_board (flash overflow) because SPI is unnecessarily enabled.

**Logs and console output**

N/A

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - 7356cb2e455c3c5cc7110ab35f7cca0404156f54

**Additional context**
Add any other context about the problem here.

jfischer-no, 2020-12-14:
FYI @jremmert-phytec-iot

jfischer-no, 2020-12-14:
There are other boards that do not follow the _guideline_.
@gmarull Do you want to open a general issue about this?

boards/arc/hsdk/hsdk_2cores_defconfig:CONFIG_SPI=y
boards/arc/hsdk/hsdk_defconfig:CONFIG_SPI=y
boards/arm/96b_carbon_nrf51/96b_carbon_nrf51_defconfig:CONFIG_SPI=y
boards/arm/adafruit_feather_m0_basic_proto/adafruit_feather_m0_basic_proto_defconfig:CONFIG_SPI=y
boards/arm/atsamd20_xpro/atsamd20_xpro_defconfig:CONFIG_SPI=y
boards/arm/atsamd21_xpro/atsamd21_xpro_defconfig:CONFIG_SPI=y
boards/arm/atsamr21_xpro/atsamr21_xpro_defconfig:CONFIG_SPI=y
boards/arm/ip_k66f/ip_k66f_defconfig:CONFIG_SPI=y
boards/arm/pinetime_devkit0/pinetime_devkit0_defconfig:CONFIG_SPI=y
boards/arm/reel_board/reel_board_defconfig:CONFIG_SPI=y
boards/arm/reel_board/reel_board_v2_defconfig:CONFIG_SPI=y
boards/arm/sam4l_ek/sam4l_ek_defconfig:CONFIG_SPI=y
boards/arm/serpente/serpente_defconfig:CONFIG_SPI=y
boards/riscv/hifive1/hifive1_defconfig:CONFIG_SPI=y
boards/riscv/litex_vexriscv/litex_vexriscv_defconfig:CONFIG_SPI=y

gmarull, 2020-12-14:
@jfischer-no converted to a generic issue

mbolivar-nordic, 2021-08-02:
@jfischer-no @gmarull do you have some list of priorities for configs that are enabled that should not be? I'm trying to figure out how to write a script to grep for this issue.

gmarull, 2021-08-02:
I think that other than UART/GPIO/PINMUX, the rest should be disabled by default (and UART is there because it's the most frequent way to see that samples do something).

mbolivar-nordic, 2021-08-02:
"the rest" is not a list :stuck_out_tongue:

carlescufi, 2021-09-08:
@gmarull please follow-up with a list of Kconfig options that we think should be disabled by default on all boards, so that @mbolivar-nordic can work on the script.

gmarull, 2021-09-08:
> @gmarull please follow-up with a list of Kconfig options that we think should be disabled by default on all boards, so that @mbolivar-nordic can work on the script.

It's hard to tell the minimum defaults, since it may be platform-specific. As I mentioned, I think the bare minimum would be gpio, pinmux and uart (as noted, uart because it's usually how all samples/tests are _observed_).

Some options that are not minimal:

```
git grep CONFIG_SENSOR -- boards/**/*defconfig
boards/arm/bt510/bt510_defconfig:CONFIG_SENSOR=y
boards/arm/mec1501modular_assy6885/mec1501modular_assy6885_defconfig:CONFIG_SENSOR=y
boards/arm/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig:CONFIG_SENSOR=y
```

```
git grep CONFIG_I2C -- boards/**/*defconfig
boards/arc/hsdk/hsdk_defconfig:CONFIG_I2C=y
boards/arm/adafruit_feather_m0_basic_proto/adafruit_feather_m0_basic_proto_defconfig:CONFIG_I2C=y
boards/arm/adafruit_feather_m0_basic_proto/adafruit_feather_m0_basic_proto_defconfig:CONFIG_I2C_SAM0=y
boards/arm/bt510/bt510_defconfig:CONFIG_I2C=y
boards/arm/degu_evk/degu_evk_defconfig:CONFIG_I2C=y
boards/arm/efr32mg_sltb004a/efr32mg_sltb004a_defconfig:CONFIG_I2C=y
boards/arm/faze/faze_defconfig:CONFIG_I2C=y
boards/arm/faze/faze_defconfig:CONFIG_I2C_0=y
boards/arm/faze/faze_defconfig:CONFIG_I2C_1=y
boards/arm/google_kukui/google_kukui_defconfig:CONFIG_I2C=y
boards/arm/mec1501modular_assy6885/mec1501modular_assy6885_defconfig:CONFIG_I2C=y
boards/arm/mec1501modular_assy6885/mec1501modular_assy6885_defconfig:CONFIG_I2C_INIT_PRIORITY=60
boards/arm/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig:CONFIG_I2C=y
boards/arm/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig:CONFIG_I2C_INIT_PRIORITY=60
boards/arm/mec172xevb_assy6906/mec172xevb_assy6906_defconfig:CONFIG_I2C=y
boards/arm/mps2_an385/mps2_an385_defconfig:CONFIG_I2C=y
boards/arm/mps2_an521/mps2_an521_defconfig:CONFIG_I2C=y
boards/arm/mps2_an521/mps2_an521_ns_defconfig:CONFIG_I2C=y
boards/arm/mps2_an521/mps2_an521_remote_defconfig:CONFIG_I2C=y
boards/arm/mps3_an547/mps3_an547_defconfig:CONFIG_I2C=y
boards/arm/npcx7m6fb_evb/npcx7m6fb_evb_defconfig:CONFIG_I2C=y
boards/arm/npcx9m6f_evb/npcx9m6f_evb_defconfig:CONFIG_I2C=y
boards/arm/particle_boron/particle_boron_defconfig:CONFIG_I2C=y
boards/arm/pinetime_devkit0/pinetime_devkit0_defconfig:CONFIG_I2C=y
boards/arm/rcar_h3ulcb/rcar_h3ulcb_cr7_defconfig:CONFIG_I2C=y
boards/arm/rcar_h3ulcb/rcar_h3ulcb_cr7_defconfig:CONFIG_I2C_RCAR=y
boards/riscv/it8xxx2_evb/it8xxx2_evb_defconfig:CONFIG_I2C=y
boards/riscv/it8xxx2_evb/it8xxx2_evb_defconfig:CONFIG_I2C_ITE_IT8XXX2=y
boards/riscv/litex_vexriscv/litex_vexriscv_defconfig:CONFIG_I2C=y
boards/riscv/litex_vexriscv/litex_vexriscv_defconfig:CONFIG_I2C_LITEX=y
boards/xtensa/esp32/esp32_defconfig:CONFIG_I2C=y
boards/xtensa/esp32/esp32_defconfig:CONFIG_I2C_ESP32=y
boards/xtensa/esp32/esp32_defconfig:CONFIG_I2C_0=y
boards/xtensa/esp32/esp32_defconfig:CONFIG_I2C_1=y
boards/xtensa/odroid_go/odroid_go_defconfig:CONFIG_I2C=y
boards/xtensa/odroid_go/odroid_go_defconfig:CONFIG_I2C_ESP32=y
boards/xtensa/odroid_go/odroid_go_defconfig:CONFIG_I2C_0=y
```

```
git grep CONFIG_SPI -- boards/**/*defconfig
boards/arc/hsdk/hsdk_2cores_defconfig:CONFIG_SPI=y
boards/arc/hsdk/hsdk_defconfig:CONFIG_SPI=y
boards/arm/96b_carbon_nrf51/96b_carbon_nrf51_defconfig:CONFIG_SPI=y
boards/arm/96b_carbon_nrf51/96b_carbon_nrf51_defconfig:CONFIG_SPI_SLAVE=y
boards/arm/ip_k66f/ip_k66f_defconfig:CONFIG_SPI=y
boards/arm/pinetime_devkit0/pinetime_devkit0_defconfig:CONFIG_SPI=y
boards/riscv/hifive1/hifive1_defconfig:CONFIG_SPI=y
boards/riscv/hifive1/hifive1_defconfig:CONFIG_SPI_SIFIVE=y
boards/riscv/hifive_unleashed/hifive_unleashed_defconfig:CONFIG_SPI=y
boards/riscv/hifive_unleashed/hifive_unleashed_defconfig:CONFIG_SPI_SIFIVE=y
boards/riscv/hifive_unmatched/hifive_unmatched_defconfig:CONFIG_SPI=y
boards/riscv/hifive_unmatched/hifive_unmatched_defconfig:CONFIG_SPI_SIFIVE=y
boards/riscv/litex_vexriscv/litex_vexriscv_defconfig:CONFIG_SPI=y
boards/riscv/litex_vexriscv/litex_vexriscv_defconfig:CONFIG_SPI_LITESPI=y
```

```
git grep CONFIG_PWM -- boards/**/*defconfig
boards/arm/mec1501modular_assy6885/mec1501modular_assy6885_defconfig:CONFIG_PWM=y
boards/arm/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig:CONFIG_PWM=y
boards/arm/npcx7m6fb_evb/npcx7m6fb_evb_defconfig:CONFIG_PWM=y
boards/arm/npcx9m6f_evb/npcx9m6f_evb_defconfig:CONFIG_PWM=y
boards/riscv/it8xxx2_evb/it8xxx2_evb_defconfig:CONFIG_PWM=y
boards/riscv/it8xxx2_evb/it8xxx2_evb_defconfig:CONFIG_PWM_ITE_IT8XXX2=y
boards/riscv/litex_vexriscv/litex_vexriscv_defconfig:CONFIG_PWM=y
boards/riscv/litex_vexriscv/litex_vexriscv_defconfig:CONFIG_PWM_LITEX=y
```

I think it will be hard to come up with a script that solves all cases...

PR #48574 (2022-08-05): boards: remove non-minimal peripherals from defconfig
According to the board porting guidelines, boards should "leave
peripherals and their drivers disabled by default". In Zephyr we
tipically enable GPIO and SERIAL, as they are virtually required by all
samples/tests in tree. However, for the rest of peripherals it is up to
the application/test to enable the necessary driver classes. It is also
useful that board's Kconfig.defconfig enables certain driver peripherals
based on a condition, e.g. enable I2C if SENSOR=y.

Ref. https://docs.zephyrproject.org/latest/hardware/porting/board_porting.ht [... cut]
commit ada8d728 (2022-08-05): boards: remove non-minimal peripherals from defconfig

According to the board porting guidelines, boards should "leave
peripherals and their drivers disabled by default". In Zephyr we
tipically enable GPIO and SERIAL, as they are virtually required by all
samples/tests in tree. However, for the rest of peripherals it is up to
the application/test to enable the necessary driver classes. It is also
useful that board's Kconfig.defconfig enables certain driver peripherals
based on a condition, e.g. enable I2C if SENSOR=y.

Ref. https://docs.zephyrproject.org/latest/hardware/porting/
board_porting. [... cut]
closed 2022-08-05 by carlescufi