zephyrproject-rtos/zephyr #30699: Persisting connectable extended advertising

Thalley, 2020-12-14:
**Is your enhancement proposal related to a problem? Please describe.**
The scenario is that an application wants to have a persisting connectable advertising set, i.e. an advertising set that starts advertising again after connection (if `CONFIG_BT_MAX_CONN` allows it). 

It is technically possible to do this to some extent with the current API/implementation. However, there are two challenges with the current implementation/API:

1) Multiple applications
Even with extended advertising, it is still only possible for a single connectable extended advertising set (let's call that a CEAS)  in the controller. So if two applications on the same host tries to create a CEAS each, one will fair, but can be dealt with, as one of them can simply retry on connection/disconnect. However, there's no for `Application 2` to know if `Application 1` cancels/aborts the CEAS. Such a scenario could easily happen if there's a application which uses the Mesh Proxy and also tries to do CEAS itself. 

2) "Resuming" advertising on disconnection
If e.g. `CONFIG_BT_MAX_CONN=1` and a client disconnects, then the CEAS should be restarted. However, since the connection isn't free in the `disconnect` callback, it is not possible to restart a CEAS in the disconnected callback, and will have to be scheduled to a later time. 

**Describe the solution you'd like**
For 1) allowing multiple CEAS in the host could be done, and then use e.g. a round-robin approach towards the controller; either time based or when there's a connection on the CEAS. 

For 2) allowing a CEAS to reuse a connection pointer if the connection pointer can be matched with the CEAS. 

**Describe alternatives you've considered**
Alternatively resuming a CEAS could be handled in the stack itself as legacy adv is done. 

**Additional context**
N/A.

alwa-nordic, 2022-10-21:
I have some loose ideas I wanted to share:

### Super advertiser

I imagine a "super advertiser" -- a collection of advertisers. Advertisers in a super advertiser collectively share a pool of bt_conns. First come first serve. If the pool is empty, the advertisers are all paused.

For simplicity, we can automatically define a super advertiser that consist of all started advertisers marked with some new option.

#### Application managed bt_conns

A application supplies the bt_conn allowance. The simplest approach is to make applications explicitly manage bt_conns. `bt_superadv_conn_accept(DEFAULT_SHARED_SUPERADV, bt_conn *)`. This API would have to come along with a `bt_conn_alloc()` API that would return a bt_conn in an idle state. A bt_conn would also become idle and ready for reuse at some point after disconnection.

##### Note on scanning and bt_conn management

The above scheme for letting an application control bt_conns is also useful for scanning. An application would be able to reserve a bt_conn before starting the scanner with the intention to create a connection. But it can't be mandatory. Other applications may instead want to opportunistically scan while connections slots are available, and stop the scanner if the connection pool is empty.

#### Stack managed bt_conns

Alternatively, we can come up with some scheme to track the bt_conns for the application. E.g. The application can just say how many bt_conns to reserve for the super advertiser. And a connection created from that super advertiser will be tracked and returned to the pool when that bt_conn would reach ref-count zero.

### Connection pools

Maybe we instead want a concept of a user-defined connection pool? Multiple advertisers could draw from the same pool, and pause when it is empty.

We might want to make it possible to react to the available connection count. Some applications might want to advertise something as non-connectable when the connection pool is empty.

Thalley, 2022-10-25:
@alwa-nordic I have trouble understanding your reasoning for the super advertiser. You define it as a collection of advertisers; would there be more than 1 super advertiser on a device? 

The first come, first served approach seems like something we can implement today without the concept of a super advertiser. With the `connected` callback, the code starting the advertising can then also stop it again if the connection that just connected is what it wanted. 

The idea of allocating `bt_conn`s from the app via `bt_conn_alloc` isn't bad, though I am not entirely convinced it will be an improve over the current architecture.

alwa-nordic, 2022-10-28:
Currently, starting a connectable advertiser (`bt_le_adv_start_ext`/`bt_le_adv_start`) will grab a `bt_conn`. You will not be able to start multiple advertisers if there is only one `bt_conn` available.

A super advertiser would allow you to start multiple connectable advertisers, while having only one `bt_conn` available. The first advertiser to create a connection will get the `bt_conn`. The rest will be stopped.

This is an alternative to your idea of starting advertisers in a round-robin order.

Thalley, 2022-10-28:
> A super advertiser would allow you to start multiple connectable advertisers while having only one `bt_conn` available. The first advertiser to create a connection will get the `bt_conn`. The rest will be stopped.

I get that. I am, however, worried that we then return 0 for success on ADV start, when the advertising set doesn't actually start. That gives a mismatch between what the caller think the state is, and what the actual state is.

alwa-nordic, 2022-10-28:
Yes, that's a good point. We want the application alone to manage the started/stopped property.

What I have in mind is: The advertisers can be in a new state "paused waiting on: at least one bt_conn available in super advertiser". The advertisers will go into that state instead of stopping when the super advertiser runs out of bt_conns. The applications can still use `bt_le_adv_stop` at any time to stop the advertiser completely.

Thalley, 2022-10-28:
> Yes, that's a good point. We want the application alone to manage the started/stopped property.
>
> What I have in mind is: The advertisers can be in a new state "paused waiting on: at least one [...]

That could work. We already have a callback structure in place for the advertising objects, so we can provide any state changes to the owner of the advertising object.

jori-nordic, 2024-02-09:
I consider this fixed by https://github.com/zephyrproject-rtos/zephyr/pull/68266
@Thalley please re-open if still valid.

closed 2024-02-09 by Thalley