zephyrproject-rtos/zephyr #30702: Shell module broken on LiteX/VexRiscv after release zephyr-v2.1.0

lindemer, 2020-12-14:
**Description**
In releases after [zephyr-v2.1.0](https://github.com/zephyrproject-rtos/zephyr/releases/tag/zephyr-v2.1.0) the shell module no longer accepts input on LiteX/VexRiscv running on an Arty A7 FPGA. 

**To reproduce**
Steps to reproduce the behavior:
1. Clone and follow the installation steps in https://github.com/litex-hub/zephyr-on-litex-vexriscv.
2. Load the FPGA bitstream according to those same instructions.
3. From the Zephyr repository
- `git checkout zephyr-v2.1.0`
- `cd samples/subsys/shell/shell_module`
- `west build -b litex_vexriscv`
- `lxterm /dev/ttyUSB1 --kernel build/zephyr/zephyr.bin --serial-boot` OR `flterm --port /dev/ttyUSB1 --kernel build/zephyr/zephyr.bin --kernel-adr 0x40000000`

This should flash the firmware onto the soft CPU and bring up the Zephyr shell. Repeating the above with any release after `zephyr-v2.1.0` will display the shell prompt but it does not accept any user input.

**Environment**
- Ubuntu Linux 5.4.0-58-generic
- Zephyr SDK 0.11.4

lindemer, 2020-12-16:
This may be unrelated, but Zephyr's memory footprint appears to have increased significantly after that release. Here's the RAM usage for two samples on `litex_vexriscv`:

|  | `zephyr-v2.1.0` | `zephyr-v2.2.0` |
|-|-----------------|-----------------|
| `hello_world` | 13024 B | 80560 B |
| `shell_module` | 73584 B | 138672 B |

tgorochowik, 2021-01-27:
The problem here was that the ethernet driver was getting stuck waiting for buffers to be ready with interrupts disabled. This appeared as a problem with broken shell because the serial driver needed those interrupts to work. This has been resolved by #31152.

In addition the underlying problem in the platform that caused the ethernet driver to get stuck has been resolved in the platform as well (as noted here: https://github.com/litex-hub/zephyr-on-litex-vexriscv/issues/3).

I think this can be closed too.

closed 2021-01-27 by lindemer