zephyrproject-rtos/zephyr #30705: STM32 PWM driver generates signal with wrong duty cycle on STM32G4

martinjaeger, 2020-12-14:
**Describe the bug**

The following code snipped should generate a 5 kHz PWM signal with 50% duty cycle.

```C
#define PWM_CONTROLLER "PWM_8"
#define PWM_PERIOD 200000 /* 200 us period = 5 kHz */
#define PWM_CHANNEL 1

void start_pwm()
{
    const struct device *pwm_dev = device_get_binding(PWM_CONTROLLER);
    pwm_pin_set_nsec(pwm_dev, PWM_CHANNEL, PWM_PERIOD, PWM_PERIOD / 2, 0);
}
```

The actual signal generated is 7.5 kHz (duty cycle of 50% is correct).

Tested MCUs: STM32G431 and STM32G474.

**Environment:**
 - OS: Linux
 - Zephyr master, SHA: 1b680c82345577cf81f9e3333ae6708c81df85c9

@gmarull I'm trying to find a fix for the issue myself, but if you already have an idea what could be wrong, please let me know.

gmarull, 2020-12-14:
It may be related with https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/pwm/pwm_stm32.c#L109 function. Timer clock frequency calculation has some differences amongst STM32 series.

martinjaeger, 2020-12-15:
It turns out that the reason was a too high prescaler value, which I left at the default value of `10000`.

https://github.com/zephyrproject-rtos/zephyr/blob/55df308025e29d133e4b29759e3c483534a48e84/dts/arm/st/g4/stm32g4.dtsi#L425

So the issue was my fault and it's solved with a smaller prescaler.

@gmarull Do you know the reason why the prescaler for STM32 PWM nodes is sometimes set to `10000` and sometimes to `0`? It doesn't make much sense to me... as far as I understand, `10000` does not even lead to a clean division of the system clock frequency by `10000` as the counter starts at `0` and we'd have to substract `1`.

gmarull, 2020-12-15:
@martinjaeger I don't know, I've always seen it set to 10000. @erwango do you know why?

martinjaeger, 2020-12-15:
Yeah, `10000` is much more common, but there are also quite a few occasions of `0`

```
$ git grep "st,prescaler = <10000>;" | wc -l
107
$ git grep "st,prescaler = <0>;" | wc -l
20
```

martinjaeger, 2020-12-15:
Raised the question of correct prescaler setting also in #30740, so I'm closing this issue as the original problem is resolved.

closed 2020-12-15 by martinjaeger