zephyrproject-rtos/zephyr #30708: BLE SW controller PA/LNA pins configuration in device tree

hubertmis, 2020-12-14:
It looks that currently there are two ways to define PA/LNA pins for radio drivers in Zephyr. One of them is device tree like here: https://github.com/zephyrproject-rtos/zephyr/blob/1b680c82345577cf81f9e3333ae6708c81df85c9/boards/arm/nrf21540dk_nrf52840/nrf21540dk_nrf52840.dts#L113 or https://github.com/nrfconnect/sdk-nrf/tree/master/dts/bindings/radio_fem . The other one is to use Kconfig like here: https://github.com/zephyrproject-rtos/zephyr/blob/1b680c82345577cf81f9e3333ae6708c81df85c9/subsys/bluetooth/controller/Kconfig.ll_sw_split#L549

I think it should be unified and the device tree seems like a better place to define pin numbers.

I would like Software BLE Controller to use pins description in the device tree and deprecate pins description in Kconfig.

cvinayak, 2020-12-22:
This is a welcome request. If anyone with better knowledge to port Kconfig define used, to using the devicetree then its quicker to have this enhancement.

hubertmis, 2020-12-22:
While developing FEM support for 802.15.4 we have gained enough knowledge to port Generic and Skyworks FEM bindings https://github.com/nrfconnect/sdk-nrf/tree/master/dts/bindings/radio_fem to Zephyr. However, we don't have enough knowledge about BLE Controller to help starting using DTS description in this module. I think a good starting point could be an example how MPSL is using DTS to configure FEMs:
https://github.com/nrfconnect/sdk-nrf/blob/master/subsys/mpsl/Kconfig.fem
https://github.com/nrfconnect/sdk-nrf/blob/master/subsys/mpsl/mpsl_fem.c

carlescufi, 2021-02-04:
This is probably related to https://github.com/zephyrproject-rtos/zephyr/pull/31336, which I think @cvinayak would like to complete soon.

mbolivar-nordic, 2021-02-23:
Happy to pick this up, but it looks like #31336 may already have an owner who will take this over. I'm going to leave this alone unless I hear otherwise and work on other DT related conversions in our IPs.

mbolivar-nordic, 2021-04-29:
@hubertmis I have reviewed the existing materials and I have a proposal for how to proceed.

I think the request is for me to write a single DT binding that will apply to the Skyworks 66112 device and other similar devices, as well as the nRF21540 and potentially other Nordic FEMs. I think that has to be the case since we want to be able to use all of these devices as a FEM within the split controller code in a generic way as we do now.

In order to do this in an upstream-friendly way, this will require some changes to our downstream bindings, linked here for context:

- https://github.com/nrfconnect/sdk-nrf/blob/master/dts/bindings/radio_fem/generic-fem-two-ctrl-pins.yaml
- https://github.com/nrfconnect/sdk-nrf/blob/master/dts/bindings/radio_fem/skyworks%2Csky66112-11.yaml
- https://github.com/nrfconnect/sdk-zephyr/blob/master/dts/bindings/net/wireless/nordic%2Cnrf21540-fem.yaml

The issue is that available documents use different terms:

Re: CONFIG_BT_CTLR_GPIO_PA_PIN:

- generic-fem-two-ctrl-pins.yaml and skyworks,sky66112-11.yaml seem to   call this "ctx-gpios"
- nordic,nrf21540-fem.yaml seems to call this "tx-en-gpios"
- the docs for the MPSL library which consumes these refers to this as the "PA" pin, like the existing Zephyr Kconfig option

Re: CONFIG_BT_CTLR_GPIO_LNA_PIN:

- generic-fem-two-ctrl-pins.yaml and skyworks,sky66112-11.yaml seem to call this "crx-gpios"
- nordic,nrf21540-fem.yaml seems to call this "rx-en-gpios"
- MPSL refer to this as the "LNA" pin, like the existing Zephyr Kconfig option

We need to standardize property names in the upstream binding. I suggest:

```yaml
# generic-two-pin-fem.yaml
properties:
  pa-gpios:
        type: phandle-array
        required: true
        description: |
          Configuration for the power amplifier enable pin. For
          example, to use an active-high pin 3 on GPIO controller
          gpio1, write the following:

            pa-gpios = <&gpio1 3 GPIO_ACTIVE_HIGH>;

          This pin should be activated during transmission, and
          deactivated during reception.

  lna-gpios:
        type: phandle-array
        required: true
        description: |
          Configuration for the low noise amplifier enable pin. This
          should be written in the same way as pa-gpios.

          This pin should be activated during reception, and
          deactivated during transmission.
```

```yaml
# nordic,nrf21540.yaml
compatible: "nordic,nrf21540.yaml"
include: generic-two-pin-fem.yaml
description: |
  The SoC GPIO controlling the TX_EN pin of the nRF21540 should
  be set via the pa-gpios property. Similarly, RX_EN should be
  set via lna-gpios.
properties:
  # device-specific properties, e.g. 'pdn-gpios', ...
```

```yaml
# skyworks,sky66112-11.yaml
compatible: "skyworks,sky66112-11"
include: generic-two-pin-fem.yaml
description: |
  The SoC GPIO controlling the CTX pin of the SKY66112-11 should
  be set via the pa-gpios property. Similarly, CRX should be
  set via lna-gpios.
properties:
  # device-specific properties ...
```

That is, the binding would standardize on the terms "PA" and "LNA", like the Kconfig option does now.

Developers using the device-specific bindings would need to change terms from what their datasheets use to fit in this framework E.g. CTX becomes PA for Skyworks, and RX_EN becomes LNA for nRF21540.

Work for you?

hubertmis, 2021-04-30:
@mbolivar-nordic Thanks for looking into this issue. Let me shortly explain the idea behind separated bindings downstream.

PA/LNA modules are managed with different interfaces. Some of them use simple GPIO signals (some of them as simple as one line to trigger PA and second one to trigger LNA), other use more complicated serial data interface like SPI. SKY66112 is one example of PA/LNA module with the simplest interface, while nRF21540 is much more complicated (it can use GPIO alongside SPI, it uses more lines to select TX power, reduce power consumption, select antenna).

generic-fem-two-ctrl-pins.yaml was introduced to easily create bindings for the simplest devices and reuse single software driver for all of them. It looks pin naming is copied here from SKY66112 datasheet that we used as an example. I'm fine to rename it to something more generic like `pa-gpio`.

nordic,nrf21540-fem.yaml describes much more complicated device. Pin naming in the binding comes from its datasheet. I think it would be better to keep it this way instead of renaming two of these pins to something generic. I think it should be ok to include these two pins in the binding from the generic one, as long as we can add name aliases that would allow device tree creator easily identify these pins in the design. But if the binding description language does not support aliases, I would prefer to keep nRF21540 binding as it is defined now and not include generic one. My intention is to maintain pin naming consistency between nrf21540 datasheet and its binding in Zephyr.

> I think the request is for me to write a single DT binding that will apply to the Skyworks 66112 device and other similar devices, as well as the nRF21540 and potentially other Nordic FEMs. 

I don't agree. I think we need single software driver for all simple devices like SKY, and a set of drivers for complicated devices like nRF21540. For nRF21540 we should at least manage PDN pin alongside TX and RX_EN. If drivers are separated, bindings should be separated as well. They can import common base if it would make things easier, but it is not the goal of this activity.

> We need to standardize property names in the upstream binding.

Where does this need come from? I thought it is better to copy property names from datasheet. Because of other pins, nRF21540 requires a separated driver anyway.

hubertmis, 2021-04-30:
I see now that BLE SW controller uses single severely ifdefed driver for all PA/LNA modules. I think that's the source of confusion. I'm in favor of refactoring this driver and split it into smaller parts without all these ifdefs. It should be easier to read and maintain, and would be aligned with bindings naming pins according to naming for given device.

hubertmis, 2021-04-30:
Alternatively, if it is preferred to minimize changes to BLE SW Controller, I would recommend to create aliases in there like:
```
#if NRF21540_FEM_PRESENT
#define PA_PIN DT_GET_PROPERTY(tx-en-pin)
#elif GENERIC_FEM_PRESENT
#define PA_PIN DT_GET_PROPERTY(pa-gpios)
#endif
```

cvinayak, 2021-04-30:
To reduce work, we keep the nRF21540 bindings as-is, use aliases as suggested here https://github.com/zephyrproject-rtos/zephyr/issues/30708#issuecomment-829903166 . We add the new generic and skyworks bindings inline with @mbolivar thoughts.

mbolivar-nordic, 2021-04-30:
@hubertmis hmm, maybe I'm missing something here but I'm not sure your counterproposal is workable as a generic replacement for the existing Kconfig options. It seems to me that the openisa controller also supports these Kconfig options, and we can't just go about adding SOME_VENDOR_FEM_PRESENT ifdefs everywhere for every supported controller, can we?

I agree we could write a FEM driver API that might be able to abstract differences across these modules, but that would probably have to be written against the GPIO driver API, which doesn't seem like something that would fit into the task/event system we are using in our controller, presumably for performance.

hubertmis, 2021-05-04:
@mbolivar-nordic I think DTS structure and binding shall be designed to reflect hardware devices. We should not design DTS bindings to be similar to some software implementations. Because of this I don't want to discuss how PA/LNA management is implemented in BLE SW Controller, openisa, proprietary closed-source protocols (like downstream MPSL, SD Controller, 802.15.4 radio driver) yet. I want to focus on hardware devices first.

And about hardware devices:

There are PA/LNA chips that we can manage with a generic two-line GPIO interface (RFX2401C, SKY66112). It makes sense to include "generic" PA/LNA dts for bindings of these devices, because these are compatible and can share the same driver.

There are PA/LNA chips that have more complicated interface and we cannot manage them using the "generic" interface. One example is nRF21540. I agree two GPIO lines in this device have similar meaning to the "generic" GPIO interface, but it is not enough to use these lines to manage this chip. Because of this, nRF21540 is __incompatible__ with the "generic" GPIO interface and we cannot share software drivers between "generic" GPIO PA/LNAs and nRF21540. And as a result, we should not include "generic" PA/LNA dts in nRF21540 binding. The current binding of nRF21540 device is correct. We can extend it with TX power control, but we shall not modify its basics.

After we decide how DTS bindings shall describe hardware devices, we shall start discussion how to align software implementations with the bindings.

mbolivar-nordic, 2021-11-28:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/39168

closed 2021-11-28 by mbolivar-nordic