zephyrproject-rtos/zephyr #30712: "make zephyr_generated_headers" regressed again - ";" separator for Z_CFLAGS instead of spaces

pfalcon, 2020-12-14:
`make zephyr_generated_headers`, as used to integrate Zephyr's build system with 3rd-party build systems, regressed again (previous case: https://github.com/zephyrproject-rtos/zephyr/issues/28230). Now it produces following Makefile.exports.C:

> CC = /home/pfalcon/opt/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
CXX = /home/pfalcon/opt/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-g++
OBJCOPY = /home/pfalcon/opt/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-objcopy
OBJDUMP = /home/pfalcon/opt/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-objdump
AS = /home/pfalcon/opt/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-as
AR = /home/pfalcon/opt/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-ar
NM = /home/pfalcon/opt/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-nm
GDB = /home/pfalcon/opt/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gdb
Z_CFLAGS = -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include -I/home/pfalcon/projects/pycopy/ports/zephyr/outdir/qemu_x86/zephyr/include/generated -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/soc/x86/ia32 -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/lib/libc/newlib/include -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/subsys/net/lib/sockets/. -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/subsys/net/lib/dns/. -I/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/subsys/net/l2 -isystem  -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DBUILD_VERSION=zephyr-v2.4.0-2375-g6d8fe6eae923 -D__LINUX_ERRNO_EXTENSIONS__  -Os;-imacros;/home/pfalcon/projects/pycopy/ports/zephyr/outdir/qemu_x86/zephyr/include/generated/autoconf.h;-ffreestanding;-fno-common;-g;-m32;-Wa,--divide;-imacros;/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/toolchain/zephyr_stdint.h;-Wall;-Wformat;-Wformat-security;-Wno-format-zero-length;-Wno-main;-Wno-pointer-sign;-Wpointer-arith;-Wno-address-of-packed-member;-Wno-unused-but-set-variable;-Werror=implicit-int;-fno-asynchronous-unwind-tables;-fno-pie;-fno-pic;-fno-strict-overflow;-fno-reorder-functions;-fno-defer-pop;-fmacro-prefix-map=/home/pfalcon/projects/pycopy/ports/zephyr=CMAKE_SOURCE_DIR;-fmacro-prefix-map=/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr=ZEPHYR_BASE;-fmacro-prefix-map=/mnt/hdd/projects-3rdparty/Embedded/Zephyr=WEST_TOPDIR;-ffunction-sections;-fdata-sections;-mpreferred-stack-boundary=2;-mno-sse;-std=c99 

The culprit is in the generated `Z_CFLAGS` - after some option, the separator suddenly changes from " " (space) to ";".

pfalcon, 2020-12-14:
I didn't do detailed triaging/bisecting yet, pretty late for me, may fall asleep before I finish. Assigning to you, @tejlmand, just in case, hopefully you'll be able to help, just as with the previous case above. Thanks.

pfalcon, 2020-12-14:
Ok, 8275d39d4dbac85ffa095751b5529cf78e2962f2 broke it, with telling title "cmake: adding $<SEMICOLON> instead of space on zephyr_get_compile_<type>".

tejlmand, 2020-12-14:
Fived here: https://github.com/zephyrproject-rtos/zephyr/pull/30714

PR #30714 (2020-12-15): cmake: using zephyr_get_compile_options_for_lang_as_string for export
Fixes: #30712

Now using zephyr_get_compile_options_for_lang_as_string for makefile
export of Zephyr build flags.
commit a019b6f2 (2020-12-15): cmake: using zephyr_get_compile_options_for_lang_as_string for export

Fixes: #30712

Now using zephyr_get_compile_options_for_lang_as_string for makefile
export of Zephyr build flags.
closed 2020-12-15 by nashif