zephyrproject-rtos/zephyr #30713: doc: "Variable ZEPHYR_TOOLCHAIN_VARIANT is not defined"

andrewboie, 2020-12-14:
**Describe the bug**
Imaged a new Ubuntu machine and followed all the directions on https://docs.zephyrproject.org/latest/getting_started/index.html

I used the default location for the SDK.

Unfortunately, this page doesn't mention that ZEPHYR_TOOLCHAIN_VARIANT has to be set for `twister` to work.

**To Reproduce**
```
andrew@meatwad:~/projects/zephyr/zephyr$  twister -p qemu_x86
Renaming output directory to /home/andrew/projects/zephyr/zephyr/twister-out.1
INFO    - Zephyr version: zephyr-v2.4.0-2399-g7883eedfde
INFO    - JOBS: 32
E: Variable ZEPHYR_TOOLCHAIN_VARIANT is not defined
```

**Expected behavior**
Doc page would mention this detail.

**Impact**
Minor

SeppoTakalo, 2020-12-15:
I was just searching for the same issue.
Apparently `export ZEPHYR_TOOLCHAIN_VARIANT=zephyr` is the right choice, if you installed the Zephyr SDK, and it has been visible in the documentation in the past:
https://docs.zephyrproject.org/1.14.1/getting_started/installation_linux.html#zephyr-sdk

But current documentation does not mention it. Probably at some point all the build tools detected Zephyr SDK automatically, so they removed the mention about setting that variable.

tejlmand, 2021-02-04:
> Probably at some point all the build tools detected Zephyr SDK automatically, so they removed the mention about setting that variable.

Correct. CMake handles this automatically when using newer Zephyr SDK's. (from 0.11.3 and beyond).

This issue is related to running twister, which seems to also test for the Toolchain variable, so i'm looking into aligning code so this can be done in a single place.
See also:
https://github.com/zephyrproject-rtos/zephyr/pull/31936/files#r570036292

tejlmand, 2021-02-04:
if trying to build using west or CMake / Ninja, then you should not see this issue.

SeppoTakalo, 2021-02-04:
west+CMake+Ninja works OK, as expected.
This issue is only within Twister.

PR #32003 (2021-02-05): cmake: twister: Using common script for toolchain setting verification
Fixes: #30713

Now using the common script `verify-toolchain.cmake` for verifying
toolchain settings.
commit d162e9e1 (2021-02-05): cmake: twister: Using common script for toolchain setting verification

Fixes: #30713

Now using the common script `verify-toolchain.cmake` for verifying
toolchain settings.
closed 2021-02-05 by nashif