zephyrproject-rtos/zephyr #30724: CAN J1939 Support

Navin-Sankar, 2020-12-15:
**Is your feature request related to a problem? Please describe.**
    SAE J1939 is the vehicle bus recommended practice used for communication
    and diagnostics among vehicle components. Originating in the car and
    heavy-duty truck industry in the United States, it is now widely used in
    other parts of the world.
    J1939, ISO 11783 and NMEA 2000 all share the same high level protocol.
    SAE J1939 can be considered the replacement for the older SAE J1708 and
    SAE J1587 specifications.

**Describe the solution you'd like**
In linux there is separate implementation for SAE J1939 protocol (net/can/j1939/).
Likewise I try to add j1939 protocol support in zephyr by creating **sockets_j1939.c** file in subsys/net/lib/sockets directory to handle the  socket type: **SOCK_DGRAM** and protocol: **CAN_J1939**

Navin-Sankar, 2020-12-15:
@jukkar @alexanderwachter requesting for comments and feedback.

jukkar, 2020-12-15:
> Likewise I try to add j1939 protocol support in zephyr by creating **sockets_j1939.c** file in subsys/net/lib/sockets directory to handle the socket type: **SOCK_DGRAM** and protocol: **CAN_J1939**

That is the way to go if you want support socket interface for this protocol. Your plan looks ok to me.

alexanderwachter, 2020-12-18:
@Navin-Sankar don't forget that you are not allowed to copy any code from the Linux kernel (GPL licensed) to the Zephyr repository. Only Apache licensed or compatible code is allowed.

wysman, 2021-03-17:
@Navin-Sankar do you have some news about you work on J1939 ? I may be interested to use it too.

henrikbrixandersen, 2022-01-18:
@Navin-Sankar Any work done on this yet that you can share?

alexanderwachter, 2022-02-23:
No progress here. We can reopen if there is progress on this issue

closed 2022-02-23 by alexanderwachter