zephyrproject-rtos/zephyr #30730: tests: nrf: Tests in tests/drivers/timer/nrf_rtc_timer are flaky

PerMac, 2020-12-15:
**Describe the bug**
The tests in tests/drivers/timer/nrf_rtc_timer are flaky (both nrf_rtc_timer and nrf_rtc_timer_no_zli). I didn't run statistics but the chances for a fail look high and it is hard to get a clear run. I only saw test_reseting_cc failing in this suite. 

**To Reproduce**
Steps to reproduce the behavior:
1. go to zephyr dir
2. have nrf platform connected (e.g. nrf52dk_nrf52832)
3. run ./scripts/sanitycheck --device-testing -T tests/drivers/timer/nrf_rtc_timer/ -p nrf52dk_nrf52832 --device-serial /dev/ttyACM0 -v -v
4. See error

**Expected behavior**
Test results are stable

**Impact**
Annoyance, the tests requires multiple retries

**Logs and console output**
Examples of failures
```
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_reseting_cc
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/timer/nrf_rtc_timer/src/main.c:62: timeout_handler: (diff <= data->delay is false)
DEBUG   - DEVICE: Handler called in wrong time (187863), set cc: 187866, got cc: 187866
DEBUG   - DEVICE: FAIL - test_reseting_cc
DEBUG   - DEVICE: ===================================================================

DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_reseting_cc
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/timer/nrf_rtc_timer/src/main.c:62: timeout_handler: (diff <= data->delay is false)
DEBUG   - DEVICE: Handler called in wrong time (187185), set cc: 187188, got cc: 187188
DEBUG   - DEVICE: FAIL - test_reseting_cc
DEBUG   - DEVICE: ===================================================================

```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain ZEphyr SDK
 - Commit SHA or Version used zephyr-v2.4.0-2419-gd35d26d2500a

JordanYates, 2021-01-08:
I believe I am seeing this same issue in real applications.
`rtc_nrf_isr` is running before `RTC->COUNTER` reaches the value present in `RTC->CC[0]`.
`dticks`becomes negative if `rtc_nrf_isr` runs again before the original value in `RTC->CC[0]` is reached by the counter.
This is happening under a relatively high interrupt load from a UART accelerometer.

nordic-krch, 2021-01-08:
@JordanYates are you on the latest master? Should have been fixed by #30487

nordic-krch, 2021-01-08:
@PerMac can you close if test is no longer failing?
@JordanYates can you open new issue if you still see it?

PerMac, 2021-01-08:
@nordic-krch but was anything change in regard to the reported issue? With the current master, I still see these tests failing more often than passing.

nordic-krch, 2021-01-08:
@PerMac reproduced it locally. Investigating...

PR #31187 (2021-01-10): drivers: timer: nrf_rtc_timer: Increase half tick busy wait
Due to clock discrepancy, busy waiting for 15us was not covering for
half tick in certain cases. Busy wait runs from HF clock source.

Increased to 19us to cover it. Anyway, this case is hit very rarely,
only when there was aborted, not-cancelled compare value that was
about to expire. Because of that, increase shall not impact the
performance.

Fixes #30730.
closed 2021-01-10 by nashif