zephyrproject-rtos/zephyr #30736: Deadlock with usb_transfer_sync()

ktolonen, 2020-12-15:
Calling usb_transfer_sync() from multiple thread with same endpoint address can cause deadlock.

Example code to reproduce can found from wpanusb sample where same end point is used from tx and rx context. USB_MAX_NUMBER_TRANSFERS shall be greater than 1 (default 4). Race condition is hard to get in practice and I don't have test code to reproduce the issue.

Based on debugging and code review issue seems to be with usb_ep_get_transfer() function implementation when called from usb_transfer_ep_callback().

```c
static struct usb_transfer_data *usb_ep_get_transfer(uint8_t ep)
{
	for (int i = 0; i < ARRAY_SIZE(ut_data); i++) {
		if (ut_data[i].ep == ep) {
			return &ut_data[i];
		}
	}
	return NULL;
}
```

This function always returns first transfer with endpoint address and doesn't care the transfer status. When there is 2 or more transfers with same endpoint address, this might return completed transfer and the ongoing transfer is not handled. This will lead to situation when transfer completion callback is never called to release caller semaphore.

An easy fix is just to return first ongoing transfer for endpoint callback.
```c
static struct usb_transfer_data *usb_ep_get_transfer(uint8_t ep)
{
	for (int i = 0; i < ARRAY_SIZE(ut_data); i++) {
		if (ut_data[i].ep == ep && ut_data[i].status != 0) {
			return &ut_data[i];
		}
	}
	return NULL;
}
```
If this is a valid fix to above issue, I can make a pull request for the change.

jfischer-no, 2020-12-15:
@ktolonen yes usb_transfers were not made to handle multiple transfers for the same endpoint simultaneously.
Please go ahead and submit PR.

ktolonen, 2020-12-15:
PR is ready for review.

PR #30743 (2021-01-24): usb: fix for parallel transfer deadlock with usb_transfer_sync() 
Transfer status was not checked when accessing to transfer from endpoint callback. Adding status check, to prevent double complete. Double complete will cause deadlock when ongoing transfer complete callback is not called.

Fixes #30736
commit 22c0d015 (2021-01-24): usb: fix for parallel transfer deadlock with usb_transfer_sync()

Parallel transfer to same endpoint is not supported and
may cause a deadlock. Adding a check to prevent starting
the transfer if tranfer is already ongoing on same endpoint.

Transfer status was not checked when accessing to transfer
from endpoint callback. Adding status check, to prevent
a double completion.

Fixes #30736
closed 2021-01-24 by nashif