zephyrproject-rtos/zephyr #30771: Logging: Fault instruction address

LingaoM, 2020-12-16:
**Describe the bug**
Every time power on, it will trigger the crash exception.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. west build -b nrf52840dk_nrf52840 zephyr/tests/bluetooth/mesh_shell/
3. west build -t menuconfig

[00:00:17.732,574] <dbg> bt_mesh_crypto.bt_mesh_beacon_auth: BeaconKey 001245a59b55cee65ede2ecd3706dba~
[00:00:17.732,574] <dbg> bt_mesh_crypto.bt_mesh_beacon_auth: NetId c358e9488149e865
[00:00:17.732,574] <dbg> bt_mesh_crypto.bt_mesh_beacon_auth: IV Index 0x00000000
[00:00:17.732,604] <dbg> bt_mesh_crypto.bt_mesh_beacon_auth: BeaconMsg 00c358e9488149e86500000000
[00:00:17.733,123] <dbg> bt_mesh_proxy.bt_mesh_proxy_adv_stop: adv_enabled 0
[00:00:17.734,802] <dbg> bt_mesh_proxy.bt_mesh_proxy_gatt_enable: 
[00:00:17.833,435] <err> os: r0/a1:  0x00000004  r1/a2:  0x0000006d  r2/a3:  0x200038d0
[00:00:17.833,465] <err> os: r3/a4:  0x00000001 r12/ip:  0x00000000 r14/lr:  0x00004e8b
[00:00:17.833,465] <err> os:  xpsr:  0x61000000
[00:00:17.833,496] <err> os: Faulting instruction address (r15/pc): 0x0002ef4c
[00:00:17.833,496] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:17.833,526] <err> os: Current thread: 0x20000e08 (logging)
[00:00:18.510,375] <err> os: Halting system

LingaoM, 2020-12-16:
[autoconf.zip](https://github.com/zephyrproject-rtos/zephyr/files/5701281/autoconf.zip)

ioannisg, 2021-01-04:
@trond-snekvik maybe you could take a look at this mesh-related test issue?

trond-snekvik, 2021-01-04:
@ioannisg This error is happening on the logging thread, I can't quite see how the mesh shell module is interfering.
Deferring to shell module code owners, but I'll stick around in case there's something wrong in the mesh usage of shell.

carlescufi, 2021-01-26:
/home/lingao/zephyrproject/zephyr/subsys/shell/shell_log_backend.c:190:
https://github.com/zephyrproject-rtos/zephyr/blob/889e37b8d9918f28287a2b0dda2529910dd0c068/subsys/shell/shell_log_backend.c#L190

@LingaoM is this the same line you see on your local copy?

nordic-krch, 2021-01-27:
@LingaoM can you attach `prj.conf` file that was used. It is hard to compare autoconf.

closed 2021-04-12 by github-actions[bot]