zephyrproject-rtos/zephyr #30792: Cannot build network echo_server for nucleo_f767zi

jukkar, 2020-12-16:
**Describe the bug**
Following dts errors are shown when trying to compile echo_server sample app for nucleo_f767zi

```
west build -d build -b nucleo_f767zi samples/net/sockets/echo_server
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): ./zephyr/cmake/app/boilerplate.cmake
-- Application: ./zephyr/samples/net/sockets/echo_server
-- Zephyr version: 2.4.99 (./zephyr)
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nucleo_f767zi
-- Cache files will be written to: ~/.cache/zephyr
-- Found toolchain: zephyr (/opt/zephyr-sdk)
-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: ./zephyr/boards/arm/nucleo_f767zi/nucleo_f767zi.dts
nucleo_f767zi.dts.pre.tmp:329.8-335.6: ERROR (phandle_references): /soc/timers@40010000/pwm: Reference to non-existent node or label "tim1_ch3_pe13"

  also defined at nucleo_f767zi.dts.pre.tmp:1937.12-1940.4
nucleo_f767zi.dts.pre.tmp:580.22-588.5: ERROR (phandle_references): /soc/adc@40012000: Reference to non-existent node or label "adc1_in0_pa0"

  also defined at nucleo_f767zi.dts.pre.tmp:1962.7-1965.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_mdc_pc1"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_rxd0_pc4"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_rxd1_pc5"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_ref_clk_pa1"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_mdio_pa2"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_crs_dv_pa7"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_tx_en_pg11"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_txd0_pg13"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
nucleo_f767zi.dts.pre.tmp:689.26-701.5: ERROR (phandle_references): /soc/ethernet@40028000: Reference to non-existent node or label "eth_txd1_pb13"

  also defined at nucleo_f767zi.dts.pre.tmp:1969.6-1980.3
ERROR: Input tree has errors, aborting (use -f to force output)
CMake Error at ./zephyr/cmake/dts.cmake:209 (message):
  command failed with return code: 2
Call Stack (most recent call first):
  ./zephyr/cmake/app/boilerplate.cmake:506 (include)
  ./zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  ./zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: ~/.local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B./zephyr/build -S./zephyr/samples/net/sockets/echo_server -GNinja
```

**To Reproduce**
`west build -d build -b nucleo_f767zi samples/net/sockets/echo_server`

**Environment (please complete the following information):**
 - OS: Fedora 33
 - SDK 0.11.3
 - Commit SHA: 7a3f9c4e391f75558c74cbebcd503f16c6ed047c

erwango, 2020-12-17:
@jukkar can you try `west udpate` ?

jukkar, 2020-12-17:
Indeed that was the case. I forgot to update the HAL. Sorry for the noise.

closed 2020-12-17 by jukkar