zephyrproject-rtos/zephyr #30800: STM32 usb clock from PLLSAI1

qin-zou, 2020-12-17:
I have external clock source (8MHz) for my STM32L452.
I want to configure my USB to use PLL as clock source (PLLSAI1).

Here is my current config.
```
# 80MHz system clock
CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=80000000

# Clock Configuration
CONFIG_CLOCK_CONTROL=y

# Use PLLCLK for SYSCLK
CONFIG_CLOCK_STM32_SYSCLK_SRC_PLL=y

# External clock source (8MHz)
CONFIG_CLOCK_STM32_PLL_SRC_HSE=y
CONFIG_CLOCK_STM32_HSE_BYPASS=y
CONFIG_CLOCK_STM32_PLL_M_DIVISOR=1
CONFIG_CLOCK_STM32_PLL_N_MULTIPLIER=20
CONFIG_CLOCK_STM32_PLL_P_DIVISOR=7
CONFIG_CLOCK_STM32_PLL_Q_DIVISOR=2
CONFIG_CLOCK_STM32_PLL_R_DIVISOR=2

CONFIG_CLOCK_STM32_AHB_PRESCALER=1
CONFIG_CLOCK_STM32_APB1_PRESCALER=1
CONFIG_CLOCK_STM32_APB2_PRESCALER=1
```

With the above configuration, USB clock source is still from HSI48.
I am wondering if it's possible to create the clock configuration like the picture attached using zephyr config parameters.

[image]

erwango, 2020-12-17:
@qin-zou This setting is not available yet. It has to be implemented.
It would need to be implemented directly in usb driver: `drivers/usb/device/usb_dc_stm32.c`, function `usb_dc_stm32_clock_enable`. We have no plan to implement this for now, but we can provide support.

qin-zou, 2020-12-17:
Thanks, I'll do some reading on it.

qin-zou, 2020-12-17:
```
diff --git a/drivers/usb/device/usb_dc_stm32.c b/drivers/usb/device/usb_dc_stm32.c
index c2594a1..707262b 100644
--- a/drivers/usb/device/usb_dc_stm32.c
+++ b/drivers/usb/device/usb_dc_stm32.c
@@ -232,8 +232,18 @@ static int usb_dc_stm32_clock_enable(void)
         * that instead.  Example reference manual RM0360 for
         * STM32F030x4/x6/x8/xC and STM32F070x6/xB.
         */
-#if defined(RCC_HSI48_SUPPORT) || defined(CONFIG_SOC_SERIES_STM32WBX)
-
+#if defined(CONFIG_BOARD_STM32T)
+        /* HSE is 8Mhz for our board */
+        LL_RCC_PLLSAI1_Disable();
+        LL_RCC_PLLSAI1_ConfigDomain_48M(LL_RCC_PLLSOURCE_HSE,
+                                        LL_RCC_PLLM_DIV_1,
+                                        12, LL_RCC_PLLSAI1Q_DIV_2);
+        LL_RCC_PLLSAI1_Enable();
+        LL_RCC_PLLSAI1_EnableDomain_48M();
+        while (LL_RCC_PLLSAI1_IsReady() != 1) {
+        }
+        LL_RCC_SetUSBClockSource(LL_RCC_USB_CLKSOURCE_PLLSAI1);
+#elif defined(RCC_HSI48_SUPPORT) || defined(CONFIG_SOC_SERIES_STM32WBX)
        /*
         * In STM32L0 series, HSI48 requires VREFINT and its buffer
         * with 48 MHz RC to be enabled.
```

@erwango Is this implementation ok? The drawback is that it's board specific.

erwango, 2020-12-18:
@qin-zou Seems correct. What you can do is add a new Kconfig symbol (such as `USB_DC_STM32_CLOCK_PLLSAI` for instance) in `drivers/usb/device/Kconfig` to control this.
This new symbol would be STM32 specific. You just need to care about the dependencies (at least depends on SOC_STM32L4..).

qin-zou, 2020-12-18:
Thanks,
Qin

erwango, 2021-01-04:
@qin-zou Let me know if this point can be closed or if you have further questions

qin-zou, 2021-01-04:
@erwango You can go ahead and close this. Thanks.

closed 2021-01-04 by erwango