zephyrproject-rtos/zephyr #30805: Build error at tests/kernel/queue in mec15xxevb_assy6853 platform

Zhaoningx, 2020-12-17:
I used command `twister -p mec15xxevb_assy6853 -T tests/kernel/queue/` to run in my master branch,
a error is met, the error log as below:
...
[ 86%] Linking C static library libdrivers__gpio.a
[ 86%] Built target drivers__gpio
Scanning dependencies of target drivers__i2c
[ 87%] Building C object zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_common.c.obj
[ 88%] Building C object zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_mchp_xec.c.obj
/home/zn/zephyrproject/zephyr/drivers/i2c/i2c_mchp_xec.c: In function 'xec_spin_yield':
/home/zn/zephyrproject/zephyr/drivers/i2c/i2c_mchp_xec.c:127:17: error: 'TIMEOUT' undeclared (first use in this function); did you mean 'ETIMEDOUT'?
  127 |  if (*counter > TIMEOUT) {
      |                 ^~~~~~~
      |                 ETIMEDOUT
/home/zn/zephyrproject/zephyr/drivers/i2c/i2c_mchp_xec.c:127:17: note: each undeclared identifier is reported only once for each function it appears in
zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/build.make:95: recipe for target 'zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_mchp_xec.c.obj' failed
...

And I have already used command `git pull` and `west update` to get the lasting commit to run.

So how to deal with this?

cfriedt, 2020-12-17:
This bug seems to be causing tests to fail for all of the PR's I have open. I presume it's affecting all PR's at the moment.

cfriedt, 2020-12-17:
Looks like it was fixed in af2ba71e18bd7f8383d897b36b1a54e62655bee7?

cfriedt, 2020-12-17:
Yeah, lookes like it was fixed by af2ba71e18bd7f8383d897b36b1a54e62655bee7

nashif, 2020-12-17:
fix merged.

closed 2020-12-17 by nashif